ARMmbed/mbed-ls #22: Consistently map platform name to board rather than target

c1728p9, 2015-09-23:
Currently platform name is a mix of board names and target processor names.  In the case that they map to target names there can be duplicate boards with the same target such as:
        "1040": "LPC11U24",
        "1045": "LPC11U24",

This can cause problems for automated testing since you can't be certain of the board you are connecting to.  It would be nice if these consistently mapped to only the board name.  It would be even better if these names mapped to the names in https://developer.mbed.org/platforms.  This would allow the platform name to be used with the compile API.  Let me know what you think.  I can open a pull request to change the names over if no one has concerns with this.

sg-, 2015-09-24:
I think an additional parameter for the platform page (compile api) would be good but the other problem is that there is no mcu specific identifier. I think this is a good step but we need much more though about a device database.

PrzemekWirkus, 2015-09-24:
Hi,

_Currently platform name is a mix of board names and target processor names._.

Platform name is not mix of both, it is a regular target name from mbed 2.0 - poor naming convention aside those are particular target names. See: https://github.com/mbedmicro/mbed/blob/master/workspace_tools/targets.py

I think we have to be consistent and can't just change that.

PrzemekWirkus, 2015-09-24:
I think I understand your concern better after having a coffee :)
I think ```platform_name``` is a very unfortunate name given and as you said can create a lot of confusion:
```
$mbedls
+--------------+---------------------+------------+------------+-------------------------+
|platform_name |platform_name_unique |mount_point |serial_port |target_id                |
+--------------+---------------------+------------+------------+-------------------------+
|K64F          |K64F[0]              |E:          |COM142      |024002011E031E6AE3FDE3D2 |
+--------------+---------------------+------------+------------+-------------------------+
```
Should maybe be:
* platform_name = ```FRDM-K64F```
* target_name = ```K64F``` (for mbed 2.0 backward compatibility.

I've requested web service from web team which should return proper tuples of data about each device according to TargetID. Unfortunately web team has other priorities now.

In the future: I want ```mbed-ls``` to just fetch data from mbed.com web service. This service would provide read-only data about device from "vendor/platform" database table/view.

0xc0170, 2015-09-24:
@PrzemekWirkus We understand limited python :grin:, why not to create a python module with this type of information - mbed board definitions or database or name it as you wish.

We did some small exercise for mbed classic with that class wrapper. That could be another one - transition phase for a while (we can even print warnings - using deprecated target name for example for K64F, a message: "Using deprecated target name - K64F - please switch to (fetch new name for K64F from database - FRDM-K64F)".

PrzemekWirkus, 2015-09-29:
### Do not read this....
1. There should be a web-service with more detailed target/platform/vendor information available soon I hope. I can show you relevant JIRA ticket when you are back to office.
2. We can't just create a module, who will maintain it and track changes.
3. Who will make sure it is compatible with online IDE (mbed 2.0?).
4. I can't just release mbed-ls with platform_name reporting different thing now, there are too many dependencies to other module to this naming convention.
5. We are now developing test tools, 0.1.x. When we update to 0.2.x we can change interface (sic!) because will will not keep interface compatibility between 0.1.x and 0.2.x -> Implication is 
This is because we increment Agile way on tools.
6. I know we've prepared nice wrapper for names in mbed 2.0, it will not propagate to mbed 3.0 just like that, you need to make it 2 & 3.0 ready. Note, we are using yotta registry to map targets to mbed 2.0 platform_names (```yotta search``` command). So your changes would impact yotta registry entries.

I still opt for one place we store target/platform/vendor information (database with RESTful API so mbed-ls is a very thin client and only queries host computer for USB devices) so we can fetch it and internally sync between it and e.g. yotta registry.

### Boring history lesson ;(
Things like:
```
"1040": "LPC11U24",
"1045": "LPC11U24",
```
is a consequence of decisions made long time ago: that maybe different platforms (HW boards) have the same HAL and we do not have to differentiate at all. That time it was probably a short-cut or just simplification but project got bigger and here we are!

### Just read this part...
Let's patiently wait for web service with proper information and we can add new field(s) to each detected device to ease automation! What do you think!?

closed 2016-07-04 by PrzemekWirkus