ARMmbed/mbed-ls #25: Why is the Linux support Ubuntu specific?

jrobeson, 2015-10-25:
The code seems pretty generic for linux, but maybe I missed something.  I think the only thing I see that's distro specific is the usage of the /media directory by default.

simonqhughes, 2015-10-25:
ARM Internal Ref: IOTSYST-476

PrzemekWirkus, 2015-10-26:
Hi Johnny,
For Ubuntu Linux distro we have a proper port we've developed, we maintain and, test on our infrastructure.

Nothing stops us as a community from adding a new Linux distro support to mbed-ls's code base but we have to remember that we then have to commit to support and test. It is all about working software others can use seamlessly in the end.

We are still improving our infrastructure and this is why we are currently concentrating (and limiting) our efforts of limited number of supported platforms.

If anyone wants to port mbed-ls they can, and to verify their port they can use packages's unit tests:
```
$ git clone <mbed-ls>
$ cd mbed-ls
... Make your changes ...
$ python setup.py test
```

jrobeson, 2015-10-26:
isn't it recommended to use udisks api on ubuntu instead of calling the programs directly?

jrobeson, 2015-10-26:
that's another issue though. I will file that separately.

PrzemekWirkus, 2015-10-26:
Cool, you are also more than welcome to send us pull request using udisks on Ubuntu to replace current implementation!

PrzemekWirkus, 2015-11-05:
We have promising PR https://github.com/ARMmbed/mbed-ls/pull/26 to fix this issue.

PrzemekWirkus, 2015-12-23:
LInux support was merged to master. We will have release soon! :)

closed 2015-12-23 by PrzemekWirkus