ARMmbed/mbed-ls #38: mbedls fails to list devices on OS X El Capitan

Patater, 2016-01-07:
This is due to the new unified (between iOS and OS X) USB stack. The data returned from `ioreg -a -r -n AppleUSBXHCI -l` is different on El Capitan.

PrzemekWirkus, 2016-01-13:
Ooly thing I can do for now is to add documentation entry regarding this issue: https://github.com/ARMmbed/mbed-ls/pull/39

flit, 2016-03-04:
All the required data is available in the ioreg tool output. This is on El Capitan 10.11.3.

Run `ioreg -a -r -c IOUSBHostDevice -l'. Walk the tree looking for the CMSIS-DAP `IOUSBDevice` instance. Once found, examine its children recursively. You should look for children with classes in this order:

1. `IOUSBInterface` with `bInterfaceClass` and `bInterfaceSubClass` properties matching USB MSC (class==0x8, subclass==0x6)
2. `IOUSBMassStorageInterfaceNub`
3. `IOUSBMassStorageDriverNub`
4. `IOUSBMassStorageDriver`
5. `IOSCSILogicalUnitNub`
6. `IOSCSIPeripheralDeviceType00`
7. `IOBlockStorageServices`
8. `IOBlockStorageDriver`
9. `IOMedia`. This will actually have an `IOObjectClass` property value of something like "MBED microcontroller Media", but it's the only child of the `IOBlockStorageDriver`.

The `IOMedia` object has the `BSD Name` property you are looking for.

Or you can just search all children of the matching `IOUSBDevice` for the `BSD Name` property.

sg-, 2016-07-27:
Seems like this should be fixed given we've got a suggestion on how to fix. Thanks @flit

sg-, 2016-08-04:
https://github.com/ARMmbed/mbed-ls/pull/115

closed 2016-08-04 by sg-