ARMmbed/mbed-ls #44: Boards in bootloader mode show up on Windows but not Linux

c1728p9, 2016-01-26:
Running mbedls on Linux does not show any boards in bootloader mode.  Windows shows connected boards in bootloader mode even though they are unknown:
```
+---------------------+----------------------------+-------------------+-------------------+--------------------------------------------------------+
|platform_name        |platform_name_unique        |mount_point        |serial_port        |target_id                                               |
+---------------------+----------------------------+-------------------+-------------------+--------------------------------------------------------+
|unknown              |None[0]                     |D:                 |unknown            |0000000026334e450053400f5e0e00134eb10000646c0000        |
+---------------------+----------------------------+-------------------+-------------------+--------------------------------------------------------+
```

Is this a bug in the Linux implementation?  Would it be possible to add support for boards in bootloader mode?  This would be very useful for DAPLink testing.

c1728p9, 2016-01-26:
Pushed up #45 which fixes this on Linux.  Not sure if this is the appropriate fix, but it works for me.  Let me know what you think.

PrzemekWirkus, 2016-01-26:
This is not a bug, we just do not support boards in bootloader mode at all in mbed-ls.
But I can see you wanna add this feature... so this is a feature request. I will check your PR and if it is OK I will merge, if not I will let you know how it looks like.

c1728p9, 2016-01-26:
I put up PR #46 to add support for this on Linux.

PrzemekWirkus, 2016-01-26:
Released with v0.1.27

closed 2016-01-26 by PrzemekWirkus