ARMmbed/mbed-ls #53: daplink version is not shown when calling mbedls

jupe, 2016-02-24:
When calling mbedls without parameters it show `unknown` as daplink_version, json format seems to work correctly. Tested in ubuntu 14.04 VM.
```
jusvat01@ubuntu:~/git/github/ARMmbed/raas-daemon$ mbedls 
+---------------+----------------------+-------------------------+--------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point             | serial_port  | target_id                                        | daplink_version |
+---------------+----------------------+-------------------------+--------------+--------------------------------------------------+-----------------+
| K64F          | K64F[0]              | /media/jusvat01/DAPLINK | /dev/ttyACM0 | 0240000028884e45002500106bf000448021000097969900 | unknown         **|**
```
json format show correctly
```
jusvat01@ubuntu:~/git/github/ARMmbed/raas-daemon$ mbedls --json
[
    {
        "daplink_#_daplink_firmware_-_see_https": "//mbed.com/daplink", 
        "daplink_auto_reset": "0", 
        "daplink_automation_allowed": "0", 
        "daplink_daplink_mode": "Interface", 
        "daplink_git_sha": "c765cbb590f57598756683254ca38b211693ae5e", 
        "daplink_hif_id": "97969900", 
        "daplink_interface_crc": "0x26764ebf", 
        "daplink_interface_version": "0240", 
        "daplink_local_mods": "0", 
        "daplink_unique_id": "0240000028884e45002500106bf000448021000097969900", 
        "daplink_usb_interfaces": "MSD, CDC, HID", 
        "mount_point": "/media/jusvat01/DAPLINK", 
        "platform_name": "K64F", 
        "platform_name_unique": "K64F[0]", 
        "serial_port": "/dev/ttyACM0", 
        "target_id": "0240000028884e45002500106bf000448021000097969900", 
        "target_id_mbed_htm": "0240000028884e45002500106bf000448021000097969900", 
        "target_id_usb_id": "0240000028884e45002500106bf000448021000097969900"
    }
]
```

jupe, 2016-02-24:
looks that this related some how to custom daplink firmware.

PrzemekWirkus, 2016-02-24:
Please copy-paste DETAILS.TXT here and MBED.HTM for debug :)

PrzemekWirkus, 2016-02-24:
mbed.html:
```html
<!doctype html>
<!-- mbed Platform Website and Authentication Shortcut -->
<html>
<head>
<meta charset="utf-8">
<title>mbed Website Shortcut</title>
</head>
<body>
<script>
window.location.replace("https://mbed.org/device/?code=0240000029164e45002f0012706e0006f301000097969900?version=0240?target_id=0007ffffffffffff4e45315450090023");
</script>
</body>
</html>
```

DETAILS.TXT:
```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 0240000029164e45002f0012706e0006f301000097969900
HIF ID: 97969900
Auto Reset: 0
Automation allowed: 0
Daplink Mode: Interface
Interface Version: 0240
Git SHA: c765cbb590f57598756683254ca38b211693ae5e
Local Mods: 0
USB Interfaces: MSD, CDC, HID
Interface CRC: 0x26764ebf
```

PrzemekWirkus, 2016-02-25:
There is an issue because you are using new version of DAPlink, 0240. I will try to align with it anyway!

closed 2016-02-28 by PrzemekWirkus