ARMmbed/mbed-ls #57: mbedls produces inexistent devices when run on VM

VeliMattiLahtela, 2016-03-02:
Host:
Windows 7, VMware Player.
VM:
Ubuntu 14.04 LTS.

Host machine has 5 mbed devices connected to it. But only 4 of them could be attached to the VM.
The fifth is for some reason pointing to the same "sdc" as one of the other devices as seen on the debug.
[error_situation.txt](https://github.com/ARMmbed/mbed-ls/files/154667/error_situation.txt)

This creates a nice effect on the output of mbedls.
The working device is shown as two separate connected devices with the exception of missing serial_port on the other.

PrzemekWirkus, 2016-03-02:
Can you please paste here output of these commands when all devices are connected:
```
$ ls -oA /dev/serial/by-id/
$ ls -oA /dev/disk/by-id/
```

PS: It looks like some VM configuration error because two different mbed devices are mounted under the same /media/DAPLINK directory. But before we establish this let's dig deeper! :)
```
ata-VMware_Virtual_IDE_CDROM_Drive_10000000000000000001 -> ../../sr0
usb-MBED_VFS_0240000028634e4500135006691700105f21000097969900-0:0 -> ../../sdb
usb-MBED_VFS_0240000028884e450019700f6bf0000f8021000097969900-0:0 -> ../../sdd
usb-MBED_VFS_0240000028884e45002f700f6bf0000e8021000097969900-0:0 -> ../../sdc
usb-MBED_VFS_0240000028884e450051700f6bf000128021000097969900-0:0 -> ../../sdc
usb-MBED_VFS_0240000029164e4500590012706e001df301000097969900-0:0 -> ../../sde
```

VeliMattiLahtela, 2016-03-15:
Hi.

Sorry for the wait.

Just tried it and straight away hit the same or even worse situation:

```
ls -oA /dev/serial/by-id/
total 0
lrwxrwxrwx 1 root 13 Mar 15 08:35 usb-ARM_DAPLink_CMSIS-DAP_0240000028634e4500135006691700105f21000097969900-if01 -> ../../ttyACM0
lrwxrwxrwx 1 root 13 Mar 15 08:35 usb-ARM_DAPLink_CMSIS-DAP_0240000028884e450018700f6bf000338021000097969900-if01 -> ../../ttyACM1
lrwxrwxrwx 1 root 13 Mar 15 08:35 usb-ARM_DAPLink_CMSIS-DAP_0240000028884e450036700f6bf000118021000097969900-if01 -> ../../ttyACM3
lrwxrwxrwx 1 root 13 Mar 15 08:35 usb-ARM_DAPLink_CMSIS-DAP_0240000029164e45001b0012706e000df301000097969900-if01 -> ../../ttyACM2
```

```
ls -oA /dev/disk/by-id/
total 0
lrwxrwxrwx 1 root 9 Mar 14 07:58 ata-VMware_Virtual_IDE_CDROM_Drive_10000000000000000001 -> ../../sr0
lrwxrwxrwx 1 root 9 Mar 15 08:35 usb-MBED_VFS_0240000028634e4500135006691700105f21000097969900-0:0 -> ../../sdb
lrwxrwxrwx 1 root 9 Mar 15 08:35 usb-MBED_VFS_0240000028884e450018700f6bf000338021000097969900-0:0 -> ../../sdc
lrwxrwxrwx 1 root 9 Mar 14 08:44 usb-MBED_VFS_0240000028884e45001f700f6bf000118021000097969900-0:0 -> ../../sdd
lrwxrwxrwx 1 root 9 Mar 15 08:35 usb-MBED_VFS_0240000028884e450036700f6bf000118021000097969900-0:0 -> ../../sde
lrwxrwxrwx 1 root 9 Mar 15 08:35 usb-MBED_VFS_0240000029164e45001b0012706e000df301000097969900-0:0 -> ../../sdd
lrwxrwxrwx 1 root 9 Mar 14 08:44 usb-MBED_VFS_0240000029164e45002f0012706e0006f301000097969900-0:0 -> ../../sdc
```

marhil01, 2016-03-18:
Got this repeated with native linux computer: raspberry pi:
pi@raas_daemon_1:~ $ mbedls
+---------------+----------------------+-------------+--------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port  | target_id                                        | daplink_version |
+---------------+----------------------+-------------+--------------+--------------------------------------------------+-----------------+
| K64F          | K64F[0]              | /media/usb1 | unknown      | 0240000028884e45002d700f6bf000138021000097969900 | 0241            |
| K64F          | K64F[1]              | /media/usb2 | unknown      | 0240000028884e450012700f6bf000338021000097969900 | 0241            |
| K64F          | K64F[2]              | /media/usb1 | /dev/ttyACM0 | 0240000028884e45002d700f6bf000138021000097969900 | 0241            |
| K64F          | K64F[3]              | /media/usb0 | /dev/ttyACM1 | 0240000028884e450052700f6bf000288021000097969900 | 0241            |
| K64F          | K64F[4]              | /media/usb2 | /dev/ttyACM2 | 0240000028884e450012700f6bf000338021000097969900 | 0241            |
+---------------+----------------------+-------------+--------------+--------------------------------------------------+-----------------+

PrzemekWirkus, 2016-04-04:
We do not support officially Raspbery Pi linux distro so please give us some time to make sure we are getting mbedls right on this platform.

marhil01, 2016-04-04:
Distro is: Raspbian GNU/Linux 8 (jessie)

PrzemekWirkus, 2016-04-20:
@VeliMattiLahtela 

In your system (I checked log you attached) two devices are assigned to the same ```sdc``` device name:
```
lrwxrwxrwx 1 root 9 Mar  2 09:39 usb-MBED_VFS_0240000028884e45002f700f6bf0000e8021000097969900-0:0 -> ../../sdc
lrwxrwxrwx 1 root 9 Mar  2 08:57 usb-MBED_VFS_0240000028884e450051700f6bf000128021000097969900-0:0 -> ../../sdc
```

So there is a conflict for ``` K64F[3]``` later in table with device names. This makes ```mbed-ls``` under Linux confused. I will check the code but actually it is good to see some abnormal behavior.

From your log (see ```sdc``` assigned to device twice).
```
$ ls -oA /dev/disk/by-id/
lrwxrwxrwx 1 root 9 Mar  1 10:06 ata-VMware_Virtual_IDE_CDROM_Drive_10000000000000000001 -> ../../sr0
lrwxrwxrwx 1 root 9 Mar  2 09:39 usb-MBED_VFS_0240000028634e4500135006691700105f21000097969900-0:0 -> ../../sdb
lrwxrwxrwx 1 root 9 Mar  2 09:39 usb-MBED_VFS_0240000028884e450019700f6bf0000f8021000097969900-0:0 -> ../../sdd
lrwxrwxrwx 1 root 9 Mar  2 09:39 usb-MBED_VFS_0240000028884e45002f700f6bf0000e8021000097969900-0:0 -> ../../sdc
lrwxrwxrwx 1 root 9 Mar  2 08:57 usb-MBED_VFS_0240000028884e450051700f6bf000128021000097969900-0:0 -> ../../sdc
lrwxrwxrwx 1 root 9 Mar  2 09:39 usb-MBED_VFS_0240000029164e4500590012706e001df301000097969900-0:0 -> ../../sde

$ ls -oA /dev/serial/by-id/
lrwxrwxrwx 1 root 13 Mar  2 09:01 usb-ARM_DAPLink_CMSIS-DAP_0240000028634e4500135006691700105f21000097969900-if01 -> ../../ttyACM0
lrwxrwxrwx 1 root 13 Mar  2 09:02 usb-ARM_DAPLink_CMSIS-DAP_0240000028884e450019700f6bf0000f8021000097969900-if01 -> ../../ttyACM2
lrwxrwxrwx 1 root 13 Mar  2 09:01 usb-ARM_DAPLink_CMSIS-DAP_0240000028884e45002f700f6bf0000e8021000097969900-if01 -> ../../ttyACM1
lrwxrwxrwx 1 root 13 Mar  2 09:02 usb-ARM_DAPLink_CMSIS-DAP_0240000029164e4500590012706e001df301000097969900-if01 -> ../../ttyACM3

$ mount | grep vfat
/dev/sdd on /media/iot/DAPLINK1 type vfat (rw,nosuid,nodev,uid=1000,gid=1000,shortname=mixed,dmask=0077,utf8=1,showexec,flush,uhelper=udisks2)
/dev/sdb on /media/iot/DAPLINK2 type vfat (rw,nosuid,nodev,uid=1000,gid=1000,shortname=mixed,dmask=0077,utf8=1,showexec,flush,uhelper=udisks2)
/dev/sde on /media/iot/DAPLINK3 type vfat (rw,nosuid,nodev,uid=1000,gid=1000,shortname=mixed,dmask=0077,utf8=1,showexec,flush,uhelper=udisks2)
/dev/sdc on /media/iot/DAPLINK type vfat (rw,nosuid,nodev,uid=1000,gid=1000,shortname=mixed,dmask=0077,utf8=1,showexec,flush,uhelper=udisks2)

+---------------+----------------------+---------------------+--------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point         | serial_port  | target_id                                        | daplink_version |
+---------------+----------------------+---------------------+--------------+--------------------------------------------------+-----------------+
| K64F          | K64F[0]              | /media/iot/DAPLINK1 | /dev/ttyACM2 | 0240000028884e450019700f6bf0000f8021000097969900 | 0240            |
| K64F          | K64F[1]              | /media/iot/DAPLINK  | /dev/ttyACM1 | 0240000028884e45002f700f6bf0000e8021000097969900 | 0240            |
| K64F          | K64F[2]              | /media/iot/DAPLINK2 | /dev/ttyACM0 | 0240000028634e4500135006691700105f21000097969900 | 0240            |
| K64F          | K64F[3]              | /media/iot/DAPLINK  | unknown      | 0240000028884e45002f700f6bf0000e8021000097969900 | 0240            |
| K64F          | K64F[4]              | /media/iot/DAPLINK3 | /dev/ttyACM3 | 0240000029164e4500590012706e001df301000097969900 | 0240            |
+---------------+----------------------+---------------------+--------------+--------------------------------------------------+-----------------+
```

VeliMattiLahtela, 2016-04-20:
This problem has not been seen anymore, as we have transferred to another device manager.

closed 2016-04-20 by PrzemekWirkus

PrzemekWirkus, 2016-04-22:
Just a note for people using Raspberry Pi and have this issues. Apparently usage of LDM device manager helps.