ARMmbed/mbed-ls #69: Install failed with python3.4.4 in Win10

Neeil, 2016-04-13:
There are about 3 issues occur.
1. Something seems wrong with in the README.md
`$ python setup.py install
Traceback (most recent call last):
  File "setup.py", line 38, in <module>
    long_description=read('README.md'),
  File "setup.py", line 33, in read
    return open(os.path.join(os.path.dirname(__file__), fname)).read()
UnicodeDecodeError: 'gbk' codec can't decode byte 0x9d in position 11668: illega                               l multibyte sequence
`
I replace the README.md with a new empty file. No longer show this.

2. There is a tab in main.py while all others are spaces. Python throw this issue.
 `Sorry: TabError: inconsistent use of tabs and spaces in indentation (lstools_dar                               win.py, line 119)
  File "build\bdist.win-amd64\egg\mbed_lstools\main.py", line 166
    print json.dumps(mbeds.mock_read(), indent=4)
             ^
SyntaxError: invalid syntax`
I remvoe the tab with space, passed.

3. print still follows py2 standard.
`  File "build\bdist.win-amd64\egg\mbed_lstools\lstools_base.py", line 386
    print 'error: %s'% text
                    ^
SyntaxError: Missing parentheses in call to 'print'
`
It seems you may update these to py3.

0xc0170, 2016-04-13:
Hi, thanks for reporting. As I am aware, mbed-ls is not python 3 compatible, neither greentea module for instance.

@PrzemekWirkus

PrzemekWirkus, 2016-04-18:
@Neeil Neil, we do not support Python 3.x with mbed-ls tools. Currently our tools are developed to support Python 2.7.
I'm slowly working on [Python 3 support](https://github.com/ARMmbed/mbed-ls/pull/64) but works goes slowly because we have other higher priorities.

PrzemekWirkus, 2016-04-22:
All I can do is to remove information about Python 3.x from docs and add info that we support 2.7 in description: https://github.com/ARMmbed/mbed-ls/pull/71

closed 2016-04-22 by PrzemekWirkus