ARMmbed/mbed-ls #83: K64F is not detected on OS X

RomanSaveljev, 2016-06-02:
```
$ sudo pip install mbed-ls --upgrade
The directory '/Users/romsav01/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/Users/romsav01/Library/Caches/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting mbed-ls
  Downloading mbed-ls-0.2.10.zip (65kB)
    100% |████████████████████████████████| 65kB 364kB/s 
Requirement already up-to-date: PrettyTable>=0.7.2 in /Library/Python/2.7/site-packages (from mbed-ls)
Installing collected packages: mbed-ls
  Found existing installation: mbed-ls 0.2.8
    Uninstalling mbed-ls-0.2.8:
      Successfully uninstalled mbed-ls-0.2.8
  Running setup.py install for mbed-ls ... done
Successfully installed mbed-ls-0.2.10
You are using pip version 8.0.2, however version 8.1.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
$ mbedls --version
0.2.10
$ mbedls -d

debug @MbedLsToolsDarwin.mbed_lstools.main: Return code: 0
$ mount
/dev/disk3 on /Volumes/DAPLINK (msdos, local, nodev, nosuid, noowners)
$ ls /Volumes/DAPLINK/
DETAILS.TXT	MBED.HTM
$ cat /Volumes/DAPLINK/DETAILS.TXT 
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 0240000028884e450031700f6bf000118021000097969900
HIC ID: 97969900
Auto Reset: 0
Automation allowed: 0
Daplink Mode: Interface
Interface Version: 0241
Git SHA: 34182e2cce4ca99073443ef29fbcfaab9e18caec
Local Mods: 0
USB Interfaces: MSD, CDC, HID
Interface CRC: 0x33e2dd18
$ ls /dev/tty.*
/dev/tty.Bluetooth-Incoming-Port	/dev/tty.usbmodem1411422
```

I suppose everything is in place, but mbedls does not list anything

PrzemekWirkus, 2016-06-03:
Hello @RomanSaveljev,
I think you might be owner of Mac with OSX we do not support due to some porting issues / [known issues](https://github.com/ARMmbed/mbed-ls#known-issues).

RomanSaveljev, 2016-06-03:
@PrzemekWirkus yes, it is exactly the case. I have tried to debug it a little and it looks like it happens due to incompatible output from `ioreg`.

Not sure what is meant by "mounted serial port"? I have a device node for daplink's CDC and I can use it with screen or miniterm.py:

```
$ ls /dev/tty.*
/dev/tty.usbmodem1411422
```

RomanSaveljev, 2016-06-03:
Here you go:

```
$ mbedls -d

debug @MbedLsToolsDarwin.mbed_lstools.main: Return code: 0
```

Is this a correct way to collect the `ioreg` output in a similar format expected by mbedls?

```
for I in AppleUSBXHCI AppleUSBUHCI AppleUSBEHCI AppleUSBOHCI; do echo $I; done | xargs -n 1 -- ioreg -a -r -l -n | tee /tmp/ioreg.log
```
[ioreg.log.zip](https://github.com/ARMmbed/mbed-ls/files/297389/ioreg.log.zip)

PrzemekWirkus, 2016-07-29:
@RomanSaveljev Please try with mbed-ls version v1.2.0 !

RomanSaveljev, 2016-07-29:
Success! Thank you very much @PrzemekWirkus:

```
(.virtualenv) e107920:mbed-ls romsav01$ mbedls
+---------------+----------------------+------------------+-------------------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point      | serial_port             | target_id                                        | daplink_version |
+---------------+----------------------+------------------+-------------------------+--------------------------------------------------+-----------------+
| K64F          | K64F[0]              | /Volumes/DAPLINK | /dev/tty.usbmodem142422 | 0240000033514e450041500585d40006e981000097969900 | 0241            |
+---------------+----------------------+------------------+-------------------------+--------------------------------------------------+-----------------+
```

PrzemekWirkus, 2016-07-29:
@RomanSaveljev Please thank @bulislaw :) I will close this one. Another achievement unlocked! :)

closed 2016-07-29 by PrzemekWirkus