ARMmbed/mbed-ls #85: identify board when daplink is in bootloader mode

jupe, 2016-06-10:
in automation point of view we would like to update daplink softare and identify devices with mbed-ls just like before. Problem is now that mbed-ls give totally different target_id when device is in bootloader mode and therefore we cannot identify same board than when daplink is in normal mode.

Proposal changes:
* `mbed-ls` give information when device is in bootloader or normal mode (in json result only is enough for me)
* `mbed-ls` give exactly same `target_id` in bootloader mode as well as normal mode.
   
   This probably needs more work and investigate how to do, because it might affect daplink implementation as well (@c1728p9 comments?)

Anyway, this is almost mandatory because otherwise we have to create hacky solution to avoid these "missing features".

@PrzemekWirkus @bridadan @c1728p9 @RomanSaveljev

RomanSaveljev, 2016-06-10:
> mbed-ls give information when device is in bootloader or in normal mode (in json result only is enough for me)

How does it help?

jupe, 2016-06-10:
when we are uploading binary we know where it goes ;)

c1728p9, 2016-06-10:
You should be able to use host ID for this.  It has the format:
<4 chars - board ID><4 chars - 4 reserved bytes><32 chars - host ID><8 chars - hic ID>
For example:
0240000025354e45005c400d47d00030bb91000097969900 - K64F interface
0000000025354e45005c400d47d00030bb91000097969900 - K64F bootloader

jupe, 2016-06-10:
makes sense, so there is common part inside target_id:

02400000`25354e45005c400d47d00030bb910000`97969900 - K64F interface
00000000`25354e45005c400d47d00030bb910000`97969900 - K64F bootloader

perhaps mbed-ls should give only that (hostID)  part as target_id, what do you think?
then results are same even device are in bootloader mode..

PrzemekWirkus, 2016-06-14:
Hi guys,
So there is no way to detect device type when device is in bootloader mode at all.
You can see for your K64F example that bootloader `target_id` is stripped from VENDOR/PLATFORM prefix: In case of K64F it is `0240`.
So it will be impossible without storing original `target_id` to detect board correctly.

It boils down to *the way* we can use to detect platform in bootloader mode.
I will be appy to draft a feature to support this but we may need to talk to Russ to check how we can:
* Properly detect bootloader mode
* Detect platform name when platform is in bootloader mode.

@c1728p9 Russ, any way we can detect platform type when in bootloader mode?

PrzemekWirkus, 2016-06-14:
@jupe @RomanSaveljev Up!

c1728p9, 2016-06-14:
Hi @PrzemekWirkus, the same bootloader image can be used across several different boards, so it has no knowledge of which target the board has.  Only the interface firmware known that.  

The reason your bootloader has a different ID is because it is out of date.  This bootloader is from before any automation features were in daplink.

jupe, 2016-06-14:
I think mbedls shouldn't give any platform specific (=e.g. name) when device is in bootloader mode.
Bootloader mode can be detected from `DETAILS.txt` -file:

```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 0240000033514e450041500585d40006e981000097969900
HIC ID: 97969900
Auto Reset: 0
Automation allowed: 0
Daplink Mode: Interface
Interface Version: 0241
Git SHA: 34182e2cce4ca99073443ef29fbcfaab9e18caec
Local Mods: 0
USB Interfaces: MSD, CDC, HID
Interface CRC: 0x33e2dd18
```

**Daplink Mode: Interface**

when in daplink is in bootloader mode there should be text like `Bootloader` or something. @c1728p9 could you verify right text?

c1728p9, 2016-06-14:
Hi @jupe, that is correct the details.txt will show the mode as ```Bootloader```.

PrzemekWirkus, 2016-07-29:
There is no 'DETAILS.txt` in bootloader mode!

RomanSaveljev, 2016-07-29:
@PrzemekWirkus I could not see 'DETAILS.txt' in bootloader mode as well, but then I was told the earlier stage bootloader (not the public one) has to be updated. I did not succeed doing it though..

jupe, 2016-12-05:
This is still blocker for update daplink automatically, any news?

bridadan, 2016-12-08:
@jupe I'd like understand what you want exactly here. Is the following summary correct?

- Detect when a board is in bootloader mode and list the `platform_name` as `bootloader`
    - If we can assume that all boards show up with a target id that begins with `0000` are in bootloader mode then this should be easily doable!
- Change the target ID to only include the 32 character host ID bit that Russ mentioned above.
    - I don't think we should change this, as the full target ID provides lots of good information about the board. Instead I would recommend that the script that's driving the automated DAPLink update look for the host ID part from the target ID to properly match the board.

Am I missing anything?

jupe, 2017-03-08:
> Detect when a board is in bootloader mode and list the platform_name as bootloader

noup, board is still same platform even its in bootloader mode. So please do not touch that field. 

> Change the target ID to only include the 32 character host ID bit that Russ mentioned above.

I agree, this is not what we want to do. Instead I propose to create separate property: `host_id`, which stay exactly same for each device even its in bootloader mode or interface mode. We could use this as "unique id" in test beds.

bridadan, 2017-03-08:
@jupe I think I understand now, please see below:

When I plugin an NRF51_DK for example, I get the following output from mbed-ls:

```
C:\Users\bridan01>mbedls
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| NRF51_DK      | NRF51_DK[0]          | D:          | COM353      | 1100000044203120324c4638323032323932303297969903 | 0243            |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
```

When I then put that device into bootloader mode:
```
C:\Users\bridan01>mbedls
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| unknown       | None[0]              | D:          | unknown     | 0000000044203120324c4638323032323932303297969903 | 0243            |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
```

Note how most of target_id is the same!

So are you proposing to add a separate field called `host_id` that would just contain the characters (in this case) `44203120324c4638323032323932303297969903`?

jupe, 2017-03-08:
exactly. then application doesn't need to have extra logic to detect unique board in different mode..

bridadan, 2017-03-08:
Seems pretty reasonable to me, it'd take some investigation to see how STLink and JLink handle this. Or are really just interested in DAPLink devices?

jupe, 2017-12-08:
@c1728p9 - whether there is any risk of collision if automation rely on "`host_id`" as device unique id ?

c1728p9, 2017-12-12:
Hi @jupe this is intended to be unique and avoid collisions, so it should be safe