ARMmbed/mbed-ls #100: Mocked platforms timeout on tests

BlackstoneEngineering, 2016-06-22:
## Problem
I tried to mock the hexiwear device by doing `mbedls --mock=0214:HEXIWEAR`, this allowed me to then run `singletest.py -j 0 --tc GCC_ARM -V`, but all tests time out. I then instead went and modified mbedls, adding the appropriate line to mbedls to enable hexiwear, re-installed, deleted the build directory, and re-ran the tests. Then all tests worked. 

## Conclusion
Something in the `mock` subcommand doesnt work with tests. Not sure what

## Solution
make mock work with testing infrastructure as this is the primary way we expect users to test their platforms before submitting a PR.

BlackstoneEngineering, 2016-06-22:
Testing on mbedmicro/mbed

PrzemekWirkus, 2016-06-25:
@BlackstoneEngineering Austin did you take under consideration that `--mock` works only in the same directory not globally. Description is in [this](https://github.com/ARMmbed/mbed-ls#mocking-example-with-freescale-k64f-platform) chapter in the description under third block.

I quote: **"We can observe changes immediately. Please note this change only works in the same directory because we save .mbedls-mock file locally:"**.

Please try with pre HEX mbed-ls and this workflow:
```
$ cd mbed
$ cd workspace_tools
$ mbedls --mock=0214:HEXIWEAR
$ singletest.py --auto --tc GCC_ARM -V
```

I was unsure if command you've put here (`singletest.py -j 0 --tc GCC_ARM -V) is exactly what you were using.

I'm assuming that you were actually using `--auto` to detect HEX platform and forced `GCC_ARM` toolchain using `--tc` command line switch.

BlackstoneEngineering, 2016-06-26:
Yes, i took the local folder into consideration. I even had @bridadan double check to make sure I wasn't crazy. Using `--mock` worked in so far as identifying the board correctly, but did not work when trying to run the test suite (singletest.py), however when I modified mbed-ls-git and reinstalled the test suite worked correctly.

This is going to be a terrible usability bug when we try to get people in the general populace / partners to port their stuff with minimal support.

PrzemekWirkus, 2016-07-05:
Done via #106

BlackstoneEngineering, 2016-07-05:
Awesome, closing issue

closed 2016-07-05 by BlackstoneEngineering