ARMmbed/mbed-ls #98: Add Cygwin Support

BlackstoneEngineering, 2016-06-22:
## Problem
Cygwin is not a supported platform and results in this message: 
```
This platform is not supported! Pull requests welcome at github.com/ARMmbed/mbed-ls
```

## Solution
Add support for Cygwin

BlackstoneEngineering, 2016-06-22:
Extra output, using latest of mbed-ls (0.2.14)

```
Requirement already up-to-date: pip in /usr/lib/python2.7/site-packages
╭─ausbla01@U201184 ~
╰─➤  pip search mbed-ls
ls-django-easytests (0.9.5)        - Modified test utils from django-cms as a own module
git-ls-date (0.1.1)                - git-ls-date is git sub command shows first and last commit date.
mbed-connector-api (1.0.3)         - REST interface library for mbed Device Connector (connector.mbed.com)
  INSTALLED: 1.0.3 (latest)
mbed-tools (0.1.14)                - Build and test system for mbed
mbed-greentea (0.2.24)             - mbed 3.0 test suite, codename Greentea. The test suite is a collection of tools that enable
                                     automated testing on mbed-enabled platforms
ls-thrift-py-hadoop (1-cdh4.3.0)   - Hadoop and Hive Python Thrift Libs
ls-django-shop-categories (1.0.4)  - A extendable category app using django-mptt for django-shop
ls-django-treeadmin (0.4.8)        - Tree UI for mptt-managed models, extracted from FeinCMS
mbed-host-tests (0.2.18)           - mbed tools used to flash, reset and supervise test execution for mbed-enabled devices
ls-s3-logs (0.1.2)                 - A library to parse S3 log files.
mbed-ls (0.2.14)                   - mbed-ls is a Python module that detects and lists mbed-enabled devices connected to the host
                                     computer
  INSTALLED: 0.2.14 (latest)
mbed-test-wrapper (0.0.2)          - Wrap the mbed test loader for easy use from yotta targets.
  INSTALLED: 0.0.2 (latest)
╭─ausbla01@U201184 ~
╰─➤  mbed-ls
zsh: command not found: mbed-ls
╭─ausbla01@U201184 ~
╰─➤  mbedls                                                                                                                        127 ↵
This platform is not supported! Pull requests welcome at github.com/ARMmbed/mbed-ls

```

PrzemekWirkus, 2016-07-22:
Just checked my Cygwin and everything looks fine!

```
przwir01@E104178 /usr/bin
$ mbedls
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| K64F          | K64F[0]              | E:          | COM231      | 0240000025354e45003f400b47d00039bb91000097969900 | 0241            |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
```

```
przwir01@E104178 /usr/bin
$ mbedls --version
1.1.4
```

PrzemekWirkus, 2016-07-22:
@BlackstoneEngineering 

Can you update mbed-ls to latest version and:
```
przwir01@E104178 /usr/bin
$ mbedls -d
debug @MbedLsToolsWin7.mbed_lstools.main: mbed-ls ver. 1.1.4
debug @MbedLsToolsWin7.mbed_lstools.main: host: ('nt', 'Windows', '7', '6.1.7601', 'win32')
...
```
I'm interested in info about your Cygwin host.

BlackstoneEngineering, 2016-07-31:
Updated, still seeing the problem. 

```shell
╭─ ~/mbed/test/mbed-os-example-blinky  ‹master*›
╰─➤  pip install -U mbed-ls                                                                       1 ↵
Collecting mbed-ls
  Downloading mbed-ls-1.2.0.zip (73kB)
    100% |████████████████████████████████| 81kB 1.4MB/s
Requirement already up-to-date: PrettyTable>=0.7.2 in /usr/lib/python2.7/site-packages/prettytable-0.7.2-py2.7.egg (from mbed-ls)
Requirement already up-to-date: lockfile in /usr/lib/python2.7/site-packages (from mbed-ls)
Installing collected packages: mbed-ls
  Found existing installation: mbed-ls 1.1.5
    Uninstalling mbed-ls-1.1.5:
      Successfully uninstalled mbed-ls-1.1.5
  Running setup.py install for mbed-ls ... done
Successfully installed mbed-ls-1.2.0
╭─~/mbed/test/mbed-os-example-blinky  ‹master*›
╰─➤  mbedls -d
This platform is not supported! Pull requests welcome at github.com/ARMmbed/mbed-ls
```

bridadan, 2019-01-11:
I am not going to transfer this particular issue since its mbed-ls specific. Instead I'm going to make a general "investigate cygwin support" issue in mbed-os-tools (since the other packages don't support it either): https://github.com/ARMmbed/mbed-os-tools/issues/86.

closed 2019-01-11 by bridadan