ARMmbed/mbed-ls #99: mock needs to be global

BlackstoneEngineering, 2016-06-22:
## Problem
When using the `--mock` subcommand it creates a folder in what ever directory I am in, I then must use that mocked target from within that folder. This use case makes sense as a backend tool but not as a user facing tool.

## Solution
When adding a 'mocked` device it should be added globally by default with the option to be folder specific for when needed on the backend. This reflects the transition of this tool from a backend tool to a frontend user facing tool.

PrzemekWirkus, 2016-06-25:
Mock can be global only per OS user. Do you think this would be enough?

BlackstoneEngineering, 2016-06-26:
Yes, that should be good enough, for the typical users (windows, single user laptop/desktop machine) this should net the same effect.

PrzemekWirkus, 2016-07-05:
Done via #106

BlackstoneEngineering, 2016-07-05:
cool, closing issue

closed 2016-07-05 by BlackstoneEngineering