ARMmbed/mbed-ls #119: Add support for public endpoint with boards, procuct codes etc.

PrzemekWirkus, 2016-08-09:
There is now a public https://www.mbed.com/api/v1/boards/ endpoint. It exposes all public boards. It contains information about product code, features, targets (classic, mbed os 5, yotta) and more. There is also similar endpoint on developer.mbed.org site but it requires token to get this information.

See `IOTWEB-1105` for more details.

PrzemekWirkus, 2016-08-09:
@mazimkhan @bridadan @adbridge @ConorPKeegan @screamerbg @sg- FYI

bridadan, 2016-08-09:
Ohhhhhh boy! This is gonna be sweet! :D

sg-, 2016-08-11:
Wowwwwwzzzzers 👍

jupe, 2016-08-11:
cool :+1: . is there also plain json API?

bridadan, 2016-08-11:
@jupe : https://www.mbed.com/api/v1/boards/?format=json

:)

bridadan, 2017-10-17:
FYI the api has moved to https://os.mbed.com/api/v3/platforms/

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

mazimkhan, 2018-10-02:
Seems like this is implemented by #341. But I see there was a revert as well.