ARMmbed/mbed-ls #128: mbed-ls does not work on Ubuntu 14.04 anymore

teetak01, 2016-09-09:
Updated mbed-flasher to 1.2.1, now getting non-functional mbed-ls and failures like this:

Traceback (most recent call last):
  File "/home/mbed-client/mbed-clitest/mbed_clitest/bench.py", line 718, in run
    self.__setUpBench()
  File "/home/mbed-client/mbed-clitest/mbed_clitest/bench.py", line 891, in __setUpBench
    self.__initDuts()
  File "/home/mbed-client/mbed-clitest/mbed_clitest/bench.py", line 542, in __initDuts
    self.duts = self.resource_provider.initialize_duts(self.DEFAULT_BIN)
  File "/home/mbed-client/mbed-clitest/mbed_clitest/ResourceProvider/ResourceProvider.py", line 68, in initialize_duts
    self.allocator = self.__get_allocator()
  File "/home/mbed-client/mbed-clitest/mbed_clitest/ResourceProvider/ResourceProvider.py", line 128, in __get_allocator
    return LocalAllocator(logger = self.logger)
  File "/home/mbed-client/mbed-clitest/mbed_clitest/ResourceProvider/Allocators/LocalAllocator.py", line 16, in __init__
    self._available_devices = DutDetection().get_available_devices()
  File "/home/mbed-client/mbed-clitest/mbed_clitest/DeviceConnectors/DutDetection.py", line 11, in __init__
    self.mbeds = create()
  File "/usr/local/lib/python2.7/dist-packages/mbed_lstools/main.py", line 43, in create
    elif mbed_os == 'Ubuntu': result = MbedLsToolsUbuntu()
  File "/usr/local/lib/python2.7/dist-packages/mbed_lstools/lstools_ubuntu.py", line 27, in __init__
    MbedLsToolsLinuxGeneric.__init__(self)
  File "/usr/local/lib/python2.7/dist-packages/mbed_lstools/lstools_linux_generic.py", line 29, in __init__
    MbedLsToolsBase.__init__(self)
  File "/usr/local/lib/python2.7/dist-packages/mbed_lstools/lstools_base.py", line 41, in __init__
    self.mbedls_get_mocks()
  File "/usr/local/lib/python2.7/dist-packages/mbed_lstools/lstools_base.py", line 225, in mbedls_get_mocks
    mock_ids = self.mock_read()
  File "/usr/local/lib/python2.7/dist-packages/mbed_lstools/lstools_base.py", line 276, in mock_read
    lock = self.mbedls_get_global_lock()
  File "/usr/local/lib/python2.7/dist-packages/mbed_lstools/lstools_base.py", line 237, in mbedls_get_global_lock
    lock = lockfile.LockFile(file_path)
AttributeError: 'module' object has no attribute 'LockFile'

Downgrading back to 0.2.17 works fine.

bridadan, 2016-09-09:
Hi @teetak01,

Could please run `pip list` and paste the log please?

You may also try running `pip install -U lockfile` to see if the most recent version of the pip package solves your issue.

teetak01, 2016-09-12:
@bridadan here's the pip list. lockfile is already up to date. Installed now the newest mbed-ls 1.2.2, same issue still.

```
adium-theme-ubuntu (0.3.4)
apt-xapian-index (0.45)
argcomplete (1.1.1)
certifi (2016.2.28)
cffi (1.6.0)
chardet (2.0.1)
CoAPthon (2.0.0)
colorama (0.3.7)
coloredlogs (5.0)
command-not-found (0.3)
cryptography (1.3.2)
debtagshw (0.1)
defer (1.0.6)
dirspec (13.10)
docutils (0.12)
duplicity (0.6.23)
enum34 (1.1.6)
functools32 (3.2.3.post2)
futures (2.1.6)
hgapi (1.7.2)
httplib2 (0.8)
humanfriendly (1.44.7)
idna (2.1)
intelhex (2.1)
ipaddress (1.0.16)
Jinja2 (2.8)
jsonmerge (1.1.0)
jsonpointer (1.10)
jsonschema (2.5.1)
junit-xml (1.7)
lockfile (0.12.2)
Logbook (0.12.5)
lxml (3.3.3)
MarkupSafe (0.23)
mbed-cli (0.7.3)
mbed-clitest (0.7.0)
mbed-flasher (0.3.1)
mbed-ls (1.2.2)
mbed-test-wrapper (1.0.0)
mercurial (2.8.2)
ndg-httpsclient (0.4.0)
oauthlib (0.6.1)
oneconf (0.3.7.14.4.1)
PAM (0.4.2)
pathlib (1.0.1)
pexpect (3.1)
Pillow (2.3.0)
pip (8.1.2)
piston-mini-client (0.7.5)
prettytable (0.7.2)
project-generator (0.8.16)
project-generator-definitions (0.2.21)
psutil (4.1.0)
py (1.4.31)
pyasn1 (0.1.9)
pycparser (2.14)
pycrypto (2.6.1)
pycups (1.9.66)
pyelftools (0.23)
PyGithub (1.26.0)
Pygments (2.1.3)
pygobject (3.12.0)
PyJWT (1.4.0)
pyOCD (0.7.1)
pyOpenSSL (16.0.0)
pyserial (3.1.1)
pyshark (0.3.6.1)
pysmbc (1.0.14.1)
python-apt (0.9.3.5ubuntu2)
python-debian (0.1.21-nmu2ubuntu2)
pyusb (1.0.0)
pyxdg (0.25)
PyYAML (3.11)
reportlab (3.0)
requests (2.10.0)
semantic-version (2.5.0)
semver (2.6.0)
sessioninstaller (0.0.0)
setuptools (21.1.0)
six (1.10.0)
software-center-aptd-plugins (0.0.0)
Sphinx (1.2.2)
system-service (0.1.6)
trollius (1.0.4)
Twisted (16.2.0)
Twisted-Core (13.2.0)
Twisted-Web (13.2.0)
unity-lens-photos (1.0)
urllib3 (1.7.1)
valinor (0.0.15)
xdiagnose (3.6.3build2)
xmltodict (0.10.2)
yattag (1.5.2)
yotta (0.16.4)
zope.interface (4.0.5)
```

bridadan, 2016-09-12:
I tried replicating your error on a clean Ubuntu 14.04 server but everything was working for me. Here were the steps that I followed:

1. Start with the following global pip packages:
    ```
    adium-theme-ubuntu (0.3.4)
    apt-xapian-index (0.45)
    argparse (1.2.1)
    chardet (2.0.1)
    colorama (0.2.5)
    command-not-found (0.3)
    debtagshw (0.1)
    defer (1.0.6)
    dirspec (13.10)
    duplicity (0.6.23)
    html5lib (0.999)
    httplib2 (0.8)
    lockfile (0.8)
    lxml (3.3.3)
    oauthlib (0.6.1)
    oneconf (0.3.7.14.04.1)
    PAM (0.4.2)
    pexpect (3.1)
    Pillow (2.3.0)
    pip (1.5.4)
    piston-mini-client (0.7.5)
    pycrypto (2.6.1)
    pycups (1.9.66)
    pygobject (3.12.0)
    pyOpenSSL (0.13)
    pyserial (2.6)
    pysmbc (1.0.14.1)
    python-apt (0.9.3.5ubuntu2)
    python-debian (0.1.21-nmu2ubuntu2)
    pyxdg (0.25)
    reportlab (3.0)
    requests (2.2.1)
    sessioninstaller (0.0.0)
    setuptools (3.3)
    six (1.5.2)
    software-center-aptd-plugins (0.0.0)
    ssh-import-id (3.21)
    system-service (0.1.6)
    Twisted-Core (13.2.0)
    Twisted-Web (13.2.0)
    unity-lens-photos (1.0)
    urllib3 (1.7.1)
    vboxapi (1.0)
    virtualenv (15.0.3)
    wheel (0.24.0)
    wsgiref (0.1.2)
    xdiagnose (3.6.3build2)
    zope.interface (4.0.5)
    ```

2. Create a virtual environment: `virtualenv mbedlsenv`
3. Activate the virtual env: `. mbedlsenv/bin/activate`
4. Install mbed-ls `pip install mbed-ls`
5. pip packages after `pip install mbed-ls`:
    ```
    lockfile (0.12.2)
    mbed-ls (1.2.2)
    pip (8.1.2)
    prettytable (0.7.2)
    setuptools (27.1.2)
    wheel (0.29.0)
    ```

6. Plug in NUCLEO board into machine and mount it: `sudo mount /dev/sdd /media/NUCLEO`
7. Run `mbedls`:

    ```
    $ mbedls
    +---------------+----------------------+---------------+--------------+--------------------------+-----------------+
    | platform_name | platform_name_unique | mount_point   | serial_port  | target_id                | daplink_version |
    +---------------+----------------------+---------------+--------------+--------------------------+-----------------+
    | NUCLEO_F411RE | NUCLEO_F411RE[0]     | /media/NUCLEO | /dev/ttyACM0 | 074002210959692A361DFF57 | 0221            |
    +---------------+----------------------+---------------+--------------+--------------------------+-----------------+
    ```

Can you reproduce the failure when running those steps? I did notice you don't have the `wheel` package installed, maybe that's causing an issue? Are you seeing any errors when installing `mbed-ls`?

teetak01, 2016-09-13:
At least wheel or setuptools did not fix anything. Did not try to go through the whole list.

VeliMattiLahtela, 2016-09-13:
At least in my setup the problem was the python-lockfile that was installed with apt-get during system setup.

```
sudo apt-get purge python-lockfile
```

Solved my issues.

teetak01, 2016-09-13:
Same for me. Ubuntu components were conflicting with pip installed packages.

bridadan, 2016-09-13:
@teetak01 Glad you were able to solve the issue. I noticed that the `lockfile` package is now deprecated, so it seems like we now may have two good reasons to switch to a different pip package for global locking. I've captured this in the new issue #130.

Feel free to close this issue if you feel like you have an appropriate resolution.

teetak01, 2016-09-14:
The switch sounds like a good idea. Issue is resolved for us so I'll close this.

closed 2016-09-14 by teetak01

RidaJichi, 2016-09-30:
I got the same LockFile error when executing "mbed detect" (mbed-cli)
**Solution:** `pip install mbed-ls --upgrade`
**Details:** On a fresh Ubuntu 14.04, I had installed necessary dependencies and then "pip install mbed-cli" which installs mbed-ls. This is verified by the following:
`> pip install mbed-ls
Requirement already satisfied (use --upgrade to upgrade): mbed-ls in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied (use --upgrade to upgrade): PrettyTable>=0.7.2 in /usr/local/lib/python2.7/dist-packages (from mbed-ls)
Requirement already satisfied (use --upgrade to upgrade): lockfile in /usr/lib/python2.7/dist-packages (from mbed-ls)
Cleaning up...`
So the dependency appears to be satisfied. However, upon upgrade, it appears that lockfile is yet to be installed.
` > pip install mbed-ls --upgrade
Requirement already up-to-date: mbed-ls in /usr/local/lib/python2.7/dist-packages
Requirement already up-to-date: PrettyTable>=0.7.2 in /usr/local/lib/python2.7/dist-packages (from mbed-ls)
Downloading/unpacking lockfile from https://pypi.python.org/packages/c8/22/9460e311f340cb62d26a38c419b1381b8593b0bb6b5d1f056938b086d362/lockfile-0.12.2-py2.py3-none-any.whl#md5=07b04864472c90cdf4452cf25068
7334 (from mbed-ls)
  Downloading lockfile-0.12.2-py2.py3-none-any.whl
Installing collected packages: lockfile
  Found existing installation: lockfile 0.8
    Not uninstalling lockfile at /usr/lib/python2.7/dist-packages, owned by OS
Successfully installed lockfile
Cleaning up...
`