ARMmbed/mbed-ls #130: Move 'lockfile' dependency to 'fasteners' or 'oslo.concurrency'

bridadan, 2016-09-13:
As mentioned at the top of the [lockfile documentation](https://pypi.python.org/pypi/lockfile/0.12.2), the package is now deprecated. It also a conflict with a default Ubuntu pip package as described here: https://github.com/ARMmbed/mbed-ls/issues/128#issuecomment-246587081

We can probably take care of both issues by moving to one of the recommended packages from the lockfile docs: [fasteners](https://pypi.python.org/pypi/fasteners) or [oslo.concurrency](http://docs.openstack.org/developer/oslo.concurrency/).

mazimkhan, 2016-09-15:
@bridadan are you working  on it?

bridadan, 2016-09-15:
@mazimkhan Not currently :smile:

adbridge, 2016-09-19:
ARM Internal Ref: IOTSFW-2924

mazimkhan, 2016-10-11:
Fixed by https://github.com/ARMmbed/mbed-ls/pull/139

closed 2016-10-11 by mazimkhan

bridadan, 2016-10-11:
Thanks @mazimkhan !