ARMmbed/mbed-ls #137: Latest ST-Link fails to list serial port on Windows

bridadan, 2016-10-04:
Using the latest version of ST-Link (version 2.28.16 found here: http://www.st.com/content/st_com/en/products/embedded-software/development-tool-software/stsw-link007.html), my nucleo devices do not have a serial port that is picked up be mbedls (though I can see it in Device Manager)

Using Windows 7 64 bit and mbed-ls version 1.2.5.

**NOTE:** This is not an issue in mbed-ls version 1.2.4.

```
debug @MbedLsToolsWin7.mbed_lstools.main: mbed-ls ver. 1.2.5
debug @MbedLsToolsWin7.mbed_lstools.main: host: ('nt', 'Windows', '7', '6.1.7601', 'win32')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\E:', '_??_USBSTOR#Disk&Ven_MBED&Prod_microcontroller&Rev_1.0#a&19934e84&0&066BFF494951785087022947&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\F:', '_??_USBSTOR#Disk&Ven_MBED&Prod_microcontroller&Rev_1.0#8&4bb3ea4&0&0669FF515456707067164717&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\G:', '_??_USBSTOR#Disk&Ven_MBED&Prod_microcontroller&Rev_1.0#a&1a08f641&0&0670FF494951785087022432&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\H:', '_??_USBSTOR#Disk&Ven_MBED&Prod_microcontroller&Rev_1.0#a&1d9f9a1d&0&066EFF485649785087174252&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbeds: ('E:', 'microcontroller')
debug @MbedLsToolsWin7.get_mbeds: ('F:', 'microcontroller')
debug @MbedLsToolsWin7.get_mbeds: ('G:', 'microcontroller')
debug @MbedLsToolsWin7.get_mbeds: ('H:', 'microcontroller')
debug @MbedLsToolsWin7.mount_point_ready: Mount point E: is ready
debug @MbedLsToolsWin7.mount_point_ready: Mount point F: is ready
debug @MbedLsToolsWin7.mount_point_ready: Mount point G: is ready
debug @MbedLsToolsWin7.mount_point_ready: Mount point H: is ready
debug @MbedLsToolsWin7.scan_html_line_for_target_id: <meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=072002210726657A3862F367"/>
debug @MbedLsToolsWin7.scan_html_line_for_target_id: (('072002210726657A3862F367',), '072002210726657A3862F367')
debug @MbedLsToolsWin7.get_mbed_com_port: ID: microcontroller
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\E:', '_??_USBSTOR#Disk&Ven_MBED&Prod_microcontroller&Rev_1.0#a&19934e84&0&066BFF494951785087022947&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\F:', '_??_USBSTOR#Disk&Ven_MBED&Prod_microcontroller&Rev_1.0#8&4bb3ea4&0&0669FF515456707067164717&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\G:', '_??_USBSTOR#Disk&Ven_MBED&Prod_microcontroller&Rev_1.0#a&1a08f641&0&0670FF494951785087022432&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\H:', '_??_USBSTOR#Disk&Ven_MBED&Prod_microcontroller&Rev_1.0#a&1d9f9a1d&0&066EFF485649785087174252&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbeds: ('E:', 'microcontroller')
debug @MbedLsToolsWin7.get_mbeds: ('F:', 'microcontroller')
debug @MbedLsToolsWin7.get_mbeds: ('G:', 'microcontroller')
debug @MbedLsToolsWin7.get_mbeds: ('H:', 'microcontroller')
debug @MbedLsToolsWin7.mount_point_ready: Mount point E: is ready
debug @MbedLsToolsWin7.mount_point_ready: Mount point F: is ready
debug @MbedLsToolsWin7.mount_point_ready: Mount point G: is ready
debug @MbedLsToolsWin7.mount_point_ready: Mount point H: is ready
debug @MbedLsToolsWin7.scan_html_line_for_target_id: <meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=072002210726657A3862F367"/>
debug @MbedLsToolsWin7.scan_html_line_for_target_id: (('072002210726657A3862F367',), '072002210726657A3862F367')
debug @MbedLsToolsWin7.scan_html_line_for_target_id: <meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=074002210959692A361DFF57"/>
debug @MbedLsToolsWin7.scan_html_line_for_target_id: (('074002210959692A361DFF57',), '074002210959692A361DFF57')
debug @MbedLsToolsWin7.scan_html_line_for_target_id: <meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=07200221076F6570382BF36D"/>
debug @MbedLsToolsWin7.scan_html_line_for_target_id: (('07200221076F6570382BF36D',), '07200221076F6570382BF36D')
debug @MbedLsToolsWin7.scan_html_line_for_target_id: <meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=07400221075C610D3818F770"/>
debug @MbedLsToolsWin7.scan_html_line_for_target_id: (('07400221075C610D3818F770',), '07400221075C610D3818F770')
debug @MbedLsToolsWin7.scan_html_line_for_target_id: <meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=074002210959692A361DFF57"/>
debug @MbedLsToolsWin7.scan_html_line_for_target_id: (('074002210959692A361DFF57',), '074002210959692A361DFF57')
debug @MbedLsToolsWin7.get_mbed_com_port: ID: microcontroller
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\E:', '_??_USBSTOR#Disk&Ven_MBED&Prod_microcontroller&Rev_1.0#a&19934e84&0&066BFF494951785087022947&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\F:', '_??_USBSTOR#Disk&Ven_MBED&Prod_microcontroller&Rev_1.0#8&4bb3ea4&0&0669FF515456707067164717&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\G:', '_??_USBSTOR#Disk&Ven_MBED&Prod_microcontroller&Rev_1.0#a&1a08f641&0&0670FF494951785087022432&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\H:', '_??_USBSTOR#Disk&Ven_MBED&Prod_microcontroller&Rev_1.0#a&1d9f9a1d&0&066EFF485649785087174252&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbeds: ('E:', 'microcontroller')
debug @MbedLsToolsWin7.get_mbeds: ('F:', 'microcontroller')
debug @MbedLsToolsWin7.get_mbeds: ('G:', 'microcontroller')
debug @MbedLsToolsWin7.get_mbeds: ('H:', 'microcontroller')
debug @MbedLsToolsWin7.mount_point_ready: Mount point E: is ready
debug @MbedLsToolsWin7.mount_point_ready: Mount point F: is ready
debug @MbedLsToolsWin7.mount_point_ready: Mount point G: is ready
debug @MbedLsToolsWin7.mount_point_ready: Mount point H: is ready
debug @MbedLsToolsWin7.scan_html_line_for_target_id: <meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=072002210726657A3862F367"/>
debug @MbedLsToolsWin7.scan_html_line_for_target_id: (('072002210726657A3862F367',), '072002210726657A3862F367')
debug @MbedLsToolsWin7.scan_html_line_for_target_id: <meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=074002210959692A361DFF57"/>
debug @MbedLsToolsWin7.scan_html_line_for_target_id: (('074002210959692A361DFF57',), '074002210959692A361DFF57')
debug @MbedLsToolsWin7.scan_html_line_for_target_id: <meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=07200221076F6570382BF36D"/>
debug @MbedLsToolsWin7.scan_html_line_for_target_id: (('07200221076F6570382BF36D',), '07200221076F6570382BF36D')
debug @MbedLsToolsWin7.scan_html_line_for_target_id: <meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=07400221075C610D3818F770"/>
debug @MbedLsToolsWin7.scan_html_line_for_target_id: (('07400221075C610D3818F770',), '07400221075C610D3818F770')
debug @MbedLsToolsWin7.scan_html_line_for_target_id: <meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=07200221076F6570382BF36D"/>
debug @MbedLsToolsWin7.scan_html_line_for_target_id: (('07200221076F6570382BF36D',), '07200221076F6570382BF36D')
debug @MbedLsToolsWin7.get_mbed_com_port: ID: microcontroller
[... 32 lines cut ...]
debug @MbedLsToolsWin7.mount_point_ready: Mount point F: is ready
debug @MbedLsToolsWin7.mount_point_ready: Mount point G: is ready
debug @MbedLsToolsWin7.mount_point_ready: Mount point H: is ready
debug @MbedLsToolsWin7.scan_html_line_for_target_id: <meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=072002210726657A3862F367"/>
debug @MbedLsToolsWin7.scan_html_line_for_target_id: (('072002210726657A3862F367',), '072002210726657A3862F367')
debug @MbedLsToolsWin7.scan_html_line_for_target_id: <meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=074002210959692A361DFF57"/>
debug @MbedLsToolsWin7.scan_html_line_for_target_id: (('074002210959692A361DFF57',), '074002210959692A361DFF57')
debug @MbedLsToolsWin7.scan_html_line_for_target_id: <meta http-equiv="refresh" content="0; url=http://mbed.org/device/?code=07200221076F6570382BF36D"/>
[... cut ...]
| NUCLEO_F401RE | NUCLEO_F401RE[1]     | G:          | unknown     | 07200221076F6570382BF36D | 0221            |
| NUCLEO_F411RE | NUCLEO_F411RE[0]     | F:          | unknown     | 074002210959692A361DFF57 | 0221            |
| NUCLEO_F411RE | NUCLEO_F411RE[1]     | H:          | unknown     | 07400221075C610D3818F770 | 0221            |
+---------------+----------------------+-------------+-------------+--------------------------+-----------------+
debug @MbedLsToolsWin7.mbed_lstools.main: Return code: -1
```

jeromecoutant, 2016-10-05:
Hi
I don't think issue is related to ST-Link version....
I didn't update my boards, and new mbed-ls version doesn't work as well :-(

bridadan, 2016-10-05:
@jeromecoutant It's definitely an issue with mbed-ls, I have a patch up here: https://github.com/ARMmbed/mbed-ls/pull/138

bridadan, 2016-10-05:
Closing because #138 was merged

closed 2016-10-05 by bridadan