ARMmbed/mbed-ls #141: master doesn't work on windows 7

jupe, 2016-10-17:
Commit 9a3121e003e5a436f48f6b62a17373d3deb148f1

```
c:\github\ARMmbed\mbed-ls>python setup.py install
...
c:\github\ARMmbed\mbed-ls>mbedls
Traceback (most recent call last):
  File "C:\Python27\Scripts\mbedls-script.py", line 9, in <module>
    load_entry_point('mbed-ls==1.2.8', 'console_scripts', 'mbedls')()
  File "build\bdist.win32\egg\mbed_lstools\main.py", line 217, in mbedls_main
  File "build\bdist.win32\egg\mbed_lstools\lstools_base.py", line 560, in get_string
  File "build\bdist.win32\egg\mbed_lstools\lstools_base.py", line 425, in list_mbeds_ext
  File "build\bdist.win32\egg\mbed_lstools\lstools_win7.py", line 48, in list_mbeds
  File "build\bdist.win32\egg\mbed_lstools\lstools_win7.py", line 69, in discover_connected_mbeds
  File "build\bdist.win32\egg\mbed_lstools\lstools_win7.py", line 141, in get_connected_mbeds
  File "build\bdist.win32\egg\mbed_lstools\lstools_win7.py", line 164, in get_mbeds
AttributeError: 'NoneType' object has no attribute 'group'

c:\github\ARMmbed\mbed-ls>git rev-parse HEAD
9a3121e003e5a436f48f6b62a17373d3deb148f1
```

jupe, 2016-10-17:
seems that this patch fix my issue:
```
diff --git a/mbed_lstools/lstools_win7.py b/mbed_lstools/lstools_win7.py
index 675f8ec..ae5fd45 100644
--- a/mbed_lstools/lstools_win7.py
+++ b/mbed_lstools/lstools_win7.py
@@ -161,7 +161,9 @@ class MbedLsToolsWin7(MbedLsToolsBase):
         for mbed in self.get_mbed_devices():
             mountpoint = re.match('.*\\\\(.:)$', mbed[0]).group(1)
             # TargetID is a hex string with 10-48 chars
-            tid = re.search('[&#]([0-9A-Za-z]{10,48})[&#]', mbed[1]).group(1)
+            m = re.search('[&#]([0-9A-Za-z]{10,48})[&#]', mbed[1])
+            if not m: continue
+            tid = m.group(1)
```

bridadan, 2016-10-17:
@jupe Could you submit this as a PR? Good catch.

closed 2016-11-01 by adbridge