ARMmbed/mbed-ls #145: setup install return an error 【UnicodeDecodeError: 'gbk' codec can't decode byte 0x9d】

zhangshengping, 2016-11-05:
I am trying to install mbed-ls with source code, but it return an error.  the log is:`
Traceback (most recent call last):
  File "setup.py", line 41, in <module>
    long_description=read('README.md'),
  File "setup.py", line 36, in read
    return open(os.path.join(os.path.dirname(__file__), fname)).read()
UnicodeDecodeError: 'gbk' codec can't decode byte 0x9d in position 13877: illegal multibyte sequence`

What should I do to solve the error.

PrzemekWirkus, 2016-11-08:
You are using Python 3 to install `mbed-ls` and `mbed-ls` is Python 2.7.x compatible.
Please use Python 2.7 instead.

PrzemekWirkus, 2016-11-08:
BTW, same as https://github.com/ARMmbed/mbed-ls/issues/69

mazimkhan, 2016-12-12:
Does this PR fixes this issue? Although still Python 2.7 is the officially supported version.

PrzemekWirkus, 2016-12-20:
There was no PR for Py3 support for mbed-ls  AFAIK @mazimkhan.

mazimkhan, 2016-12-21:
Sorry failed to copy the PR URL https://github.com/ARMmbed/mbed-ls/pull/151

closed 2016-12-22 by zhangshengping

adbridge, 2017-01-09:
@zhangshengping Also could you please not embedd unicode characters within the Issue titles ? Thanks