ARMmbed/mbed-ls #149: Same device listed twice

RomanSaveljev, 2016-12-05:
This reproduces on Raspberry Pi in one of our CI shoot-out jobs (same test set is run continuously to pinpoint the flaky parts). Eventually, the system enters some kind of messed state and `mbedls` begins to show the same target board twice:

```
pi@mikanpi:~/usbreset $ mbedls --debug
debug @MbedLsToolsLinuxGeneric.mbed_lstools.main: mbed-ls ver. 1.2.9
debug @MbedLsToolsLinuxGeneric.mbed_lstools.main: host: ('posix', 'Linux', '4.1.19-v7+', '#858 SMP Tue Mar 15 15:56:00 GMT 2016', 'linux2')
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 13 Dec  5 13:24 mmc-USDU1_0x4b2f54c1 -> ../../mmcblk0
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 15 Dec  5 13:24 mmc-USDU1_0x4b2f54c1-part1 -> ../../mmcblk0p1
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 15 Dec  5 13:24 mmc-USDU1_0x4b2f54c1-part2 -> ../../mmcblk0p2
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root  9 Dec  5 21:07 usb-MBED_VFS_0240000028884e450036700f6bf000118021000097969900-0:0 -> ../../sda
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root  9 Dec  5 22:50 usb-MBED_VFS_0240000029164e45001b0012706e000df301000097969900-0:0 -> ../../sda
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 13 Dec  5 22:50 usb-ARM_DAPLink_CMSIS-DAP_0240000029164e45001b0012706e000df301000097969900-if01 -> ../../ttyACM0
debug @MbedLsToolsLinuxGeneric.get_mounts: mount | grep vfat
debug @MbedLsToolsLinuxGeneric.get_mounts: /dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
debug @MbedLsToolsLinuxGeneric.get_mounts: /dev/sda on /mnt/platform-3f980000_usb-usb-0_1_2_2_1_0-scsi-0_0_0_0 type vfat (rw,relatime,uid=1000,gid=1000,fmask=0133,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,flush,errors=remount-ro,uhelper=ldm)
debug @MbedLsToolsLinuxGeneric.scan_html_line_for_target_id: window.location.replace("https://mbed.org/device/?code=0240000029164e45001b0012706e000df301000097969900?version=0242?target_id=0031ffffffffffff4e45315450090021");
debug @MbedLsToolsLinuxGeneric.scan_html_line_for_target_id: (('0240000029164e45001b0012706e000df301000097969900',), '0240000029164e45001b0012706e000df301000097969900')
debug @MbedLsToolsLinuxGeneric.scan_html_line_for_target_id: window.location.replace("https://mbed.org/device/?code=0240000029164e45001b0012706e000df301000097969900?version=0242?target_id=0031ffffffffffff4e45315450090021");
debug @MbedLsToolsLinuxGeneric.scan_html_line_for_target_id: (('0240000029164e45001b0012706e000df301000097969900',), '0240000029164e45001b0012706e000df301000097969900')
debug @MbedLsToolsLinuxGeneric.list_mbeds_ext: ('K64F[0]', '0240000029164e45001b0012706e000df301000097969900')
debug @MbedLsToolsLinuxGeneric.list_mbeds_ext: ('K64F[1]', '0240000029164e45001b0012706e000df301000097969900')
+---------------+----------------------+---------------------------------------------------------+--------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point                                             | serial_port  | target_id                                        | daplink_version |
+---------------+----------------------+---------------------------------------------------------+--------------+--------------------------------------------------+-----------------+
| K64F          | K64F[0]              | /mnt/platform-3f980000_usb-usb-0_1_2_2_1_0-scsi-0_0_0_0 | unknown      | 0240000029164e45001b0012706e000df301000097969900 | 0242            |
| K64F          | K64F[1]              | /mnt/platform-3f980000_usb-usb-0_1_2_2_1_0-scsi-0_0_0_0 | /dev/ttyACM0 | 0240000029164e45001b0012706e000df301000097969900 | 0242            |
+---------------+----------------------+---------------------------------------------------------+--------------+--------------------------------------------------+-----------------+
debug @MbedLsToolsLinuxGeneric.mbed_lstools.main: Return code: 0
```

There is two physical boards connected, but one of them seemed to have entered bad state.

```
pi@mikanpi:~/usbreset $ lsusb
Bus 001 Device 007: ID 04d8:f2f7 Microchip Technology, Inc. 
Bus 001 Device 039: ID 0d28:0204 NXP LPC1768
Bus 001 Device 004: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Below is a relevant part of kernel log, where we can see the second board beginning to misbehave:

```
[27680.732428] usb 1-1.2.1: SerialNumber: 0240000028884e450036700f6bf000118021000097969900
[27687.817600] usb 1-1.2.1: USB disconnect, device number 74
[27694.708649] usb 1-1.2.1: new full-speed USB device number 75 using dwc_otg
[27694.812336] usb 1-1.2.1: New USB device found, idVendor=0d28, idProduct=0204
[27694.812348] usb 1-1.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[27694.812355] usb 1-1.2.1: Product: DAPLink CMSIS-DAP
[27694.812362] usb 1-1.2.1: Manufacturer: ARM
[27694.812368] usb 1-1.2.1: SerialNumber: 0240000028884e450036700f6bf000118021000097969900
[27728.698762] usb 1-1.2.1: reset full-speed USB device number 75 using dwc_otg
[27735.068820] usb 1-1.2.1: reset full-speed USB device number 75 using dwc_otg
[27735.148796] usb 1-1.2.1: device descriptor read/64, error -71
[27735.338790] usb 1-1.2.1: device descriptor read/64, error -71
[27735.528827] usb 1-1.2.1: reset full-speed USB device number 75 using dwc_otg
[27735.608764] usb 1-1.2.1: device descriptor read/64, error -71
[27735.798760] usb 1-1.2.1: device descriptor read/64, error -71
[27735.988799] usb 1-1.2.1: reset full-speed USB device number 75 using dwc_otg
[27736.408763] usb 1-1.2.1: device not accepting address 75, error -71
[27736.488764] usb 1-1.2.1: reset full-speed USB device number 75 using dwc_otg
[27736.908763] usb 1-1.2.1: device not accepting address 75, error -71
[27736.909122] usb 1-1.2.1: USB disconnect, device number 75
[27737.008832] usb 1-1.2.1: new full-speed USB device number 76 using dwc_otg
[27737.088781] usb 1-1.2.1: device descriptor read/64, error -71
[27737.278770] usb 1-1.2.1: device descriptor read/64, error -71
[27737.468757] usb 1-1.2.1: new full-speed USB device number 77 using dwc_otg
[27737.548763] usb 1-1.2.1: device descriptor read/64, error -71
[27737.738764] usb 1-1.2.1: device descriptor read/64, error -71
```

I think the messed up board should not show up

mazimkhan, 2016-12-08:
it seems like because of a messed up board it is showing working board twice. I will have a look in the code.

bridadan, 2016-12-08:
Here's the funky bit in the debug log that I think is the root of the problem:

```
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root  9 Dec  5 21:07 usb-MBED_VFS_0240000028884e450036700f6bf000118021000097969900-0:0 -> ../../sda
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root  9 Dec  5 22:50 usb-MBED_VFS_0240000029164e45001b0012706e000df301000097969900-0:0 -> ../../sda
```

Both mbed MSD endpoints are showing up as `/dev/sda`!

This seems like a problem with the board and OS interaction, like you described @RomanSaveljev.

Does the board with target ID `0240000029164e45001b0012706e000df301000097969900` still work correctly? If so then I agree that the board entry with the `unknown` serial port should be removed from the listing. Otherwise it might make sense for mbed-ls to error if detects this bad device configuration.

RomanSaveljev, 2016-12-09:
I will have to check it out, when the situation re-appears again

bridadan, 2016-12-09:
@RomanSaveljev That'd be great. The debug log is really helpful, we could probably even patch it just from that. But it'd be good to know how the boards behave when they're in this state. Does this occur often on your setup?

RomanSaveljev, 2016-12-09:
Every morning I find the setup in a messed up state (after the nightly blast out). I will keep you posted, when precisely this issue manifests itself

mazimkhan, 2016-12-12:
Analysed this issue. The behaviour is because of mbed-ls looping over found disks to find mbed devices. Since in the logs there are two disk devices mbed-ls knows two devices. Since they both point to same device ```/dev/sda``` mbed-ls finds same target Id and other params. Related code is here https://github.com/ARMmbed/mbed-ls/blob/master/mbed_lstools/lstools_linux_generic.py#L220

This issues complains that faulty duplicate device shouldn't be shown. I think this is happening because of faulty kernel. mbed-ls has fair assumption that kernel will not report two devices with same device node. Even if we put validation for this situation. We don't know what can come next. For example if there was another serial port present. But there were two devices with same disk node. Which one will you select. In my opinion we should try to fix the issue outside mbed-ls. @bridadan what do you say?

@RomanSaveljev @jupe For a workaround for time being and keeping the assumption that faulty device is the one with no serial port. Can you not put some checks where you invoke mbed-ls to discard device with "unknown" serial port. In parallel try fixing the kernel issue. Or not getting into the issue.

bridadan, 2016-12-12:
I've been trying to think of a way that both reports the error and lists only "known good" mbeds, but I haven't come up with anything yet. We probably don't want to require a board to have a serial port to be listed because it can be useful when debugging problems with serial drives and stuff (ST-Link for instance). I think I would tend to agree with @mazimkhan here and recommend that the thing using mbedls should probably check the device info before using them.

I'd like to hear @RomanSaveljev's opinion on that though, since there very well could be a case I haven't considered.

jupe, 2016-12-12:
I think by default mbed-ls should list just "known good" mbeds because otherwise we might be ended situations like this which just make confusion for end user. There could be option to list also "bad mbeds" so end user have to understand possible consequences. But because every mbed has unique id maybe mbedls shouldn't list any id twice even there is some conflict (especially when listing "known good" mbeds). Maybe there could be some other property when conflict case.

RomanSaveljev, 2016-12-12:
We assume `mbedls` returns the listing of usable devices, i.e. it is better if `mbedls` skips the boards that are not in a known good shape.

P.S. The problem has not yet re-appeared

bridadan, 2016-12-12:
> We assume mbedls returns the listing of usable devices, i.e. it is better if mbedls skips the boards that are not in a known good shape.

@mazimkhan seems like a pretty fair assumption, though we have to be pretty explicit on what we consider as "known good". Would you say a device that has a mount point and a known serial port is "good"? Would there ever be a case where a device that didn't have a serial port would still be considered "good"?

RomanSaveljev, 2016-12-15:
It looks like `mbedls` will exit with `-1`, if e.g. serial port can not be determined (just saw this situation today in Windows, where STM drivers were not installed). This makes all its output useless in automation, because the process has finished unsuccessfully and we can not trust the output of `mbedls -j`

mazimkhan, 2017-02-15:
Keep aside the arguments for unknown good devices. The issue reported here is because of error in the OS. Are trying to work around the OS issue by selecting a known good device?
Kernel issue is out of scope of mbed-ls.
We can have command line option to filter out boards that are missing interfaces. But that should be raised as a different issue.
@jupe @RomanSaveljev @bridadan