ARMmbed/mbed-ls #152: FRDM-KW24D512 listed as unknown device

VeliMattiLahtela, 2016-12-27:
FRDM-KW24D512 boards are not recognized on Windows or Linux systems correctly.

`debug @MbedLsToolsLinuxGeneric.mbed_lstools.main: mbed-ls ver. 1.2.11
debug @MbedLsToolsLinuxGeneric.mbed_lstools.main: host: ('posix', 'Linux', '4.4.34-v7+', '#930 SMP Wed Nov 23 15:20:41 GMT 2016', 'linux2')
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 13 Dec  5 10:14 mmc-USDU1_0x41046834 -> ../../mmcblk0
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 15 Dec  5 10:14 mmc-USDU1_0x41046834-part1 -> ../../mmcblk0p1
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 15 Dec  5 10:14 mmc-USDU1_0x41046834-part2 -> ../../mmcblk0p2
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root  9 Dec 27 08:39 usb-MBED_microcontroller_03200203C80E6E5935F29281-0:0 -> ../../sda
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 13 Dec 27 08:39 usb-MBED_MBED_CMSIS-DAP_03200203C80E6E5935F29281-if01 -> ../../ttyACM0
debug @MbedLsToolsLinuxGeneric.get_mounts: mount | grep vfat
debug @MbedLsToolsLinuxGeneric.get_mounts: /dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
debug @MbedLsToolsLinuxGeneric.get_mounts: /dev/sda on /mnt/platform-3f980000_usb-usb-0_1_3_1_0-scsi-0_0_0_0 type vfat (rw,relatime,uid=1000,gid=1000,fmask=0133,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,flush,errors=remount-ro,uhelper=ldm)
debug @MbedLsToolsLinuxGeneric.list_mbeds_ext: ('None[0]', u'03200203C80E6E5935F29281')
+---------------+----------------------+-------------------------------------------------------+--------------+--------------------------+-----------------+
| platform_name | platform_name_unique | mount_point                                           | serial_port  | target_id                | daplink_version |
+---------------+----------------------+-------------------------------------------------------+--------------+--------------------------+-----------------+
| unknown       | None[0]              | /mnt/platform-3f980000_usb-usb-0_1_3_1_0-scsi-0_0_0_0 | /dev/ttyACM0 | 03200203C80E6E5935F29281 | unknown         |
+---------------+----------------------+-------------------------------------------------------+--------------+--------------------------+-----------------+
debug @MbedLsToolsLinuxGeneric.mbed_lstools.main: Return code: 0
`

`lsusb
Bus 001 Device 005: ID 0d28:0204 NXP LPC1768
`

`usb 1-1.3: new full-speed USB device number 5 using dwc_otg
usb 1-1.3: New USB device found, idVendor=0d28, idProduct=0204
usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-1.3: Product: MBED CMSIS-DAP
usb 1-1.3: Manufacturer: MBED
usb 1-1.3: SerialNumber: 03200203C80E6E5935F29281
usb-storage 1-1.3:1.0: USB Mass Storage device detected
scsi host1: usb-storage 1-1.3:1.0
cdc_acm 1-1.3:1.1: ttyACM0: USB ACM device
hid-generic 0003:0D28:0204.0002: hiddev0,hidraw0: USB HID v1.00 Device [MBED MBED CMSIS-DAP] on usb-3f980000.usb-1.3/input3
scsi 1:0:0:0: Direct-Access     MBED     microcontroller  1.0  PQ: 0 ANSI: 2
sd 1:0:0:0: Attached scsi generic sg0 type 0
sd 1:0:0:0: [sda] 1072 512-byte logical blocks: (549 kB/536 KiB)
sd 1:0:0:0: [sda] Write Protect is off
sd 1:0:0:0: [sda] Mode Sense: 03 00 00 00
sd 1:0:0:0: [sda] No Caching mode page found
sd 1:0:0:0: [sda] Assuming drive cache: write through`

bridadan, 2016-12-27:
@VeliMattiLahtela I'm not sure what firmware the KW24 comes with by default, but have you tried updating it to the latest release of DAPLink? As of this post, that release is number 0242 (https://github.com/mbedmicro/DAPLink/releases/tag/0242)

VeliMattiLahtela, 2016-12-28:
@bridadan yes, already tried that, still the same.

Could the problem be the product id 0204 as [250](https://github.com/ARMmbed/mbed-ls/blob/master/mbed_lstools/lstools_base.py#L106) is linked to KW24D

bridadan, 2016-12-29:
@VeliMattiLahtela, its definitely a problem of the target id not matching up to what mbed-ls expects. Though from the debug log you posted above, the target id appears to be `03200203C80E6E5935F29281`. So in that case, it'd try to match it with the id `0320`, which is currently not a valid target.

@maclobdell are you familiar with this platform?

VeliMattiLahtela, 2016-12-30:
With newest firmware:

dmesg:

`usb 1-1.4: new full-speed USB device number 7 using dwc_otg
usb 1-1.4: New USB device found, idVendor=0d28, idProduct=0204
usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-1.4: Product: DAPLink CMSIS-DAP
usb 1-1.4: Manufacturer: ARM
usb 1-1.4: SerialNumber: 0280000030514e45001f20067d7e00051f91000097969900
usb-storage 1-1.4:1.0: USB Mass Storage device detected
scsi host2: usb-storage 1-1.4:1.0
hid-generic 0003:0D28:0204.0003: hiddev0,hidraw0: USB HID v1.00 Device [ARM DAPLink CMSIS-DAP] on usb-3f980000.usb-1.4/input3
cdc_acm 1-1.4:1.1: ttyACM0: USB ACM device
scsi 2:0:0:0: Direct-Access     MBED     VFS              0.1  PQ: 0 ANSI: 2
sd 2:0:0:0: Attached scsi generic sg0 type 0
sd 2:0:0:0: [sdb] 131200 512-byte logical blocks: (67.2 MB/64.1 MiB)
sd 2:0:0:0: [sdb] Write Protect is off
sd 2:0:0:0: [sdb] Mode Sense: 03 00 00 00
sd 2:0:0:0: [sdb] No Caching mode page found
sd 2:0:0:0: [sdb] Assuming drive cache: write through
sd 2:0:0:0: [sdb] Attached SCSI removable disk
FAT-fs (sda): unable to read boot sector to mark fs as dirty`

mbedls:

`debug @MbedLsToolsLinuxGeneric.mbed_lstools.main: mbed-ls ver. 1.2.11
debug @MbedLsToolsLinuxGeneric.mbed_lstools.main: host: ('posix', 'Linux', '4.4.34-v7+', '#930 SMP Wed Nov 23 15:20:41 GMT 2016', 'linux2')
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 13 Dec 27 09:44 mmc-USDU1_0x41046834 -> ../../mmcblk0
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 15 Dec 27 09:44 mmc-USDU1_0x41046834-part1 -> ../../mmcblk0p1
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 15 Dec 27 09:44 mmc-USDU1_0x41046834-part2 -> ../../mmcblk0p2
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root  9 Dec 30 07:52 usb-MBED_microcontroller_03200203C80E6E5935F29281-0:0 -> ../../sda
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root  9 Dec 30 07:53 usb-MBED_VFS_0280000030514e45001f20067d7e00051f91000097969900-0:0 -> ../../sdb
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 13 Dec 30 07:53 usb-ARM_DAPLink_CMSIS-DAP_0280000030514e45001f20067d7e00051f91000097969900-if01 -> ../../ttyACM0
debug @MbedLsToolsLinuxGeneric.get_mounts: mount | grep vfat
debug @MbedLsToolsLinuxGeneric.get_mounts: /dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
debug @MbedLsToolsLinuxGeneric.get_mounts: /dev/sdb on /mnt/platform-3f980000_usb-usb-0_1_4_1_0-scsi-0_0_0_0 type vfat (rw,relatime,uid=1000,gid=1000,fmask=0133,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,flush,errors=remount-ro,uhelper=ldm)
debug @MbedLsToolsLinuxGeneric.scan_html_line_for_target_id: window.location.replace("https://mbed.org/device/?code=0280000030514e45001f20067d7e00051f91000097969900?version=0242?target_id=79a30000a001001e0017200730244e45");
debug @MbedLsToolsLinuxGeneric.scan_html_line_for_target_id: (('0280000030514e45001f20067d7e00051f91000097969900',), '0280000030514e45001f20067d7e00051f91000097969900')
debug @MbedLsToolsLinuxGeneric.list_mbeds_ext: ('None[0]', u'0280000030514e45001f20067d7e00051f91000097969900')
+---------------+----------------------+-------------------------------------------------------+--------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point                                           | serial_port  | target_id                                        | daplink_version |
+---------------+----------------------+-------------------------------------------------------+--------------+--------------------------------------------------+-----------------+
| unknown       | None[0]              | /mnt/platform-3f980000_usb-usb-0_1_4_1_0-scsi-0_0_0_0 | /dev/ttyACM0 | 0280000030514e45001f20067d7e00051f91000097969900 | 0242            |
+---------------+----------------------+-------------------------------------------------------+--------------+--------------------------------------------------+-----------------+
debug @MbedLsToolsLinuxGeneric.mbed_lstools.main: Return code: 0`

bridadan, 2017-01-03:
@VeliMattiLahtela I'm having trouble reading your debug output, can you fix the line endings please? It looks like your using inline code:
```
`<your log here>`
```

instead of block code:
````
```
<your log here>
```
````

jupe, 2017-01-03:
fixed :)

bridadan, 2017-01-03:
Thanks @jupe :) I can't edit the comments here, otherwise I would have done it myself!

@VeliMattiLahtela I'm going through DAPLink's Git history, I think what you're seeing is that platform was released with an incorrect target prefix, or it was at least changed at one point from 0280 to 0250. This was the commit: https://github.com/mbedmicro/DAPLink/commit/3a21b88264a3624fffcf9b60c987503a53edf703

That commit was part of this PR: https://github.com/mbedmicro/DAPLink/pull/188

Which was merged on November 19th, 2016. The last release was from September of that year, so I'm betting that board was released in 0242 with incorrect ID. @c1728p9 does this sound right? And do you know if DAPLink will be making a new release soon?

@VeliMattiLahtela Is this a critical blocker? We could always update mbed-ls to use the currently released id `0250` until the new DAPLink release is ready. You can also "mock" the target to the correct ID on each machine that needs to use the boards: https://github.com/ARMmbed/mbed-ls#mocking-new-or-existing-target-to-custom-platform-name

jupe, 2017-01-03:
if somebody could compile fixed daplink and share for us i think we could survive with that until actual release is there. But yes this is blocker to use this platform in automation..

SeppoTakalo, 2017-01-30:
How is that you have been able to update the firmware? I did not manage to do that
https://github.com/mbedmicro/DAPLink/issues/225

Where did you find the proper firmware for this board?

mazimkhan, 2017-02-15:
Meanwhile if the ID is incorrect mock feature in mbed-ls can be used.

theotherjimmy, 2017-10-10:
@jupe @bridadan @VeliMattiLahtela Is there a way to progress this forward? Have we settled on a device id for the KW42D? If so, does it match our current database?

bridadan, 2017-10-10:
I'm going through everything again. The mbed database lists the board as 0250 and DAPLink  has been released with 0250 I believe, so it should be resolved in the latest DAPLink release. Can anyone with a KW24D confirm this?

maclobdell, 2017-10-13:
Hi all - NXP just recently updated the Daplink debug interface firmware for FRDM-KW24D512 [here](https://www.nxp.com/products/microcontrollers-and-processors/arm-based-processors-and-mcus/kinetis-cortex-m-mcus/developer-resources/ides-for-kinetis-mcus/opensda-serial-and-debug-adapter:OPENSDA?&tid=vanOpenSDA#FRDM-KW24D512).  I have tested it and it is functional.  When the mbed platform page for this board does live, the firmware update section will point to this page.  If using Windows 10, be sure to disable storage services when updating the interface firmware, otherwise it could get corrupted.   Please note that this platform is intended to go live in a matter of days.

theotherjimmy, 2018-03-01:
DAPlink released. @jupe Please reopen if this is still blocking.

closed 2018-03-01 by theotherjimmy