ARMmbed/mbed-ls #155: Support for BeagleBone Black

JanneKiiskila, 2017-01-31:
Would it be possible to have support for BeagleBone Black?

bridadan, 2017-01-31:
Hey @JanneKiiskila, I had a brief discussion with @jupe yesterday about this. The motivation behind this is to enable testing automation for these boards right? My initial reaction was to not add support for it in mbedls because it is not actually an mbed enabled device. It does seem to have a USB interface, but it doesn't seem very similar to the interfaces like DAPLink, ST Link, JLink, OpenSDA, etc.

Right now the detection method used in mbedls is the same for all devices because all the supported interfaces follow the same requirements. I'd prefer not to add anything that becomes the "exception" to this.

Now, if we start branding other Linux boards as "mbed enabled" then maybe we'll have to revisit this. But for now I would vote that this kind of detection doesn't belong in mbedls.

mazimkhan, 2017-02-15:
I agree with Brian on this.

Moreover, if you run Linux on it, why don't treat it as a Raspberry Pi and run RaaS Client on it.

theotherjimmy, 2017-09-21:
Closing this. This is not something mbedls plans to do.

closed 2017-09-21 by theotherjimmy