ARMmbed/mbed-ls #163: Python 3 compatibility and reliability

hvenev, 2017-03-06:
```
Traceback (most recent call last):
  File "/usr/bin/mbedls", line 9, in <module>
    load_entry_point('mbed-ls==1.2.12', 'console_scripts', 'mbedls')()
  File "/usr/lib/python3.5/site-packages/mbed_lstools/main.py", line 218, in mbedls_main
    print(mbeds.get_string(border=not opts.simple, header=not opts.simple))
  File "/usr/lib/python3.5/site-packages/mbed_lstools/lstools_base.py", line 577, in get_string
    mbeds = self.list_mbeds_ext()
  File "/usr/lib/python3.5/site-packages/mbed_lstools/lstools_base.py", line 442, in list_mbeds_ext
    mbeds = self.list_mbeds()
  File "/usr/lib/python3.5/site-packages/mbed_lstools/lstools_linux_generic.py", line 84, in list_mbeds
    mbeds = self.get_detected(tids, disk_ids, serial_ids, mount_ids)
  File "/usr/lib/python3.5/site-packages/mbed_lstools/lstools_linux_generic.py", line 232, in get_detected
    mbed_mount_point = self.get_mount_point(mbed_dev_disk, mount_list)
  File "/usr/lib/python3.5/site-packages/mbed_lstools/lstools_linux_generic.py", line 299, in get_mount_point
    m = mmp.search(mount)
TypeError: cannot use a string pattern on a bytes-like object
```

The current approach to listing vfat filesystems is wrong:

```
$ mount /dev/sda42 /mnt/vfat
$ mount | grep vfat
/dev/sda42 on /mnt/vfat type ext4 (rw,relatime)
```

It is also unfixable:

```
$ mkdir '/tmp/a type proc'
$ mount -t tmpfs none '/tmp/a type proc'
$ mount
none on /tmp/a type proc type tmpfs (rw,relatime)
```

Some filesystems (e.g. btrfs, overlayfs) allow arbitrary strings in mount options. This can be used to create an ambiguity.

Use libmount instead (`/etc/mtab` escapes bad characters):

```
>>> import libmount
>>> ctx=libmount.Context()
>>> while True:
>>>     fs=ctx.mtab.next_fs()
>>>     if fs is None: break
>>>     print(repr((fs.target,fs.source)))
('/tmp/a type proc', 'none')
```

Note that this is still only marginally better. It still doesn't work on most random testcases (I've reported the bug to util-linux):

```
$ mkdir $'/tmp/\xa0'
$ mount -t tmpfs none $'/tmp/\xa0'
>>> import libmount
>>> ctx=libmount.Context()
>>> while True:
>>>     fs=ctx.mtab.next_fs()
>>>     if fs is None: break
>>>     print(repr((fs.target,fs.source)))
RuntimeError: Error during object construction
```

theotherjimmy, 2017-09-19:
@hvenev Take a look again after #167 Was merged.

theotherjimmy, 2017-09-21:
It's not working just yet for me.

bridadan, 2017-09-28:
@theotherjimmy This issue is probably good to close now? Or do you want to want until a release?

theotherjimmy, 2017-09-28:
Yep. closing.

closed 2017-09-28 by theotherjimmy