ARMmbed/mbed-ls #165: identify ST boards firmware version correct way

jupe, 2017-03-08:
Seems that ST boards or any dev kit using ST-Link firmware need to be identified differently than DAPLink boards (see STLink utility tool). Now mbed-ls gives you incorrect "daplink_version" number and makes confusion.

bridadan, 2017-03-08:
@jupe True, it does report "daplink_version" for all boards, even if they aren't based on DAPLink. May I ask how you're using this information?

jupe, 2017-03-08:
in test beds we want to know what firmware version we have in boards, and update if there is old ones...

jeromecoutant, 2017-10-11:
Hi

Maybe I can propose STM32 ST-LINK utility software from
http://www.st.com/en/development-tools/stsw-link004.html

ST-LINK_CLI.exe -List
will give you the list of all connected STM32 boards

ex:
--- Available ST-LINK Probes List ---
ST-LINK Probe 0:
     SN: 066BFF545253675187212436
     FW: V2J27M15

ST-LINK Probe 1:
     SN: 0672FF504852897267175941
     FW: V2J29M18

I tried few commands to associate SN with a board.
I found this one.

If you have only 1 board connected:

$ ./ST-LINK_CLI.exe -c
ST-LINK SN : 066BFF545253675187212436
ST-LINK Firmware version : V2J27M15
Connected via SWD.
SWD Frequency = 4000K.
Target voltage = 3.3 V.
Connection mode : Normal.
Device ID:0x458
Device family :STM32F410xx

If your different boards are from different family, you can keep them connected and indicate SN like:

$ ./ST-LINK_CLI.exe -c SN=066BFF545253675187212436
ST-LINK SN : 066BFF545253675187212436
ST-LINK Firmware version : V2J27M15
Connected via SWD.
SWD Frequency = 4000K.
Target voltage = 3.3 V.
Connection mode : Normal.
Device ID:0x458
Device family :STM32F410xx

theotherjimmy, 2017-10-11:
@jeromecoutant And what do you have for a Linux host?

bridadan, 2018-07-24:
This can be closed now!

closed 2018-07-25 by theotherjimmy