ARMmbed/mbed-ls #167: python3 compatibility issues

jupe, 2017-03-29:
There is at least couple issues I've found so far which prevent to use mbedls from python3

**Environment:** win7
**python:** Python 3.6.1
**mbed-ls** tip of master (8363261250a5ff64af9b2574187d59d83a532cb8)

## Installation
```
Traceback (most recent call last):
  File "setup.py", line 38, in <module>
    long_description=read('README.md'),
  File "setup.py", line 33, in read
    return open(os.path.join(os.path.dirname(__file__), fname)).read()
  File "C:\Users\myuser\AppData\Local\Programs\Python\Python36\lib\encodings\cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
UnicodeDecodeError: 'charmap' codec can't decode byte 0x9d in position 14188: character maps to <undefined>
```

## Usage
```
  File "C:\Users\myuser\AppData\Local\Programs\Python\Python36\lib\site-packages\mbed_ls-1.2.12-py3.6.egg\mbed_lstools\lstools_win7.py", line 213, in <listcomp>
    result += [d for d in self.get_dos_devices() if ven.upper() in d[1].upper()]
AttributeError: 'filter' object has no attribute 'upper'
```

jupe, 2017-04-27:
Installation seems to work better with Mac (+py3) but when I call mbed-ls it just stuck.

bridadan, 2017-04-27:
No news on this. If this is blocking you we need to get this work scheduled in.

jupe, 2017-05-22:
blocking: IOTSYST-2314 (not critical yet)

jupe, 2017-05-30:
created IOTMORF-981. any news?

mazimkhan, 2017-05-30:
@jupe If you are expecting any work on this. I would say I don't have any bandwidth for this. Please escalate and assign it in your team.

adbridge, 2017-05-30:
@jupe Likewise, unless it is scheduled into our sprints (which could be a while as it isn't a fundamental need for us) then I would suggest taking on the work yourselves, including rigorous testing to ensure backwards compatibility...

bridadan, 2017-05-30:
@jupe perhaps this should be taken into account with my proposed mbed-ls refactor here? https://github.com/ARMmbed/mbed-ls/pull/157

I have not tested with python 3 however this could be added to tests.

jupe, 2017-06-09:
Fix PR is available: https://github.com/ARMmbed/mbed-ls/pull/180

theotherjimmy, 2017-09-19:
Done.

closed 2017-09-19 by theotherjimmy