ARMmbed/mbed-ls #186: mbedls displays "unknown" serial_port on Raspberry PI

mbanders, 2017-07-14:
I'm using a Raspberry PI:

```
$ uname -a
Linux raspberrypi 4.4.34-v7+ #930 SMP Wed Nov 23 15:20:41 GMT 2016 armv7l GNU/Linux
```

And have a K64F board connected, but `mbedls` shows "unknown" for serial_port:

```
$ mbedls 
+---------------+----------------------+-------------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point       | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------------+-------------+--------------------------------------------------+-----------------+
| K64F          | K64F[0]              | /media/pi/DAPLINK | unknown     | 0240000032044e45000a700a997b00356781000097969900 | 0243            |
+---------------+----------------------+-------------------+-------------+--------------------------------------------------+-----------------+
```

As a sanity check, when I plug that same K64F board onto a MacBook, it shows the correct serial_port:

```
$ mbedls
+---------------+----------------------+------------------+------------------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point      | serial_port            | target_id                                        | daplink_version |
+---------------+----------------------+------------------+------------------------+--------------------------------------------------+-----------------+
| K64F          | K64F[0]              | /Volumes/DAPLINK | /dev/tty.usbmodem14412 | 0240000032044e45000a700a997b00356781000097969900 | 0243            |
+---------------+----------------------+------------------+------------------------+--------------------------------------------------+-----------------+
```

bridadan, 2017-07-14:
Well this should definitely work since lots of other teams use this! Can you run the following command and post the output?

```
mbedls --debug
```

Also are you using Raspbian or a different distribution? And if you run `mbedls` with sudo does it change the output at all?

mbanders, 2017-07-14:
I'm running Raspbian:

```
$ cat /etc/issue
Raspbian GNU/Linux 8 \n \l
```

Here's mbedls debug info:

```
$ mbedls --debug
debug @MbedLsToolsLinuxGeneric.mbed_lstools.main: mbed-ls ver. 1.2.14
debug @MbedLsToolsLinuxGeneric.mbed_lstools.main: host: ('posix', 'Linux', '4.4.34-v7+', '#930 SMP Wed Nov 23 15:20:41 GMT 2016', 'linux2')
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 13 Nov 25  2016 mmc-SL32G_0xb1b32722 -> ../../mmcblk0
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 15 Nov 25  2016 mmc-SL32G_0xb1b32722-part1 -> ../../mmcblk0p1
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 15 Nov 25  2016 mmc-SL32G_0xb1b32722-part2 -> ../../mmcblk0p2
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 15 Nov 25  2016 mmc-SL32G_0xb1b32722-part5 -> ../../mmcblk0p5
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 15 Nov 25  2016 mmc-SL32G_0xb1b32722-part6 -> ../../mmcblk0p6
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root 15 Nov 25  2016 mmc-SL32G_0xb1b32722-part7 -> ../../mmcblk0p7
debug @MbedLsToolsLinuxGeneric.get_dev_by_id_process: lrwxrwxrwx 1 root  9 Jul 14 17:54 usb-MBED_VFS_0240000032044e45000a700a997b00356781000097969900-0:0 -> ../../sda
debug @MbedLsToolsLinuxGeneric.get_mounts: mount | grep vfat
debug @MbedLsToolsLinuxGeneric.get_mounts: /dev/mmcblk0p6 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
debug @MbedLsToolsLinuxGeneric.get_mounts: /dev/sda on /media/pi/DAPLINK type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0077,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
debug @MbedLsToolsLinuxGeneric.scan_html_line_for_target_id: window.location.replace("https://mbed.org/device/?code=0240000032044e45000a700a997b00356781000097969900?version=0243?target_id=001bffffffffffff4e45369320010001");
debug @MbedLsToolsLinuxGeneric.scan_html_line_for_target_id: (('0240000032044e45000a700a997b00356781000097969900',), '0240000032044e45000a700a997b00356781000097969900')
debug @MbedLsToolsLinuxGeneric.list_mbeds_ext: ('K64F[0]', '0240000032044e45000a700a997b00356781000097969900')
+---------------+----------------------+-------------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point       | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------------+-------------+--------------------------------------------------+-----------------+
| K64F          | K64F[0]              | /media/pi/DAPLINK | unknown     | 0240000032044e45000a700a997b00356781000097969900 | 0243            |
+---------------+----------------------+-------------------+-------------+--------------------------------------------------+-----------------+
debug @MbedLsToolsLinuxGeneric.mbed_lstools.main: Return code: 0
```

My theory at the moment is that `udev` isn't creating a "serial" directory in "/dev/", and that's where mbed-ls is trying to look:

```
$ ls /dev/
autofs           gpiomem  loop-control        ppp    ram6     tty0   tty22  tty36  tty5   tty63      vcs1   vcsm
block            hidraw0  mapper              ptmx   ram7     tty1   tty23  tty37  tty50  tty7       vcs2   vhci
bsg              hwrng    mem                 pts    ram8     tty10  tty24  tty38  tty51  tty8       vcs3   watchdog
btrfs-control    initctl  memory_bandwidth    ram0   ram9     tty11  tty25  tty39  tty52  tty9       vcs4   watchdog0
bus              input    mmcblk0             ram1   random   tty12  tty26  tty4   tty53  ttyAMA0    vcs5   xconsole
cachefiles       kmsg     mmcblk0p1           ram10  raw      tty13  tty27  tty40  tty54  ttyprintk  vcs6   zero
char             log      mmcblk0p2           ram11  rfkill   tty14  tty28  tty41  tty55  uhid       vcs7
console          loop0    mmcblk0p5           ram12  sda      tty15  tty29  tty42  tty56  uinput     vcsa
cpu_dma_latency  loop1    mmcblk0p6           ram13  serial1  tty16  tty3   tty43  tty57  urandom    vcsa1
cuse             loop2    mmcblk0p7           ram14  shm      tty17  tty30  tty44  tty58  usb        vcsa2
disk             loop3    mqueue              ram15  snd      tty18  tty31  tty45  tty59  vc-cma     vcsa3
fb0              loop4    net                 ram2   stderr   tty19  tty32  tty46  tty6   vchiq      vcsa4
fd               loop5    network_latency     ram3   stdin    tty2   tty33  tty47  tty60  vcio       vcsa5
full             loop6    network_throughput  ram4   stdout   tty20  tty34  tty48  tty61  vc-mem     vcsa6
fuse             loop7    null                ram5   tty      tty21  tty35  tty49  tty62  vcs        vcsa7
```

mbanders, 2017-07-14:
Also:

```
$ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

bridadan, 2017-07-14:
@mbanders We determined this is was an issue with your linux environment (it wasn't enumerating the serial device). Could you close this issue?

mbanders, 2017-07-14:
After shutting down and turning on the raspberry pi, it now works:

```
$ mbedls 
+---------------+----------------------+-------------------+--------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point       | serial_port  | target_id                                        | daplink_version |
+---------------+----------------------+-------------------+--------------+--------------------------------------------------+-----------------+
| K64F          | K64F[0]              | /media/pi/DAPLINK | /dev/ttyACM0 | 0240000032044e45000a700a997b00356781000097969900 | 0243            |
+---------------+----------------------+-------------------+--------------+--------------------------------------------------+-----------------+
```

I think here's why: I had done `sudo apt-get update && sudo apt-get upgrade`, and then I installed mbed-ls and plugged in the K64F. Maybe the raspberry pi needed to be rebooted after that? Maybe there was some kernel weirdness? It's working now though!

closed 2017-07-14 by mbanders

bridadan, 2017-07-14:
Huh too weird! Glad it worked out though, I'll keep that in mind in case I see similar problems in the future!

mbanders, 2017-07-14:
I talked to someone more familiar with linux and he said "if the apt upgrade has kernel updates, you should reboot." I didn't know that. I'll do that from now on. Thanks for your help!