ARMmbed/mbed-ls #188: Update mbed OS requirements to use latest mbed-ls

MarceloSalazar, 2017-07-25:
I had a problem on my Mac as was not properly detecting my ODIN W2 platform due to an old version of mbed-ls (had 1.2.4 instead of 1.2.14).
After manually updating mbed-ls, both 'mbed detect' and 'mbedls' were showing the correct target.

It seems the mbed OS requirements need an update on a new release (for either patch or minor versions):
https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.5/requirements.txt#L9

@mazimkhan

bridadan, 2017-07-26:
@MarceloSalazar Agreed. Could you please go ahead and submit this PR then?

MarceloSalazar, 2017-07-28:
Before I submit a PR... I'm testing my branch but I'm unable to get the mbed-ls tools updated to 1.2.14 after importing the project:
https://github.com/MarceloSalazar/mbed-os/blob/mbed-ls-requirements/requirements.txt#L9

```
mbed import / git clone --branch mbed-ls-requirements https://github.com/MarceloSalazar/mbed-os
<add main.cpp>
mbed new .
mbed deploy
mbedls --version
1.2.13
```

Do you have any thoughts on what I'm missing here?

bridadan, 2017-07-28:
I don't believe mbed-cli enforces versions at the moment, it only enforces missing packages. That may be why it's not automatically updating it.

MarceloSalazar, 2017-07-28:
Is there any other way to verify that the correct version of the tools is getting installed?
If not, then there is no point in raising a PR for now...

bridadan, 2017-07-28:
We'd need to make the change in mbed CLI, it's possible to do. Feel free to raise an issue there.

theotherjimmy, 2017-09-19:
So, as this is not as issue with mbed-ls, I'm closing it.

closed 2017-09-19 by theotherjimmy