ARMmbed/mbed-ls #195: mbedls shows different board ID depending on host OS

MarceloSalazar, 2017-09-14:
This i

bridadan, 2017-09-14:
@MarceloSalazar I can't say for sure since I don't have the board, but mbed-ls looks in a few different places for a target ID: There's the USB ID which is presented by the USB driver, but there's also the target ID contained within the `DETAILS.TXT` and `MBED.HTML` files. Most recent firmware unify all of these values, but some older ones do not. It will check the database for both of these target IDs and if it doesn't find it in there, it picks one of them to display as `unknown`.

If it happens to find the other target ID in the database, then in this case this is what's presented. I have a feeling this is what's happening here, however I can't say for sure without one of the boards.

MarceloSalazar, 2017-09-14:
@bridadan if you want to try, there are boards in your office ;)
I think you're right, there are different IDs and the mbed-ls tool seems to look at these and generates a different output depending on whether the 'main' ID is in the database.

theotherjimmy, 2017-10-10:
@MarceloSalazar This is expected behavoir. Closing.

closed 2017-10-10 by theotherjimmy