ARMmbed/mbed-ls #226: Windows python 3 install failed

YifeiZuo01, 2017-10-05:
```
11:26:00 [windows]   File "C:\Jenkins\ws\mbed-flasher_PR-60-RPJGRYMVYMYIIMXQKOTXDTY3SM4QN\mbed-flasher\py3venv\lib\site-packages\setuptools\sandbox.py", line 246, in runner
11:26:00 [windows]     _execfile(setup_script, ns)
11:26:00 [windows]   File "C:\Jenkins\ws\mbed-flasher_PR-60-RPJGRYMVYMYIIMXQKOTXDTY3SM4QN\mbed-flasher\py3venv\lib\site-packages\setuptools\sandbox.py", line 47, in _execfile
11:26:00 [windows]     exec(code, globals, locals)
11:26:00 [windows]   File "C:\WINDOWS\TEMP\easy_install-6fsantz9\mbed-ls-1.2.14\setup.py", line 38, in <module>
11:26:00 [windows]     setup(name='mbed-flasher',
11:26:00 [windows]   File "C:\WINDOWS\TEMP\easy_install-6fsantz9\mbed-ls-1.2.14\setup.py", line 33, in read
11:26:00 [windows]     """
11:26:00 [windows]   File "c:\Python36\lib\encodings\cp1252.py", line 23, in decode
11:26:00 [windows]     return codecs.charmap_decode(input,self.errors,decoding_table)[0]
11:26:00 [windows] UnicodeDecodeError: 'charmap' codec can't decode byte 0x9d in position 14188: character maps to <undefined>
```

YifeiZuo01, 2017-10-05:
seems 1.2.16 is fine to install, I need to looking more, if not an issure,  i will close

YifeiZuo01, 2017-10-05:
However, 1.2.16 doesn't have flasher compatiblity with python3.
```
11:04:33 [linux]   File "/builds/ws/mbed-flasher_PR-60-RPJGRYMVYMYIIMXQKOTXDTY3SM4QN/mbed-flasher/.py3venv/lib/python3.4/site-packages/mbed_ls-1.2.16-py3.4.egg/mbed_lstools/lstools_linux_generic.py", line 299, in get_mount_point
11:04:33 [linux]     m = mmp.search(mount)
11:04:33 [linux] TypeError: can't use a string pattern on a bytes-like object
```

theotherjimmy, 2017-10-05:
@YifeiZuo01 It looks like you are using python 3.4. mbed LS only supports python versions 2.7, 3.5 and 3.6.

Further, I think master has resolved this issue.

bridadan, 2017-10-05:
Also weirdly enough, I don't think a 1.2.16 release was ever made on GitHub @theotherjimmy. Installed ok from pip though.

YifeiZuo01, 2017-10-06:
Thanks! The CI slave installed 3.4 currently, considering upgrade version.
Yep, the 1.2.16 installed from pip. 
Master solved the flasher py3 support, but wondering why it was in 1.2.16.
Anyway, different version used on different slave. I'll close this.

closed 2017-10-06 by YifeiZuo01

theotherjimmy, 2017-10-06:
Master has not been released yet, as it needs more QE/QA. I will release what's currently on master as 1.3.0 after that happens.