ARMmbed/mbed-ls #235: DAPLink devices in bootloader mode are no longer detected

c1728p9, 2017-10-11:
DAPLink CI is broken since devices in bootloader mode are no longer detected. This worked in 1.2.14 but is broken in 1.3.1.

mbed-ls (1.3.1)
---
```
> mbedls -d
DEBUG:mbedls.main:mbed-ls ver. 1.3.1
DEBUG:mbedls.main:host: ('nt', 'Windows', '10', '10.0.14393', 'win32')
WARNING:root:Could not read htm on from usb id 0000000033514e45001c500585d40025e981000097969900. Falling back to usb id
```

mbed-ls (1.2.14)
---
```
> mbedls
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| unknown       | None[0]              | D:          | unknown     | 0000000033514e45001c500585d40025e981000097969900 | 0244            |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
```****

bridadan, 2017-10-12:
@c1728p9 Thanks for the heads up, please pin your CI at 1.2.14 for the time being while we investigate.

bridadan, 2017-10-12:
I can confirm that this is currently an issue on master.

**master**
```
$ mbedls -d -u
DEBUG:mbedls.main:mbed-ls ver. 1.2.15
DEBUG:mbedls.main:host: ('nt', 'Windows', '10', '10.0.14393', 'win32')
WARNING:root:Could not read htm on from usb id 0672FF485550755187045151. Falling back to usb id
WARNING:root:Could not read htm on from usb id 0240000032044e4500257009997b00386781000097969900. Falling back to usb id
WARNING:root:Could not read htm on from usb id 0000000032044e4500257009997b00386781000097969900. Falling back to usb id
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port |                    target_id                     | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
|      K64F     |       K64F[0]        |      E:     |    COM14    | 0240000032044e4500257009997b00386781000097969900 |     unknown     |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
```

```
mbedls -d
DEBUG:mbedls.main:mbed-ls ver. 1.2.15
DEBUG:mbedls.main:host: ('nt', 'Windows', '10', '10.0.14393', 'win32')
WARNING:mbedls.lstools_base:MBED with target id '0240000032044e4500257009997b00386781000097969900' is connected, but not mounted. Use the '-u' flag to include it in the list.
WARNING:root:Could not read htm on from usb id 0000000032044e4500257009997b00386781000097969900. Falling back to usb id
```

**1.2.14**
```
mbedls -d
debug @MbedLsToolsWin7.mbed_lstools.main: mbed-ls ver. 1.2.15
debug @MbedLsToolsWin7.mbed_lstools.main: host: ('nt', 'Windows', '10', '10.0.14393', 'win32')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\F:', '_??_USBSTOR#Disk&Ven_MBED&Prod_microcontroller&Rev_1.0#8&bcddc05&0&0672FF485550755187045151&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\E:', '_??_USBSTOR#Disk&Ven_MBED&Prod_VFS&Rev_0.1#8&ad58317&0&0240000032044e4500257009997b00386781000097969900&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\D:', '_??_USBSTOR#Disk&Ven_MBED&Prod_VFS&Rev_0.1#0000000032044e4500257009997b00386781000097969900&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbeds: ('F:', '0672FF485550755187045151')
debug @MbedLsToolsWin7.get_mbeds: ('E:', '0240000032044e4500257009997b00386781000097969900')
debug @MbedLsToolsWin7.get_mbeds: ('D:', '0000000032044e4500257009997b00386781000097969900')
debug @MbedLsToolsWin7.mount_point_ready: Mount point F: reported not ready with error 'The system cannot find the path specified.'
debug @MbedLsToolsWin7.mount_point_ready: Mount point E: reported not ready with error 'The system cannot find the path specified.'
debug @MbedLsToolsWin7.mount_point_ready: Mount point D: is ready
debug @MbedLsToolsWin7.get_mbed_com_port: ID: 0000000032044e4500257009997b00386781000097969900
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\F:', '_??_USBSTOR#Disk&Ven_MBED&Prod_microcontroller&Rev_1.0#8&bcddc05&0&0672FF485550755187045151&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\E:', '_??_USBSTOR#Disk&Ven_MBED&Prod_VFS&Rev_0.1#8&ad58317&0&0240000032044e4500257009997b00386781000097969900&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\D:', '_??_USBSTOR#Disk&Ven_MBED&Prod_VFS&Rev_0.1#0000000032044e4500257009997b00386781000097969900&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbeds: ('F:', '0672FF485550755187045151')
debug @MbedLsToolsWin7.get_mbeds: ('E:', '0240000032044e4500257009997b00386781000097969900')
debug @MbedLsToolsWin7.get_mbeds: ('D:', '0000000032044e4500257009997b00386781000097969900')
debug @MbedLsToolsWin7.mount_point_ready: Mount point F: reported not ready with error 'The system cannot find the path specified.'
debug @MbedLsToolsWin7.mount_point_ready: Mount point E: reported not ready with error 'The system cannot find the path specified.'
debug @MbedLsToolsWin7.mount_point_ready: Mount point D: is ready
debug @MbedLsToolsWin7.list_mbeds_ext: ('None[0]', '0000000032044e4500257009997b00386781000097969900')
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| unknown       | None[0]              | D:          | unknown     | 0000000032044e4500257009997b00386781000097969900 | 0243            |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
debug @MbedLsToolsWin7.mbed_lstools.main: Return code: -1
```

PR #239 (2017-10-12): Show unknown boards/devices in bootloader mode
Previously, the filtering logic was pulling out devices that didn't have
a known platform name. This restores the ability to show devices that
don't have known platform name.

Fixes #235
closed 2017-10-12 by theotherjimmy