ARMmbed/mbed-ls #238: Tests do not run on Windows

bridadan, 2017-10-12:
The tests currently fail with a traceback when ran on Windows:

```
$ python setup.py test
running test
running egg_info
writing requirements to mbed_ls.egg-info\requires.txt
writing mbed_ls.egg-info\PKG-INFO
writing top-level names to mbed_ls.egg-info\top_level.txt
writing dependency_links to mbed_ls.egg-info\dependency_links.txt
writing entry points to mbed_ls.egg-info\entry_points.txt
reading manifest file 'mbed_ls.egg-info\SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'mbed_ls.egg-info\SOURCES.txt'
running build_ext
Traceback (most recent call last):
  File "setup.py", line 63, in <module>
    "colorized_logs": ["colorlog"]
  File "C:\Python27\lib\distutils\core.py", line 151, in setup
    dist.run_commands()
  File "C:\Python27\lib\distutils\dist.py", line 953, in run_commands
    self.run_command(cmd)
  File "C:\Python27\lib\distutils\dist.py", line 972, in run_command
    cmd_obj.run()
  File "C:\Python27\lib\site-packages\setuptools\command\test.py", line 210, in run
    self.run_tests()
  File "C:\Python27\lib\site-packages\setuptools\command\test.py", line 231, in run_tests
    testRunner=self._resolve_as_ep(self.test_runner),
  File "C:\Python27\lib\unittest\main.py", line 94, in __init__
    self.parseArgs(argv)
  File "C:\Python27\lib\unittest\main.py", line 149, in parseArgs
    self.createTests()
  File "C:\Python27\lib\unittest\main.py", line 158, in createTests
    self.module)
  File "C:\Python27\lib\unittest\loader.py", line 130, in loadTestsFromNames
    suites = [self.loadTestsFromName(name, module) for name in names]
  File "C:\Python27\lib\unittest\loader.py", line 103, in loadTestsFromName
    return self.loadTestsFromModule(obj)
  File "C:\Python27\lib\site-packages\setuptools\command\test.py", line 42, in loadTestsFromModule
    tests.append(self.loadTestsFromName(submodule))
  File "C:\Python27\lib\unittest\loader.py", line 100, in loadTestsFromName
    parent, obj = obj, getattr(obj, part)
AttributeError: 'module' object has no attribute 'os_linux_generic'
```

bridadan, 2017-10-12:
I should not I tried to debug this but I didn't get very far. Leaving this here so we don't forget about it.

theotherjimmy, 2017-10-12:
I think #242 fixed this.