ARMmbed/mbed-ls #244: 1.3.2 breaks windows python 3.6.2 supports

YifeiZuo01, 2017-10-13:
### Python version
3.6.2

### Description
1.3.2 breaks windows python 3.6.2 supports for mbed-flasher

----------------------------------------------------------------------
ImportError: Failed to import test module: test_verify_flash
Traceback (most recent call last):
  File "c:\Python36\lib\unittest\loader.py", line 428, in _find_test_path
    module = self._get_module_from_name(name)
  File "c:\Python36\lib\unittest\loader.py", line 369, in _get_module_from_name
    __import__(name)
  File "C:\Jenkins\ws\mbed-flasher\test\test_verify_flash.py", line 74, in <module>
    @unittest.skipIf(mbed.list_mbeds() == [], "no hardware attached")
  File "c:\jenkins\ws\mbed-flasher\py3venv\lib\site-packages\mbed_lstools\lstools_base.py", line 128, in list_mbeds
    candidates = list(self.find_candidates())
  File "c:\jenkins\ws\mbed-flasher\py3venv\lib\site-packages\mbed_lstools\windows.py", line 50, in find_candidates
    for mnt, id in self.get_mbeds()
  File "c:\jenkins\ws\mbed-flasher\py3venv\lib\site-packages\mbed_lstools\windows.py", line 136, in get_mbeds
    for mbed in self.get_mbed_devices():
  File "c:\jenkins\ws\mbed-flasher\py3venv\lib\site-packages\mbed_lstools\windows.py", line 179, in get_mbed_devices
    if (b'DosDevices' in point and
TypeError: 'in <string>' requires string as left operand, not bytes

````

YifeiZuo01, 2017-10-13:
feels like test of mbed-ls release doesn't consider all of those conditions each time:

- python 3 with windows
- python 2 windows
- python 2 linux
- python 3 linux

YifeiZuo01, 2017-10-13:
Could you please provide some clue for this errors:

### windows, python 2, mbed-ls 1.3.2
```
AttributeError: 'MbedLsToolsWin7' object has no attribute 'manufacture_ids'
```

### linux, python 2, 
```
AttributeError: 'MbedLsToolsLinuxGeneric' object has no attribute 'manufacture_ids'
```

YifeiZuo01, 2017-10-13:
seems some refactor removed the list `manufacture_ids` which was in lstools_base, but now it was presented in a new way

theotherjimmy, 2017-10-16:
> 1.3.2 breaks windows python 3.6.2 supports for mbed-flasher

I'd bet that it's broken on other prior versions too. mbed-flasher does not work with 1.3.x. 

> feels like test of mbed-ls release doesn't consider all of those conditions each time:

No, we "intend to consider" all of the platforms we support, and have testing to mock each OS so that we don't have to have a pile of machines and boards plugged into them. It sounds like the tests were written with the incorrect unicode or bytes strings.

YifeiZuo01, 2017-10-17:
@theotherjimmy Thanks! yes, you're right, it doesn't work with 1.3.x.

>No, we "consider" all of the platforms we support, 

I guess it's Darwin, Linux and windows. Considering the docs saying it support Python 2.7.x and Python 3.6.x. It supports at least two more conditions than I have listed.  :)

theotherjimmy, 2017-10-17:
@YifeiZuo01 You are correct the full list should be:

* python 3.5.x 
  * Windows 7+
  * Darwin 10.2+
  * Linux with udev
* python 3.6.x 
  * Windows 7+
  * Darwin 10.2+
  * Linux with udev
* python 2.7.x 
  * Windows 7+
  * Darwin 10.2+
  * Linux with udev

That's what we test in CI at least. It sounds like the tests for Windows contain the wrong bytes types. Python 3+ is very picky about that sort of thing. I think @bridadan Said he was looking into this offline yesterday.

juhhov, 2017-10-17:
What is the reason python 3.4.x is not supported?

theotherjimmy, 2017-10-17:
@juhhov Because I had to draw the line somewhere. for Linux distros: Ubuntu does 3.5, Archlinux 3.6. I usually take Ubuntu as a marker for the most out of date distro, and Archlinux and the most up to date. If there is a reason you want python 3.4.x, please explain! We can always add more pythons to the list, Travis CI is pretty good about having many versions.

closed 2017-10-20 by theotherjimmy

juhhov, 2017-10-24:
We don't want python 3.4, at least currently. I was just wondering did python versions have some major differences that lead you into this.

P.s. Ubuntu 14.04 defaults 3.4.x I believe, that is still quite commonly in use.

theotherjimmy, 2017-10-24:
>  I was just wondering did python versions have some major differences that lead you into this.

Nope. I just turned a few on in Travis. If you need python 3.4 support, you can submit a PR to turn testing on for it too.

>P.s. Ubuntu 14.04 defaults 3.4.x I believe, that is still quite commonly in use.

Good to know. I recommend that people on 14.04 upgrade to 16.04 or later, as 14.04 is quite old these days.

PR #250 (2017-10-20): Correct type for Windows Registry check
Fixes #244