ARMmbed/mbed-ls #248: mbed-ls with boards mounted for another user dies

geky, 2017-10-16:
Ubuntu 17.04 zesty
mbedls --version 1.3.3

1. mount board as userA
2. run mbedls as userB

It dies without mentioning that there might be a permission issue. This can happen easily if userA has an automounter.

```bash
$ mbedls -d
DEBUG:mbedls.main:mbed-ls ver. 1.3.3
DEBUG:mbedls.main:host: ('posix', 'Linux', '4.10.0-19-generic', '#21-Ubuntu SMP Thu Apr 6 17:04:57 UTC 2017', 'linux2')
DEBUG:mbedls.lstools_linux:Converting device list ['/dev/disk/by-id/usb-MBED_VFS_0217000033514e45002b5015846c00486cb1000097969900-0:0', '/dev/disk/by-id/wwn-0x5000c500a3a88231-part1', '/dev
/disk/by-id/ata-ST1000DM003-1SB102_Z9AFJ9QY-part1', '/dev/disk/by-id/wwn-0x5000c500a3a88231-part2', '/dev/disk/by-id/ata-ST1000DM003-1SB102_Z9AFJ9QY-part2', '/dev/disk/by-id/wwn-0x5000c500a
3a88231', '/dev/disk/by-id/ata-ST1000DM003-1SB102_Z9AFJ9QY', '/dev/disk/by-id/ata-PLDS_DVD-RW_DA8AESH_DX0K84169L1CB7401493', '/dev/disk/by-id/nvme-eui.002538b671b4fd39-part1', '/dev/disk/by
-id/nvme-SAMSUNG_MZVLW256HEHP-000L7_S35ENX0J677706-part1', '/dev/disk/by-id/nvme-eui.002538b671b4fd39', '/dev/disk/by-id/nvme-SAMSUNG_MZVLW256HEHP-000L7_S35ENX0J677706']
DEBUG:mbedls.lstools_linux:Found disk devices by id {'0217000033514e45002b5015846c00486cb1000097969900': '/dev/sdb'}
DEBUG:mbedls.lstools_linux:Converting device list ['/dev/serial/by-id/usb-ARM_DAPLink_CMSIS-DAP_0217000033514e45002b5015846c00486cb1000097969900-if01']
DEBUG:mbedls.lstools_linux:Found serial devices by id {'0217000033514e45002b5015846c00486cb1000097969900': '/dev/ttyACM0'}
DEBUG:mbedls.lstools_linux:Mount mapping {u'/dev/sdb': u'/media/arm/FRDM-K82FD'}
DEBUG:mbedls.lstools_base:Candidates for display [{'mount_point': u'/media/arm/FRDM-K82FD', 'serial_port': '/dev/ttyACM0', 'target_id_usb_id': '0217000033514e45002b5015846c00486cb1000097969
900'}]
WARNING:mbedls.lstools_base:MBED with target id '0217000033514e45002b5015846c00486cb1000097969900' is connected, but not mounted. Use the '-u' flag to include it in the list.
DEBUG:mbedls.main:Return code: 0
```

cc @theotherjimmy

theotherjimmy, 2017-10-16:
Thanks! I'll look at this in the near future. Maybe sometime this week even!

theotherjimmy, 2017-10-16:
FYI, this will probably affect any OS, it's just that Linux is much more likely to run into this.

theotherjimmy, 2017-10-16:
@geky I'm thinking that a different warning should be printed, and for the board to show up with possibly incorrect details in the listing. Does that sound reasonable?

geky, 2017-10-17:
Yep, sounds good, some sort of indication that the failure is permission related (the mount point does exist) would go a long way.

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*