ARMmbed/mbed-ls #251: Mock interface difficult to use and error prone

theotherjimmy, 2017-10-20:
Currently, to add a mock, you have to do some odd things:
```bash
mbedls -m "C006:VLBUNO51"
```
```bash
mbedls --mock="C006:VLBUNO51"
```
both add a mock for the target id `C006` and the platform name `VLBUNO51`.

While that my seem reasonable, it can also be done by:
```bash
mbedls --mock="+C006:VLBUNO51"
```
Which is a bit complicated.

Removing mocks are where things get nasty:
```bash
mbedls -m "-C006"
```
is in theory part of the syntax, but `-C006` is parsed as a flag.
```bash
mbedls --mock="-C006"
```
is the only working way to actually remove a mock.
The readme also gives the impression that:
```bash
mbedls --mock="!C006"
```
should work, but it does not. Instead, it gives you a lovely trackback:
```
Traceback (most recent call last):
  File "/usr/bin/mbedls", line 11, in <module>
    load_entry_point('mbed-ls==1.3.2', 'console_scripts', 'mbedls')()
  File "/usr/lib/python3.6/site-packages/mbed_lstools/main.py", line 258, in mbedls_main
    ret_code = args.command(mbeds, args)
  File "/usr/lib/python3.6/site-packages/mbed_lstools/main.py", line 127, in mock_platform
    mbeds.mock_manufacture_id(mid, 'dummy', oper=oper)
  File "/usr/lib/python3.6/site-packages/mbed_lstools/lstools_base.py", line 231, in mock_manufacture_id
    raise ValueError("oper can only be [+-]")
ValueError: oper can only be [+-]
```

A much better interface would be:
```bash
mbedls add-mock C006 VLBUNO51
```
for add and:
```bash
mbedls rm-mock C006
```
or
```bash
mbedls remove-mock C006
```
for mock removal.

@yennster @bridadan Does this cover our offline discussion?

yennster, 2017-10-20:
Looks good to me, the readme also gives the impression that 
```
mbedls --mock="!*"
```
should work to remove all mocks.

Which would be a useful feature if mocks gets re-worked.

theotherjimmy, 2017-10-20:
@yennster I think that you would have to do 
```
mbedls --mock="-*"
```
right now

theotherjimmy, 2017-10-20:
Any thoughts on my proposed new interface?

yennster, 2017-10-20:
@theotherjimmy 😢 What does the "!" do then?

yennster, 2017-10-20:
And yes, I like the new interface better, or something that is more consistent with the existing mbed-os style commands

theotherjimmy, 2017-10-20:
Throws an exception 😆 

It should probable be turned into `-` internally

theotherjimmy, 2017-10-20:
I know, it's weird, but this is both a bug report and an enhancement request.

bridadan, 2017-10-23:
@theotherjimmy +1 for the new interface, though let's save that for the next major/minor release. We would need to add a deprecation notice for the `--mock` option.

For the next release, we should get the current implementation inline with the README. To me this would include the following:

- Add the ability to use `!` again as it is currently listed in the README
    - It also mentions being able to use `!*` to remove everything as noted by @yennster. If it's easy to do this for mocked targets only, let's go ahead and do that.
- Add the `+/-` operators to the README (with a note about using quotes to prevent the parsing error)

theotherjimmy, 2017-10-23:
@bridadan It's impossible for:
```bash
mbedls -m -C006
```
to work. That's a limitation of `argparse` as far as I know.

bridadan, 2017-10-23:
@theotherjimmy that should work with quotes correct? Or in combination with `=`? Ex, `mbedls -m="-C006"`. It's ulgy, its just a consequence of the legacy implementation.

theotherjimmy, 2017-10-23:
Surprisingly, It works with `=`. Quotes are not required.

theotherjimmy, 2017-10-23:
```
mbedls -m=-C006
```
works

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*