ARMmbed/mbed-ls #252: Warning for incorrect MBED.HTM format says it can't read the file

theotherjimmy, 2017-10-20:
This is clearly not true, and we should have a better warning.

Here's an example!

```
>mbedls
WARNING:root:Could not read htm on from usb id C006[Redacted]. Falling back to usb id
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port |                    target_id                     | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
|    VBLUNO51   |     VBLUNO51[0]      |      E:     |     COM8    | C006[Redacted] |       0244      |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
```

```
>dir e:
[SNIP]
 Directory of E:\
03/22/2016  04:30 PM               280 MBED.HTM
03/22/2016  04:30 PM               384 DETAILS.TXT
[SNIP]
```

So the `WARNING:...` line should not be printed!

@yennster @bridadan This was the other minor issue we saw offline.

bridadan, 2017-10-23:
Thanks for logging this, I think this would be a really great to feature to have, especially for people bringing up new platforms.

We should definitely to have this in either the major 2.x.x release or a minor release in the meantime.

theotherjimmy, 2017-10-23:
I was thinking 1.3.4. Is that not going to work with semver?

bridadan, 2017-10-23:
It'd be ok in 1.3.4 if you want, but if you wanted to focus on critical fixes only I'd say this is lower on the priority list.

theotherjimmy, 2017-10-23:
That's a good point, if I did not already have an implementation.

bridadan, 2017-10-23:
That works too :P

theotherjimmy, 2017-10-23:
It's like +2/-2 to make this change. I debug statement upgraded to warning, and one warning rewritten.

theotherjimmy, 2018-09-24:
Implementation is already in.

closed 2018-09-24 by theotherjimmy