ARMmbed/mbed-ls #255: Push policy decisions outside of the core library

theotherjimmy, 2017-10-24:
Currently, mbedls prints for users through logging inside of the object created by `mbed_lstools.create` This is a policy decision, and should be left up to consumers of the library or the command line interface.

In short:
 * Policy decisions, like what to do when a target does not have a mount point, should be decided by the API consumer.
 * Side effects, like logging, should be "off by default".
 * Any printing, logging or other side effect outside of the defined API indicates a bug in mbedls, or in the consumer (similar to deprecation warnings).
 * Conditions that may indicate bugs should be logged.

This implies that:
 1. several warning invocations should become debug or info
 2. the `list_mbeds()` function must provide enough information to the API user to reconstruct the warnings that are currently logged
 3.  the `list_mbeds()` function should provide an "ergonomic" method for an API user to discover that one of the warning conditions has been encountered, making this discovery easy and/or explicit. This may be a documentation task, or include new API design.

1 and 2 should come as a single PR, possibly before 3.

@c1728p9 Did this capture our offline discussion?

theotherjimmy, 2017-10-25:
Concretely, the design that @c1728p9 Recently discussed offline is:

1) Create an enum, or similar, that contains all of the current errors that we recover from
1) Add a "recoveries" key in each target
1) Change all current warnings to:
   1) An info/debug logging statement
   1) An append to the "recoveries" list

theotherjimmy, 2017-10-25:
Further, Once this API is in place, the mbedls CLI tool should use these "recoveries" to print out the warnings we once had, with mbedls context (like "add -u").

theotherjimmy, 2017-10-25:
Upon implementing this, I thought that "warnings" would better understood. I have a prototype here: https://github.com/ARMmbed/mbed-ls/compare/master...theotherjimmy:no-warning-policy

bridadan, 2017-10-25:
That prototype looks sweet

theotherjimmy, 2017-10-25:
Thanks! 😄 It fails tests though, hence "prototype".

c1728p9, 2017-11-02:
Sorry for the late response. Yep, this captures our discussion well.