ARMmbed/mbed-ls #259: mbedls v1.3.2 (and onward) is broken on Windows 7 and 10

toyowata, 2017-11-02:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

The mbedls command failed on Windows 7 and Windows 10.

**Target**
LPC1768|K64F|NUCLEO_F429ZI

**Expected behavior**
The mbedls command should execute correctly and show target information.

**Actual behavior**
The program crashed v1.3.2 and onwards.

**Steps to reproduce**

[mbedls v1.3.3]
```
PS C:\tmp> mbedls --version
1.3.3
PS C:\tmp> mbedls
Traceback (most recent call last):
  File "C:\Python27\Scripts\mbedls-script.py", line 11, in <module>
    load_entry_point('mbed-ls==1.3.3', 'console_scripts', 'mbedls')()
  File "c:\python27\lib\site-packages\mbed_lstools\main.py", line 260, in mbedls_main
    ret_code = args.command(mbeds, args)
  File "c:\python27\lib\site-packages\mbed_lstools\main.py", line 109, in print_table
    return print_mbeds(mbeds, args, False)
  File "c:\python27\lib\site-packages\mbed_lstools\main.py", line 97, in print_mbeds
    devices = mbeds.list_mbeds(unique_names=True, read_details_txt=True)
  File "c:\python27\lib\site-packages\mbed_lstools\lstools_base.py", line 128, in list_mbeds
    candidates = list(self.find_candidates())
  File "c:\python27\lib\site-packages\mbed_lstools\windows.py", line 50, in find_candidates
    for mnt, id in self.get_mbeds()
  File "c:\python27\lib\site-packages\mbed_lstools\windows.py", line 134, in get_mbeds
    drive_letters = set(self._drive_letters())
  File "c:\python27\lib\site-packages\mbed_lstools\windows.py", line 123, in _drive_letters
    drive_list = re.match('\s*Drives: (.*)', stdout.decode('utf-8')).group(1)
  File "c:\python27\lib\encodings\utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0x83 in position 2: invalid start byte
```

[mbedls v1.3.2]
```
PS C:\tmp> mbedls --version
Traceback (most recent call last):
  File "C:\Python27\Scripts\mbedls-script.py", line 11, in <module>
    load_entry_point('mbed-ls==1.3.2', 'console_scripts', 'mbedls')()
  File "c:\python27\lib\site-packages\mbed_lstools\main.py", line 236, in mbedls_main
    start_logging()
  File "c:\python27\lib\site-packages\mbed_lstools\main.py", line 230, in start_logging
    logging.basicConfig()
NameError: global name 'logging' is not defined
PS C:\tmp> mbedls
Traceback (most recent call last):
  File "C:\Python27\Scripts\mbedls-script.py", line 11, in <module>
    load_entry_point('mbed-ls==1.3.2', 'console_scripts', 'mbedls')()
  File "c:\python27\lib\site-packages\mbed_lstools\main.py", line 236, in mbedls_main
    start_logging()
  File "c:\python27\lib\site-packages\mbed_lstools\main.py", line 230, in start_logging
    logging.basicConfig()
NameError: global name 'logging' is not defined
```

[mbedls v1.3.1]
```
PS C:\tmp> mbedls --version
1.3.1
PS C:\tmp> mbedls
WARNING:mbedls.lstools_base:MBED with target id '0240000025354e450029400d47d00042bb91000097969900' is connected, but not
 mounted. Use the '-u' flag to include it in the list.
+---------------+----------------------+-------------+-------------+--------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port |        target_id         | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------+-----------------+
| NUCLEO_F429ZI |   NUCLEO_F429ZI[0]   |      D:     |    COM11    | 07960221023E68053D7AFEAE |       0221      |
+---------------+----------------------+-------------+-------------+--------------------------+-----------------+
```

toyowata, 2017-11-02:
This is probably happened in Japanese version of Windows.

bridadan, 2017-11-02:
@toyowata We will look into this. Is there anyway you can provide the output of the following:

1. `mbedls --debug` using version v1.3.3 of mbedls
2. `fsutil fsinfo drives`

As for the error with v1.3.2, that bug made it in to that release but was subsequently patched in v1.3.3.

toyowata, 2017-11-02:
@bridadan 

The result is below:

```
PS C:\Users\toyom\mbed-ls> mbedls --debug
DEBUG:mbedls.main:mbed-ls ver. 1.3.3
DEBUG:mbedls.main:host: ('nt', 'Windows', '10', '10.0.15063', 'win32')
DEBUG:mbedls.lstools_win7:running command: fsutil fsinfo drives
DEBUG:mbedls.lstools_win7:return code: 0
DEBUG:mbedls.lstools_win7:output:
ドライブ: C:\ D:\ E:\

Traceback (most recent call last):
  File "C:\Python27\Scripts\mbedls-script.py", line 11, in <module>
    load_entry_point('mbed-ls==1.3.3', 'console_scripts', 'mbedls')()
  File "c:\python27\lib\site-packages\mbed_lstools\main.py", line 260, in mbedls_main
    ret_code = args.command(mbeds, args)
  File "c:\python27\lib\site-packages\mbed_lstools\main.py", line 109, in print_table
    return print_mbeds(mbeds, args, False)
  File "c:\python27\lib\site-packages\mbed_lstools\main.py", line 97, in print_mbeds
    devices = mbeds.list_mbeds(unique_names=True, read_details_txt=True)
  File "c:\python27\lib\site-packages\mbed_lstools\lstools_base.py", line 128, in list_mbeds
    candidates = list(self.find_candidates())
  File "c:\python27\lib\site-packages\mbed_lstools\windows.py", line 50, in find_candidates
    for mnt, id in self.get_mbeds()
  File "c:\python27\lib\site-packages\mbed_lstools\windows.py", line 134, in get_mbeds
    drive_letters = set(self._drive_letters())
  File "c:\python27\lib\site-packages\mbed_lstools\windows.py", line 123, in _drive_letters
    drive_list = re.match('\s*Drives: (.*)', stdout.decode('utf-8')).group(1)
  File "c:\python27\lib\encodings\utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0x83 in position 2: invalid start byte
PS C:\Users\toyom\mbed-ls>
```
In Japanese Windows, list of drive letter does not start from ```Drives: ```, but ```ドライブ:```.

bridadan, 2017-11-02:
Ah! Perfect, good to know, that's the problem. We'll change the matching logic. I'll try to get a patch up today.

PR #260 (2017-11-03): Fix language matching in Windows
Fixes #259.

Previously the Windows implementation relied on the language of the system to work correctly (specifically, English). This dependency has been removed. It should work on all languages now.
closed 2017-11-03 by theotherjimmy