ARMmbed/mbed-ls #261: left align mbed-ls output

studavekar, 2017-11-02:
Previous version of mbed-ls 1.2.14
```
+-------------------+----------------------+---------------------------------------------------+---------------+--------------------------------------------------+-----------------+
| platform_name     | platform_name_unique | mount_point                                       | serial_port   | target_id                                        | daplink_version |
+-------------------+----------------------+---------------------------------------------------+---------------+--------------------------------------------------+-----------------+
| ARCH_PRO          | ARCH_PRO[0]          | /mnt/pci-0000_00_14_0-usb-0_1_5_1_0-scsi-0_0_0_0  | /dev/ttyACM9  | 9004000008bf3ce200000000000000000000000097969902 | 0243            |
| ARCH_PRO          | ARCH_PRO[1]          | /mnt/pci-0000_00_14_0-usb-0_1_7_1_0-scsi-0_0_0_0  | /dev/ttyACM15 | 9004000009be7aeb00000000000000000000000097969902 | 0243            |
| ARCH_PRO          | ARCH_PRO[2]          | /mnt/pci-0000_00_14_0-usb-0_1_6_1_0-scsi-0_0_0_0  | /dev/ttyACM12 | 9004000009be1b6c00000000000000000000000097969902 | 0243            |
| K22F              | K22F[0]              | /mnt/pci-0000_00_14_0-usb-0_11_1_1_0-scsi-0_0_0_0 | /dev/ttyACM3  | 0231000027944e45005030195e0e00214eb1000097969900 | 0243            |
| K22F              | K22F[1]              | /mnt/pci-0000_00_14_0-usb-0_7_2_1_0-scsi-0_0_0_0  | /dev/ttyACM10 | 0231000033514e45005a500585d40039e981000097969900 | 0243            |
| K22F              | K22F[2]              | /mnt/pci-0000_00_14_0-usb-0_6_4_1_0-scsi-0_0_0_0  | /dev/ttyACM18 | 0231000026334e45003f400d5e0e00294eb1000097969900 | 0243            |
| K64F              | K64F[0]              | /mnt/pci-0000_00_14_0-usb-0_6_7_1_0-scsi-0_0_0_0  | /dev/ttyACM33 | 0240000032044e4500507009997b00256781000097969900 | 0243            |
| K64F              | K64F[1]              | /mnt/pci-0000_00_14_0-usb-0_6_5_1_0-scsi-0_0_0_0  | /dev/ttyACM22 | 0240000031754e45000c0018948500156461000097969900 | 0243            |
| K64F              | K64F[2]              | /mnt/pci-0000_00_14_0-usb-0_6_6_1_0-scsi-0_0_0_0  | /dev/ttyACM26 | 0240000031754e4500270019948500016461000097969900 | 0243            |
| K64F              | K64F[3]              | /mnt/pci-0000_00_14_0-usb-0_7_4_1_0-scsi-0_0_0_0  | /dev/ttyACM16 | 0240000032044e45001d700a997b00076781000097969900 | 0243            |
| K66F              | K66F[0]              | /mnt/pci-0000_00_14_0-usb-0_2_1_0-scsi-0_0_0_0    | /dev/ttyACM0  | 0311000033514e450035500585d4004ee981000097969900 | 0243            |
| NCS36510          | NCS36510[0]          | /mnt/pci-0000_00_14_0-usb-0_11_6_1_0-scsi-0_0_0_0 | /dev/ttyACM31 | 120000004420312059465238313032203631303397969903 | 0243            |
| NCS36510          | NCS36510[1]          | /mnt/pci-0000_00_14_0-usb-0_11_7_1_0-scsi-0_0_0_0 | /dev/ttyACM32 | 120000004420312059465238333032203432303597969903 | 0243            |
| NCS36510          | NCS36510[2]          | /mnt/pci-0000_00_14_0-usb-0_11_5_1_0-scsi-0_0_0_0 | /dev/ttyACM29 | 120000004420312059465238333033203932303097969903 | 0243            |
| NRF51_DK          | NRF51_DK[0]          | /mnt/pci-0000_00_14_0-usb-0_7_6_1_0-scsi-0_0_0_0  | /dev/ttyACM8  | 110000005032312036393738313038313033303497969903 | 0243            |
| NRF51_DK          | NRF51_DK[1]          | /mnt/pci-0000_00_14_0-usb-0_11_4_1_0-scsi-0_0_0_0 | /dev/ttyACM25 | 1100000044203120324c4638333033323631303097969903 | 0243            |
| NRF51_DK          | NRF51_DK[2]          | /mnt/pci-0000_00_14_0-usb-0_11_3_1_0-scsi-0_0_0_0 | /dev/ttyACM21 | 1100000044203120324c4638333034313131303797969903 | 0243            |
| NRF51_DK          | NRF51_DK[3]          | /mnt/pci-0000_00_14_0-usb-0_11_2_1_0-scsi-0_0_0_0 | /dev/ttyACM17 | 110000004420312054434a38323035313232303597969903 | 0243            |
| NRF52_DK          | NRF52_DK[0]          | /mnt/pci-0000_00_14_0-usb-0_7_5_1_0-scsi-0_0_0_0  | /dev/ttyACM20 | 11010000442031204c364141313031313032303197969903 | 0243            |
| NRF52_DK          | NRF52_DK[1]          | /mnt/pci-0000_00_14_0-usb-0_7_3_1_0-scsi-0_0_0_0  | /dev/ttyACM13 | 11010000442031204c505739313030313430303697969903 | 0243            |
| NRF52_DK          | NRF52_DK[2]          | /mnt/pci-0000_00_14_0-usb-0_7_7_1_0-scsi-0_0_0_0  | /dev/ttyACM28 | 11010000442031204c364141323039203031303397969903 | 0243            |
| NUCLEO_F401RE     | NUCLEO_F401RE[0]     | /mnt/pci-0000_00_14_0-usb-0_1_2_1_1-scsi-0_0_0_0  | /dev/ttyACM23 | 07200221072265023866F31F                         | 0221            |
| NUCLEO_F401RE     | NUCLEO_F401RE[1]     | /mnt/pci-0000_00_14_0-usb-0_1_1_1_1-scsi-0_0_0_0  | /dev/ttyACM7  | 072002210774650F3830F312                         | 0221            |
| NUCLEO_F401RE     | NUCLEO_F401RE[2]     | /mnt/pci-0000_00_14_0-usb-0_1_3_1_1-scsi-0_0_0_0  | /dev/ttyACM2  | 07200221073265613876F37C                         | 0221            |
| NUCLEO_F429ZI     | NUCLEO_F429ZI[0]     | /mnt/pci-0000_00_14_0-usb-0_6_1_1_1-scsi-0_0_0_0  | /dev/ttyACM6  | 079602210138673C3E7CF197                         | 0221            |
| NUCLEO_F429ZI     | NUCLEO_F429ZI[1]     | /mnt/pci-0000_00_14_0-usb-0_7_1_1_1-scsi-0_0_0_0  | /dev/ttyACM5  | 07960221053869043A7CFFAF                         | 0221            |
| NUCLEO_F429ZI     | NUCLEO_F429ZI[2]     | /mnt/pci-0000_00_14_0-usb-0_6_2_1_1-scsi-0_0_0_0  | /dev/ttyACM30 | 07960221004C62643F08F4CF                         | 0221            |
| NUCLEO_F429ZI     | NUCLEO_F429ZI[3]     | /mnt/pci-0000_00_14_0-usb-0_6_3_1_1-scsi-0_0_0_0  | /dev/ttyACM14 | 07960221012567073E61F1AC                         | 0221            |
| NUCLEO_F746ZG     | NUCLEO_F746ZG[0]     | /mnt/pci-0000_00_14_0-usb-0_13_4_1_1-scsi-0_0_0_0 | /dev/ttyACM27 | 08160221052A601C3A6EF937                         | 0221            |
| NUCLEO_F746ZG     | NUCLEO_F746ZG[1]     | /mnt/pci-0000_00_14_0-usb-0_13_2_1_1-scsi-0_0_0_0 | /dev/ttyACM1  | 08160221053A60193A7EF932                         | 0221            |
| NUCLEO_L152RE     | NUCLEO_L152RE[0]     | /mnt/pci-0000_00_14_0-usb-0_13_7_1_1-scsi-0_0_0_0 | /dev/ttyACM4  | 071002219B2164DEA465F2F3                         | 0221            |
| NUCLEO_L152RE     | NUCLEO_L152RE[1]     | /mnt/pci-0000_00_14_0-usb-0_13_6_1_1-scsi-0_0_0_0 | /dev/ttyACM34 | 071002211C54E3AB23107586                         | 0221            |
| UBLOX_EVK_ODIN_W2 | UBLOX_EVK_ODIN_W2[0] | /mnt/pci-0000_00_14_0-usb-0_1_4_1_1-scsi-0_0_0_0  | /dev/ttyACM24 | 123602210234683C3D70EB37                         | 0221            |
| UBLOX_EVK_ODIN_W2 | UBLOX_EVK_ODIN_W2[1] | /mnt/pci-0000_00_14_0-usb-0_13_1_1_1-scsi-0_0_0_0 | /dev/ttyACM19 | 12360221022268373D66EB3C                         | 0221            |
| UBLOX_EVK_ODIN_W2 | UBLOX_EVK_ODIN_W2[2] | /mnt/pci-0000_00_14_0-usb-0_8_1_1-scsi-0_0_0_0    | /dev/ttyACM11 | 12360221023468353D70EB3E                         | 0221            |

```

latest version of mbed-ls 1.3.3

```
+-------------------+----------------------+---------------------------------------------------+---------------+--------------------------------------------------+-----------------+
|   platform_name   | platform_name_unique |                    mount_point                    |  serial_port  |                    target_id                     | daplink_version |
+-------------------+----------------------+---------------------------------------------------+---------------+--------------------------------------------------+-----------------+
|      ARCH_PRO     |     ARCH_PRO[0]      |  /mnt/pci-0000_00_14_0-usb-0_1_5_1_0-scsi-0_0_0_0 |  /dev/ttyACM9 | 9004000008bf3ce200000000000000000000000097969902 |       0243      |
|      ARCH_PRO     |     ARCH_PRO[1]      |  /mnt/pci-0000_00_14_0-usb-0_1_7_1_0-scsi-0_0_0_0 | /dev/ttyACM15 | 9004000009be7aeb00000000000000000000000097969902 |       0243      |
|      ARCH_PRO     |     ARCH_PRO[2]      |  /mnt/pci-0000_00_14_0-usb-0_1_6_1_0-scsi-0_0_0_0 | /dev/ttyACM12 | 9004000009be1b6c00000000000000000000000097969902 |       0243      |
|        K22F       |       K22F[0]        | /mnt/pci-0000_00_14_0-usb-0_11_1_1_0-scsi-0_0_0_0 |  /dev/ttyACM3 | 0231000027944e45005030195e0e00214eb1000097969900 |       0243      |
|        K22F       |       K22F[1]        |  /mnt/pci-0000_00_14_0-usb-0_7_2_1_0-scsi-0_0_0_0 | /dev/ttyACM10 | 0231000033514e45005a500585d40039e981000097969900 |       0243      |
[... cut ...]
| UBLOX_EVK_ODIN_W2 | UBLOX_EVK_ODIN_W2[1] |   /mnt/pci-0000_00_14_0-usb-0_8_1_1-scsi-0_0_0_0  | /dev/ttyACM11 |             12360221023468353D70EB3E             |       0221      |
| UBLOX_EVK_ODIN_W2 | UBLOX_EVK_ODIN_W2[2] |  /mnt/pci-0000_00_14_0-usb-0_1_4_1_1-scsi-0_0_0_0 | /dev/ttyACM24 |             123602210234683C3D70EB37             |       0221      |
+-------------------+----------------------+---------------------------------------------------+---------------+--------------------------------------------------+-----------------+

```

bridadan, 2017-11-02:
@studavekar are you parsing the mbed-ls table output? You may have better luck by parsing `mbedls -j` or `mbedls -J` since that's just plain JSON

studavekar, 2017-11-02:
@bridadan no i am not parsing the out, Its just doesn't look right since previous version.

PR #262 (2017-11-10): Left align text in table
Fixes #261
closed 2017-11-10 by theotherjimmy