ARMmbed/mbed-ls #267: deprecated messages to remove

jeromecoutant, 2017-11-15:
Hi

My command:
mbed test -m auto -t arm -v

Each test log starts by:

[1510749208.90][mbedls.lstools_base]Call to deprecated function list_mbeds_by_targetid. List formatting methods are deprecated to simplify the API. Please use 'list_mbeds' instead.
[1510749208.90][mbedls.lstools_base]Call to deprecated function list_mbeds_ext. Functionality has been moved into 'list_mbeds'. Please use list_mbeds with 'unique_names=True' and 'read_details_txt=True'
[1510749209.13][root]Found htm target id, 076402216E3565725171F32B, for usb target id 066EFF323535474B43122727
[1510749215.33][mbedls.lstools_base]Call to deprecated function list_mbeds_by_targetid. List formatting methods are deprecated to simplify the API. Please use 'list_mbeds' instead.
[1510749215.33][mbedls.lstools_base]Call to deprecated function list_mbeds_ext. Functionality has been moved into 'list_mbeds'. Please use list_mbeds with 'unique_names=True' and 'read_details_txt=True'
[1510749215.57][root]Found htm target id, 076402216E3565725171F32B, for usb target id 066EFF323535474B43122727
[1510749216.10][mbedls.lstools_base]Call to deprecated function list_mbeds_by_targetid. List formatting methods are deprecated to simplify the API. Please use 'list_mbeds' instead.
[1510749216.10][mbedls.lstools_base]Call to deprecated function list_mbeds_ext. Functionality has been moved into 'list_mbeds'. Please use list_mbeds with 'unique_names=True' and 'read_details_txt=True'
[1510749216.38][root]Found htm target id, 076402216E3565725171F32B, for usb target id 066EFF323535474B43122727
[1510749216.89][mbedls.lstools_base]Call to deprecated function list_mbeds_by_targetid. List formatting methods are deprecated to simplify the API. Please use 'list_mbeds' instead.
[1510749216.89][mbedls.lstools_base]Call to deprecated function list_mbeds_ext. Functionality has been moved into 'list_mbeds'. Please use list_mbeds with 'unique_names=True' and 'read_details_txt=True'
[1510749217.14][root]Found htm target id, 076402216E3565725171F32B, for usb target id 066EFF323535474B43122727
[1510749217.66][mbedls.lstools_base]Call to deprecated function list_mbeds_by_targetid. List formatting methods are deprecated to simplify the API. Please use 'list_mbeds' instead.
[1510749217.66][mbedls.lstools_base]Call to deprecated function list_mbeds_ext. Functionality has been moved into 'list_mbeds'. Please use list_mbeds with 'unique_names=True' and 'read_details_txt=True'
[1510749217.92][root]Found htm target id, 076402216E3565725171F32B, for usb target id 066EFF323535474B43122727
[1510749218.44][mbedls.lstools_base]Call to deprecated function list_mbeds_by_targetid. List formatting methods are deprecated to simplify the API. Please use 'list_mbeds' instead.
[1510749218.44][mbedls.lstools_base]Call to deprecated function list_mbeds_ext. Functionality has been moved into 'list_mbeds'. Please use list_mbeds with 'unique_names=True' and 'read_details_txt=True'
[1510749218.70][root]Found htm target id, 076402216E3565725171F32B, for usb target id 066EFF323535474B43122727

Please remove the deprecated message and the duplicated messages.

Thx

theotherjimmy, 2017-11-27:
This is probably in htrun. We can track it here.

drewcassidy, 2018-01-23:
this is caused by htrun calling `list_mbeds_by_targetid()`, which is also deprecated

closed 2018-02-01 by theotherjimmy