ARMmbed/mbed-ls #270: Error from creating ~/.local/share/mbedls/platforms.json

RomanSaveljev, 2017-11-20:
I am working in a somewhat strange environment, where `/root` mountpoint does not have any free space. Writing `platforms.json` fails and results in a strange error message, where, in my opinion, the utility should have already given up before. It seems like the error from creating the file is ignored.

```
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/salt/loader.py", line 740, in grains
    ret = funcs[key]()
  File "/var/cache/salt/minion/extmods/grains/mbedls.py", line 4, in mbedls
    mbeds = create(skip_retarget=True)
  File "/usr/local/lib/python2.7/dist-packages/mbed_lstools/main.py", line 50, in create
    result = MbedLsToolsLinuxGeneric(**kwargs)
  File "/usr/local/lib/python2.7/dist-packages/mbed_lstools/linux.py", line 42, in __init__
    MbedLsToolsBase.__init__(self, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/mbed_lstools/lstools_base.py", line 83, in __init__
    primary_database=platform_dbs[0])
  File "/usr/local/lib/python2.7/dist-packages/mbed_lstools/platform_database.py", line 308, in __init__
    new_db = _overwrite_or_open(db)
  File "/usr/local/lib/python2.7/dist-packages/mbed_lstools/platform_database.py", line 286, in _overwrite_or_open
    with open(db, "w", encoding="utf-8") as out:
IOError: [Errno 2] No such file or directory: '/root/.local/share/mbedls/platforms.json'
```

```
$ mbedls --version
WARNING:mbedls.platform_database:Error loading database /home/pi/.local/share/mbedls/platforms.json: Platform Database is out of date; Recreating
1.3.4
```

theotherjimmy, 2017-11-21:
@RomanSaveljev I'm a bit confused. Is the traceback what happens today? What's the expected behavior? Does mbedls still work for you?

RomanSaveljev, 2017-11-21:
I think, expected behavior is a failover to printing me connected MBED boards without writing `platforms.json`.

theotherjimmy, 2017-11-21:
@RomanSaveljev I'm confused by your response, which of my questions did you answer?

RomanSaveljev, 2017-11-22:
> Is the traceback what happens today?
Yes. The traceback is generated by this code https://github.com/ARMmbed/sandbox/blob/salt/salt/states/_grains/mbedls.py, which is loaded as a plugin into another application

> What's the expected behavior?
I think, expected behavior is a failover to printing me connected MBED boards without writing `platforms.json`

> Does mbedls still work for you?
I can not use the `mbedls` via API.

RomanSaveljev, 2017-12-11:
Sorry for the silence. Yes, I can try it out

theotherjimmy, 2017-12-11:
Better late than never! Thanks for trying it out!

PR #273 (2017-12-18): Correct platform database behavior when platforms.json is not writable
Resolves #270
closed 2017-12-18 by theotherjimmy