ARMmbed/mbed-ls #277: `get_mbed_com_port` API is missing

jupe, 2017-12-08:
`get_mbed_com_port` is dropped in some point but mbed-flasher rely on it still. 
This need to bring back including regression test for it.

```
Instance of 'MbedLsToolsWin7' has no 'get_mbed_com_port' member
Instance of 'MbedLsToolsLinuxGeneric' has no 'get_mbed_com_port' member
Instance of 'MbedLsToolsDarwin' has no 'get_mbed_com_port' member
```

bridadan, 2017-12-08:
@jupe That was only ever and API in the Windows implementation of the mbed-ls backend. It really was never meant to be consumed, it was an internal API. You should be able to get that same information from the public API, is it possible to switch mbed-flasher to the supported API?

jupe, 2017-12-08:
What is supported API's ? Would be good to have documented them first :)

jupe, 2017-12-08:
but sure we could change mbed-flasher side as well. I hope that documentation and regression tests (especially API tests) would be in place sooner or later so that we doesn't ended up in this kind of case in future anymore. Is there some "as light API" as `get_mbed_com_port` was before ?

bridadan, 2017-12-08:
@jupe Totally agree 😄 It's absolutely on the list! As you can probably tell, @theotherjimmy and I have been in the process of cleaning up this repo since it previously had basically no testing or API boundaries.

To answer your question, the public API is basically just the one function that gives you all of the available data: https://github.com/ARMmbed/mbed-ls/blob/master/mbed_lstools/lstools_base.py#L110

However, as a general rule you can generally expect to call any function in [this class](https://github.com/ARMmbed/mbed-ls/blob/master/mbed_lstools/lstools_base.py#L53) that isn't marked as `deprecated` and doesn't start with an underscore (`_`).

I think you'll be especially interested in the `fs_interaction` and `platform_name_filters` options that we added there. It allows you to prevent reading the devices' filesystems to speed up the execution of mbed-ls. Hopefully that will allow to remove some of the custom checks y'all had to do in the past for speed reasons (not sure if y'all still do this though).

I think you'll be pretty happy with all of the testing that's been added (shoutout to @theotherjimmy for setting up all of the mocks) https://github.com/ARMmbed/mbed-ls/tree/master/test. We used to not test the OSX or Windows logic at all, so its a huge improvement. We're at 74% coverage and climbing! There's a lot of deprecated code that will be pulled out in the next major release that'll bump up that number 😄

jupe, 2017-12-08:
Is there some examples or documentation about these new options (`fs_interaction` and `platform_name_filters`) ? Those sounds exactly thing we could use for this 👍

bridadan, 2017-12-08:
@jupe Not yet, however it will definitely be coming in the future. In the meantime feel free to use the following as reference (it is documented in the code)

For `platform_name_filters`, you just provide an array of regular expressions that can be used to match the targets you care about. You can also leave it empty for no filtering.

(@theotherjimmy After looking at this, I'm thinking it may be better to provide a function hook that let's filter based on the entire mbed dictionary? That way you can filter based on any of the data points mbed-ls provides)

For `fs_interaction`, the options are:

`FSInteraction.BeforeFilter` - This is the default for mbed-ls. No filtering is done on the returned results until each of the detected target's filesystems have been read and added to the list of mbeds. This ensures that the target ID returned matches the target ID in the MBED.HTML file (the most accurate source).
`FSInteraction.AfterFilter` - This moves the filtering step to before the filesystems of the detected targets are read. This ensures that you still get all of the data points, however there is a chance the target ID is **incorrect** on some old firmware versions (this isn't a problem on all up to date firmware, including DAPLink and STLink).
`FSInteraction.Never` - This never reads from the target's filesytem, instead relying on the USB ID string for all of the info. This is extremely fast, however it comes at the cost of reduced data points and the potential for incorrect target IDs to be returned. If you're using up to date firmware, the target IDs should still be correct.

jupe, 2017-12-09:
Is there way to filter by target-id ? that was argument in original APi we uses. I think name is not exactly what we would like to use in this case. Isn’t  it returning e g all K64F’s and then we have to filter again based on target-id?

bridadan, 2017-12-11:
@jupe agreed, I think we need to augment the filter functionality a bit

jupe, 2017-12-20:
This becomes obsolete - at least from mbed-flasher point of view. I just [replaced](https://github.com/ARMmbed/mbed-flasher/pull/80/commits/0b3bfe34f8025aba58786d60354e3b2fedf9dabd) old API with this filtering. Hopefully it works as quickly as original API.

bridadan, 2017-12-22:
@jupe If that's the case can you go ahead and close this issue?

closed 2017-12-22 by jupe