ARMmbed/mbed-ls #279: mbedls detects board that is not there

cederom, 2017-12-11:
Hello, we have discovered this problem on Win7 (VirtualBox), mbedls detects board NRF52_DK that is not there, probably just because there is a network device mapped to a drive that was previously detected as a board.. details attached below.. looks to happen in version 1.2.14 but also new 1.3.4.. nothing is connected to the VirtualMachine nor to the computer.. there are no VBox USB filters:

```
$ uname -a
MINGW64_NT-6.1 VM 2.8.2(0.313/5/3) 2017-07-12 15:35 x86_64 Msys

$ mbedls                                                                                                                                             
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+            
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |            
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+            
| NRF52_DK      | NRF52_DK[0]          | E:          | COM4        | 110100004420312033504139313031313233303097969903 | unknown         |            
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+            
                                                                                                                                                                                                                                                                                       
$ mbedls -d                                                                                                                                          
debug @MbedLsToolsWin7.mbed_lstools.main: mbed-ls ver. 1.2.14                                                                                        
debug @MbedLsToolsWin7.mbed_lstools.main: host: ('nt', 'Windows', '7', '6.1.7601', 'win32')                                                          
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\E:', '_??_USBSTOR#Disk&Ven_MBED&Prod_VFS&Rev_0.1#1101000044203120335041393130313132333030979
69903&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')                                                                                                     
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\G:', '_??_USBSTOR#Disk&Ven_MBED&Prod_VFS&Rev_0.1#10500000e062b0bf000000000000000000000000979
69902&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')                                                                                                     
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\H:', '_??_USBSTOR#Disk&Ven_MBED&Prod_VFS&Rev_0.1#0240000028884e45001070116bf0004780210000979
69900&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')                                                                                                     
debug @MbedLsToolsWin7.get_mbeds: ('E:', '110100004420312033504139313031313233303097969903')                                                         
debug @MbedLsToolsWin7.get_mbeds: ('G:', '10500000e062b0bf00000000000000000000000097969902')                                                         
debug @MbedLsToolsWin7.get_mbeds: ('H:', '0240000028884e45001070116bf000478021000097969900')                                                         
debug @MbedLsToolsWin7.mount_point_ready: Mount point E: is ready                                                                                    
debug @MbedLsToolsWin7.mount_point_ready: Mount point G: reported not ready with error 'The system cannot find the path specified.'                  
debug @MbedLsToolsWin7.mount_point_ready: Mount point H: reported not ready with error 'The system cannot find the path specified.'                  
debug @MbedLsToolsWin7.get_mbed_com_port: ID: 110100004420312033504139313031313233303097969903                                                       
debug @MbedLsToolsWin7.get_mbed_com_port: COM4                                                                                                       
debug @MbedLsToolsWin7.list_mbeds_ext: ('NRF52_DK[0]', '110100004420312033504139313031313233303097969903')                                           
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+            
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |            
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+            
| NRF52_DK      | NRF52_DK[0]          | E:          | COM4        | 110100004420312033504139313031313233303097969903 | unknown         |            
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+            
debug @MbedLsToolsWin7.mbed_lstools.main: Return code: -1                                                                                            
                                                                                                                                                     
$ mbedls --version                                                                                                                                   
1.2.14                                                                                                                                               

                                                                                                                                                                                                                                                  
$ source venv-win-27/Scripts/activate                                                                                                                
(venv-win-27)                                                                                                                                        

$ which python                                                                                                                                       
/T/cederom/venv-win-27/Scripts/python                                                                                                                
(venv-win-27)                                                                                                                                        

$ which mbedls                                                                                                                                       
/T/cederom/venv-win-27/Scripts/mbedls                                                                                                                
(venv-win-27)                                                                                                                                        

$ mbedls --version                                                                                                                                   
1.3.4                                                                                                                                                
(venv-win-27)                                                                                                                                        

$ mbedls                                                                                                                                             
WARNING:root:Could not read htm on from usb id 110100004420312033504139313031313233303097969903. Falling back to usb id                              
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+            
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |            
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+            
[... cut ...]
(venv-win-27)                                                                                                                                        

$ ls /e                                                                                                                                              
doc/  lab/                                                                                                                                           
(venv-win-27)

$ uname -a
MINGW64_NT-6.1 VM 2.8.2(0.313/5/3) 2017-07-12 15:35 x86_64 Msys
(venv-win-27)
```

cederom, 2017-12-11:
After rebooting the VBox I get:

```
user@VM MINGW64 ~
$ mbedls
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| LPC812        | LPC812[0]            | G:          | COM5        | 10500000e062b0bf00000000000000000000000097969902 | unknown         |
| NRF52_DK      | NRF52_DK[0]          | E:          | COM4        | 110100004420312033504139313031313233303097969903 | unknown         |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+

user@VM MINGW64 ~
$ mbedls -d
debug @MbedLsToolsWin7.mbed_lstools.main: mbed-ls ver. 1.2.14
debug @MbedLsToolsWin7.mbed_lstools.main: host: ('nt', 'Windows', '7', '6.1.7601', 'win32')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\E:', '_??_USBSTOR#Disk&Ven_MBED&Prod_VFS&Rev_0.1#110100004420312033504139313031313233303097969903&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\G:', '_??_USBSTOR#Disk&Ven_MBED&Prod_VFS&Rev_0.1#10500000e062b0bf00000000000000000000000097969902&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbed_devices: ('\\DosDevices\\H:', '_??_USBSTOR#Disk&Ven_MBED&Prod_VFS&Rev_0.1#0240000028884e45001070116bf000478021000097969900&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}')
debug @MbedLsToolsWin7.get_mbeds: ('E:', '110100004420312033504139313031313233303097969903')
debug @MbedLsToolsWin7.get_mbeds: ('G:', '10500000e062b0bf00000000000000000000000097969902')
debug @MbedLsToolsWin7.get_mbeds: ('H:', '0240000028884e45001070116bf000478021000097969900')
debug @MbedLsToolsWin7.mount_point_ready: Mount point E: is ready
debug @MbedLsToolsWin7.mount_point_ready: Mount point G: is ready
debug @MbedLsToolsWin7.mount_point_ready: Mount point H: reported not ready with error 'The system cannot find the path specified.'
debug @MbedLsToolsWin7.get_mbed_com_port: ID: 110100004420312033504139313031313233303097969903
debug @MbedLsToolsWin7.get_mbed_com_port: COM4
debug @MbedLsToolsWin7.get_mbed_com_port: ID: 10500000e062b0bf00000000000000000000000097969902
debug @MbedLsToolsWin7.get_mbed_com_port: COM5
debug @MbedLsToolsWin7.list_mbeds_ext: ('NRF52_DK[0]', '110100004420312033504139313031313233303097969903')
debug @MbedLsToolsWin7.list_mbeds_ext: ('LPC812[0]', '10500000e062b0bf00000000000000000000000097969902')
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| LPC812        | LPC812[0]            | G:          | COM5        | 10500000e062b0bf00000000000000000000000097969902 | unknown         |
| NRF52_DK      | NRF52_DK[0]          | E:          | COM4        | 110100004420312033504139313031313233303097969903 | unknown         |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
debug @MbedLsToolsWin7.mbed_lstools.main: Return code: -1

user@VM MINGW64 ~
$ source /t/cederom/venv-win-27/Scripts/activate
(venv-win-27)
user@VM MINGW64 ~
$ source /t/cederom/venv-win-27/Scripts/activate
(venv-win-27)
user@VM MINGW64 ~
$ mbedls --version
1.3.4
(venv-win-27)
user@VM MINGW64 ~
$ mbedls
WARNING:root:Could not read htm on from usb id 110100004420312033504139313031313233303097969903. Falling back to usb id
WARNING:root:Could not read htm on from usb id 10500000e062b0bf00000000000000000000000097969902. Falling back to usb id
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| LPC812        | LPC812[0]            | G:          | COM5        | 10500000e062b0bf00000000000000000000000097969902 | unknown         |
| NRF52_DK      | NRF52_DK[0]          | E:          | COM4        | 110100004420312033504139313031313233303097969903 | unknown         |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
(venv-win-27)
user@VM MINGW64 ~
$ mbedls -d
DEBUG:mbedls.main:mbed-ls ver. 1.3.4
DEBUG:mbedls.main:host: ('nt', 'Windows', '7', '6.1.7601', 'win32')
DEBUG:mbedls.lstools_win7:iter_vals <PyHKEY object at 0x02FEE6F0>
DEBUG:mbedls.lstools_win7:Skipping Mount point '\\DosDevices\\C:' label '\xb6B_Y\x00\x00P\x06\x00\x00\x00\x00'
DEBUG:mbedls.lstools_win7:Skipping Mount point '\\??\\Volume{0c496021-95b3-11e6-a723-806e6f6e6963}' label '\xb6B_Y\x00\x00\x10\x00\x00\x00\x00\x00'
[... cut ...]
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
DEBUG:mbedls.main:Return code: 0
(venv-win-27)
user@VM MINGW64 ~
$ ls /e
doc/  lab/
(venv-win-27)
user@VM MINGW64 ~
$ ls /g
att_mc8355_driver_v3717_win_xp_7_8_firmware.zip  cederom/  DAPLink/  mbed/  WM/  WWANSoft_update_R4G6606.exe*  wwansoft_v100l10m00_w7xp_d111773.exe* (venv-win-27)
user@VM MINGW64 ~
$ uname -a
MINGW64_NT-6.1 VM 2.8.2(0.313/5/3) 2017-07-12 15:35 x86_64 Msys
(venv-win-27)
user@VM MINGW64 ~
$
```

cederom, 2017-12-11:
No change after host computer reboot.. I think it is somehow related to network drives attached.. when I remove the network drive mbelds does not see the non-existing device..

bridadan, 2017-12-11:
Hi @cederom, it looks like you're using MINGW? Could you please try it in just a normal Windows command prompt in your VM and see if you still have the problem?

bridadan, 2017-12-11:
We determined offline that it was indeed the network drives being shows as mbed devices. This is a symptom of how we currently detect mbed devices on Windows. When a device is removed (USB unplugged) the Windows Registry does not go away. The real way to fix this would be to use a key that DID get removed when the physical device is removed.

...and I found some keys! I'll try to get a PR up soon with the implementation, it shouldn't be too much different from the current one.

cederom, 2017-12-12:
Thank you @bridadan for such a rapid response! You are the best! =)

bridadan, 2017-12-13:
@cederom I have a branch that should fix this problem since it's using the new detection keys. I have to clean it up a bit and add tests before I submit a PR. But maybe you can give it a try first? Here's the branch: https://github.com/bridadan/mbed-ls/tree/windows_detection_update

cederom, 2017-12-13:
Sure! Will do and report back! :-)

--
CeDeROM, SQ7MHZ, http://www.tomek.cedro.info

cederom, 2017-12-14:
Here is the log of updated detection mechanism :-)
[out.txt](https://github.com/ARMmbed/mbed-ls/files/1560138/out.txt)

cederom, 2017-12-14:
..and after some drivers installation..
[out2.txt](https://github.com/ARMmbed/mbed-ls/files/1560185/out2.txt)

cederom, 2017-12-14:
Seems to fix the problem! TANK U BRIAN! YOU ROX!! =)

closed 2018-03-01 by theotherjimmy