ARMmbed/mbed-ls #283: Detect boards with JLink firmware

sg-, 2017-12-18:
Some boards need to use JLink firmware (only one UART block) and the UART is used by the application. Has anyone investigated if JLink follows a standard pattern for identifying the board type?

@marhil01

bridadan, 2017-12-19:
I can speak at least for Windows this should be quite doable, the infrastructure is there to detect the HID endpoint we just currently aren't doing anything with it.

For Linux I think we need to do a little more investigation but I would imagine it shouldn't be too difficult. Same goes for Mac OS.

What information do you need in order to utilize the JLink/HID/Debug endpoint?

jupe, 2018-01-05:
We need at least platform name (seems that mbedls gives incorrectly this), target id (it gives something already, is it correct one?), mount point (probably ok already) and serial port. Basic things at least. Would be nice to have also jlink version etc but we can start with basic information. And support for linux is needed in our case.

bridadan, 2018-01-05:
> We need at least platform name (seems that mbedls gives incorrectly this), target id (it gives something already, is it correct one?)

Can you provide some debug output please (`mbedls --debug`)? We're working blind at this point.

Also please provide the normal info: target name, JLink version, Host OS platform, etc. These are all necessary so we can fix bugs/add features!

juhhov, 2018-01-08:
target: nRF52
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie
$ mbedls --version
1.3.5
$ mbedls -dj
DEBUG:mbedls.main:mbed-ls ver. 1.3.5
DEBUG:mbedls.main:host: ('posix', 'Linux', '4.4.26-v7+', '#915 SMP Thu Oct 20 17:08:44 BST 2016', 'linux2')
DEBUG:mbedls.lstools_linux:Converting device list ['/dev/disk/by-id/usb-SEGGER_MSD_Volume_000682245000-0:0', '/dev/disk/by-id/mmc-SD32G_0x0028012d-part2', '/dev/disk/by-id/mmc-SD32G_0x0028012d-part1', '/dev/disk/by-id/mmc-SD32G_0x0028012d']
DEBUG:mbedls.lstools_linux:Found disk devices by id {'000682245000': '/dev/sda'}
DEBUG:mbedls.lstools_linux:Converting device list ['/dev/serial/by-id/usb-SEGGER_J-Link_000682245000-if00']
DEBUG:mbedls.lstools_linux:Found serial devices by id {'000682245000': '/dev/ttyACM0'}
DEBUG:mbedls.lstools_linux:Mount mapping {u'/dev/mmcblk0p1': u'/boot', u'/dev/sdb': u'/mnt/platform-3f980000_usb-usb-0_1_2_3_1_0-scsi-0_0_0_0', u'/dev/sda': u'/mnt/platform-3f980000_usb-usb-0_1_2_1_1_3-scsi-0_0_0_0'}
DEBUG:mbedls.lstools_base:Candidates for display [{'mount_point': u'/mnt/platform-3f980000_usb-usb-0_1_2_1_1_3-scsi-0_0_0_0', 'serial_port': '/dev/ttyACM0', 'target_id_usb_id': '000682245000'}]
WARNING:root:Could not read htm on from usb id 000682245000. Falling back to usb id
DEBUG:mbedls.lstools_base:Failed to open file /mnt/platform-3f980000_usb-usb-0_1_2_1_1_3-scsi-0_0_0_0/DETAILS.TXT: [Errno 2] No such file or directory: '/mnt/platform-3f980000_usb-usb-0_1_2_1_1_3-scsi-0_0_0_0/DETAILS.TXT'
[
    {
        "daplink_url": "http://www.mbed.com/", 
        "mount_point": "/mnt/platform-3f980000_usb-usb-0_1_2_1_1_3-scsi-0_0_0_0", 
        "platform_name": "LPC2368", 
        "platform_name_unique": "LPC2368[0]", 
        "serial_port": "/dev/ttyACM0", 
        "target_id": "000682245000", 
        "target_id_mbed_htm": null, 
        "target_id_usb_id": "000682245000"
    }
]
DEBUG:mbedls.main:Return code: 0
```
also:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.2 LTS
Release:	16.04
Codename:	xenial
$ mbedls --version
1.3.5
$ mbedls -dj
DEBUG:mbedls.main:mbed-ls ver. 1.3.5
DEBUG:mbedls.main:host: ('posix', 'Linux', '4.4.0-97-generic', '#120-Ubuntu SMP Tue Sep 19 17:28:18 UTC 2017', 'linux2')
DEBUG:mbedls.lstools_linux:Converting device list ['/dev/disk/by-id/usb-SEGGER_MSD_Volume_000682245000-0:0', '/dev/disk/by-id/usb-Generic-_MS_MS-Pro_058F63626420-0:3', '/dev/disk/by-id/usb-Generic-_SM_xD_Picture_058F63626420-0:2', '/dev/disk/by-id/usb-Generic-_Compact_Flash_058F63626420-0:1', '/dev/disk/by-id/usb-Generic-_SD_MMC_058F63626420-0:0', '/dev/disk/by-id/wwn-0x5001480000000000', '/dev/disk/by-id/ata-HL-DT-ST_DVD+-RW_GT60N_KZYC5MM3148', '/dev/disk/by-id/wwn-0x5002538d412274bd-part1', '/dev/disk/by-id/ata-Samsung_SSD_850_EVO_250GB_S2R6NX0H808412F-part1', '/dev/disk/by-id/wwn-0x5002538d412274bd-part5', '/dev/disk/by-id/ata-Samsung_SSD_850_EVO_250GB_S2R6NX0H808412F-part5', '/dev/disk/by-id/wwn-0x5002538d412274bd-part2', '/dev/disk/by-id/ata-Samsung_SSD_850_EVO_250GB_S2R6NX0H808412F-part2', '/dev/disk/by-id/wwn-0x5002538d412274bd', '/dev/disk/by-id/ata-Samsung_SSD_850_EVO_250GB_S2R6NX0H808412F']
DEBUG:mbedls.lstools_linux:Found disk devices by id {'000682245000': '/dev/sdb', '058F63626420': '/dev/sdc'}
DEBUG:mbedls.lstools_linux:Converting device list ['/dev/serial/by-id/usb-SEGGER_J-Link_000682245000-if00']
DEBUG:mbedls.lstools_linux:Found serial devices by id {'000682245000': '/dev/ttyACM0'}
DEBUG:mbedls.lstools_linux:Mount mapping {u'/dev/sdb': u'/mnt/pci-0000_00_1d_0-usb-0_1_6_1_3-scsi-0_0_0_0'}
DEBUG:mbedls.lstools_base:Candidates for display [{'mount_point': u'/mnt/pci-0000_00_1d_0-usb-0_1_6_1_3-scsi-0_0_0_0', 'serial_port': '/dev/ttyACM0', 'target_id_usb_id': '000682245000'}, {'mount_point': None, 'serial_port': None, 'target_id_usb_id': '058F63626420'}]
WARNING:root:Could not read htm on from usb id 000682245000. Falling back to usb id
DEBUG:mbedls.lstools_base:Failed to open file /mnt/pci-0000_00_1d_0-usb-0_1_6_1_3-scsi-0_0_0_0/DETAILS.TXT: [Errno 2] No such file or directory: '/mnt/pci-0000_00_1d_0-usb-0_1_6_1_3-scsi-0_0_0_0/DETAILS.TXT'
[
    {
        "daplink_url": "http://www.mbed.com/", 
        "mount_point": "/mnt/pci-0000_00_1d_0-usb-0_1_6_1_3-scsi-0_0_0_0", 
        "platform_name": "LPC2368", 
        "platform_name_unique": "LPC2368[0]", 
        "serial_port": "/dev/ttyACM0", 
        "target_id": "000682245000", 
        "target_id_mbed_htm": null, 
        "target_id_usb_id": "000682245000"
    }
]
DEBUG:mbedls.main:Return code: 0
```

How to find out JLink version?

bridadan, 2018-01-08:
Thanks for posting that, it helps a lot! I was also able to get an NRF52 flashed with JLink here. I don't have it communicating over SWO yet.

However in terms of detecting as an NRF52 board, it _looks_ like it might be possible. On JLink devices it always looks like there's a `Segger.html` file to indicate that its running JLink firmware. For the NRF52 board, there's also a `User Guide.html` file that contains the following link: http://www.nordicsemi.com/X349858SLYN. That code at the end of the URL takes you to a user manual specifically for the nRF52832 dev board, so it looks to be unique.

Also the target ID that was picked up for me was 000682961642, so `0006` might correspond to NRF52 with JLink? I'd have to look at other boards to be confident about that.

The downside is we'd have to now switch our detection method based on what files are present on the device, which kind of stinks. @theotherjimmy what do you think about this?

theotherjimmy, 2018-01-08:
@bridadan I'm okay with having the _acurate_ results try to search for 2 (4 actually) files on the disk. This would degrade performance for the average use case and allow CI to have a more informed filter function that can do filtering first.

We should make sure that these changes do not impact the performance of the CI useage where QOS is set to `FsInteraciton.Never`.

bridadan, 2018-01-08:
@theotherjimmy agreed!

juhhov, 2018-01-11:
@bridadan @theotherjimmy would it be possible to add "firmware"-key (or something similar) to the mbedls json results? It should have value "daplink" or "jlink" depending on the firmware of the board.

bridadan, 2018-01-11:
@juhhov If we go with the implementation discussed above then we can definitely add something like that!

juhhov, 2018-01-19:
Any updates on this?

bridadan, 2018-01-21:
Not from my side. I've been out of the office this past week and I'll be traveling this coming week. It sounds like @theotherjimmy are relatively comfortable with this change, but we need to spend a little time planning it and actually implementing it.

Is there a timeline for this change? Or maybe some internal tickets? Just so we know how to prioritize this.

jupe, 2018-01-21:
IOTMORF-1858

theotherjimmy, 2018-03-01:
Done by #299 and released as part of 1.4

closed 2018-03-01 by theotherjimmy