ARMmbed/mbed-ls #285: Logging uses root logger

jonikula, 2018-01-02:
In mbed_lstools/lstools_base.py logging is done using direct calls to logging.debug/logging.warning etc. instead of the modules own logger. This causes the global root logger to get a debug level StreamHandler which causes duplicate logging from other loggers (since all loggers are children of the root logger).

jonikula, 2018-01-02:
ping @jupe

bridadan, 2018-01-02:
Hi @jonikula, I was going through the code and I only found a few (mistaken) calls to the global logging modules (all of them are in this file: https://github.com/ARMmbed/mbed-ls/blob/ffbcec83c01eb82cb32cb3fec4f1fe415884f64a/mbed_lstools/lstools_base.py#L211). All other logs seem to be getting their own logger.

The messages I would expect to be duplicated would be "Found htm target id, %s, for usb target id %s" and "Could not read htm on from usb id %s. Falling back to usb id". Can you confirm this is the behavior you're seeing? If you're seeing other messages, can you be sure you're on the latest version of mbed-ls before trying again?

jonikula, 2018-01-03:
Logging modules root logger is the parent of all other loggers. This means that if some other application, that happens to use mbed-ls apis for something, creates a logger using logging.getLogger("my_logger") and calls my_logger.debug("a") that call get's propagated to the root logger as well. These calls to logging.debug(...) result in a streamhandler getting attached to the root logger. These two things together result in all logging getting duplicated as all messages are propagated to the root logger and to it's streamhandler. It should be easy to confirm with a small script.

bridadan, 2018-01-03:
Thanks for the explanation @jonikula, I'll get a patch up today!

PR #286 (2018-01-03): Remove log calls to root logger
Fixes #285.

There were a few stray calls to the root logger. This caused complications when used with other modules that make use of the `logging` library. They are removed in this patch.

cc @theotherjimmy @jonikula
closed 2018-01-03 by theotherjimmy