ARMmbed/mbed-ls #302: No logging configuration if create used

jonikula, 2018-02-05:
If the create-method in mbed_lstools main is used to create the toolset, the logging module never receives any configuration so nothing ever gets logged. Is it assumed that the application calling create would configure the logging itself or should the start_logging  method be automatically called in create?

theotherjimmy, 2018-02-05:
@jonikula Really good question. I think that we should call `start_logging` in create, only if no logger currently is running. You could override the settings in your application that way, which is probably what we want.

theotherjimmy, 2018-02-05:
@jonikula If there is no crash caused by forgetting to setup logging, then I see no harm to doing nothing, like we do now.

jonikula, 2018-02-06:
Ok. We'll setup some configurations in the application then. Thanks.

jupe, 2018-02-06:
There should be at least documentation for this - actually I propose to add NullHandler for logger when create() -api is called without previous `start_logging()` call by application then this should never happen anymore. It should not be application responsibility to configure logger for mbed-ls.

jupe, 2018-02-07:
In fact there should be also API to give custom logger instead of defaults ones.

theotherjimmy, 2018-02-13:
@jupe That's the python `logging` module. I don't think that adding a duplicate API for mbed-ls is a good idea.

jupe, 2018-02-13:
I mean custom logger instance instead of one that mbedls create, which might not be suitable for each use cases.

theotherjimmy, 2018-02-13:
@jupe The Logger object itself does not have any policy decisions, and is a facade that forwards logging messages to the underlying Filter and Handler objects. It's possible to assign the Filter and Handler objects through the logging module outside of mbed-ls. As such, I don't think that there is a need to be able to replace the Logger objects (there's like 6 of them) that mbed-ls uses.

jupe, 2018-02-13:
Please document mbed-ls used logger at least so that its clear which one user can use and modify when it is needed. Browsing source code and guess logger name is not an option. Also that default ”NullHandler” would be good to have.

theotherjimmy, 2018-02-13:
@jupe Can document! I'll make sure that the logger name is clear in the documentation. I'll add the "NullHandler" default in the same PR.

theotherjimmy, 2018-02-28:
Alright, I made the default-to-null-logger PR.

theotherjimmy, 2018-02-28:
Documentation update in #312.

PR #311 (2018-02-28): Default to NullHandler when logging not configured
Resolves #302
PR #312 (2018-03-01): Document logging behavior
Resolves #302 (Docmumentation bits)

@AnotherButler Could you edit?
closed 2018-02-28 by theotherjimmy