ARMmbed/mbed-ls #313: Uninstall doesn't remove platforms.json

juhhov, 2018-03-01:
If user wants to downgrade mbed-ls to earlier release platforms.json must be manually deleted since the newer version might not be backwards compatilble. This happens at least when downgrading from 1.4 to 1.3.7.

Pip uninstall should remove platforms.json.

bridadan, 2018-03-01:
@theotherjimmy This normally should occur, however I think previously the "replace if older" logic was only using `date` instead of `datetime`. That changed in version 1.4.0.

Sounds like we could force platforms.json to be removed on uninstall though for these newer versions.

Might be able to add an `uninstall` entry to the section mentioned here: https://stackoverflow.com/questions/20288711/post-install-script-with-python-setuptools

jupe, 2018-03-01:
Why this file is needed in first place at the moment? is it related to some future ideas?

theotherjimmy, 2018-03-01:
@jupe Yes. That file will allow us to update the platform db without updating mbed-ls. That will allow us to update the platform db from the website, and stop these "add my platform to the db" PRs.