ARMmbed/mbed-ls #319: list_mbeds() raises WindowsError [Error 21] The device is not ready: 'H:*.*'

jupe, 2018-04-03:
OS: Windows 10
mbed-flasher: 0.7.1
mbed-ls version: 1.4.0

mbed-ls should catch this exception and ignore "not ready device".
Seen couple times in CI. 

Traceback:
```
  File \"x:\\app\\src\\mbed-flasher\\mbed_flasher\\flashers\\FlasherMbed.py\", line 75, in get_available_devices
    return mbeds.list_mbeds()
  File \"x:\\app\\lib\\site-packages\\mbed_lstools\\lstools_base.py\", line 146, in list_mbeds
    }[fs_interaction](device, filter_function, read_details_txt)
  File \"x:\\app\\lib\\site-packages\\mbed_lstools\\lstools_base.py\", line 183, in _fs_before_id_check
    self._update_device_from_fs(device, read_details_txt)
  File \"x:\\app\\lib\\site-packages\\mbed_lstools\\lstools_base.py\", line 213, in _update_device_from_fs
    directory_entries = os.listdir(device['mount_point'])
WindowsError: [Error 21] The device is not ready: 'H:*.*'
```

bridadan, 2018-04-03:
You're right, mbed-ls should deal with this exception correctly. In this case, the device would be classified as **unmounted**.

This is interesting, because there is a very particular sequence of events that must occur for this to happen:

1. Device is mounted and mbed-ls API is called
2. mbed-ls sees the device in the Windows Registry and lists it as a candidate
3. mbed-ls checks to see if it's mounted (in this case, it was mounted)
4. If it is mounted, mbed-ls reads the files on the device

In the traceback you posted above, the device would have had to dismount after step 3 but before step 4!

Is this something that you would expect to happen?

Either way, we'll get a fix up for this.

jupe, 2018-04-03:
> Is this something that you would expect to happen?

kind of not, but this kind of corner cases happens in CI because mbed-ls is called so much there over the time :)

PR #320 (2018-04-03): Fix unmounted traceback
This should fix #319.

@theotherjimmy I've wrapped the whole block in a `try`/`except` since each function can access the device files. Let me know if you want to go about it a different way. I also modified `list_mbeds` a bit so devices that unmount mid-read are treated like unmounted devices.

cc @jupe
closed 2018-04-03 by theotherjimmy