ARMmbed/mbed-ls #327: mbed detect:  No mbed targets were detected

Soheilica, 2018-04-13:
Hi,

Im running the MbedOS in Ubuntu Xenial. The` mbed detect` results in: [mbed] No mbed targets were detected on your system. I have connected the nucleo_f767zi to usb and have it in `lsusb`:

> vagrant@mbed-dev:~/All/All-blinks$ lsusb
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> Bus 002 Device 003: ID 0483:374b STMicroelectronics ST-LINK/V2.1 (Nucleo-F103RB)

` mbedls --version` is 1.4.2.

Here is the mbedls debug log:

> vagrant@mbed-dev:~/All/All-blinks$ mbedls --debug
> DEBUG:mbedls.main:mbed-ls ver. 1.4.2
> DEBUG:mbedls.main:host: ('posix', 'Linux', '4.4.0-119-generic', '#143-Ubuntu SMP Mon Apr 2 16:08:24 UTC 2018', 'linux2')
> DEBUG:mbedls.lstools_linux:Converting device list ['/dev/disk/by-id/usb-MBED_microcontroller_066FFF544852707267191640-0:0']
> DEBUG:mbedls.lstools_linux:Found disk devices by id {'066FFF544852707267191640': '/dev/sdc'}
> ERROR:mbedls.lstools_linux:Could not get serial devices by id. This could be because your Linux distribution does not use udev, or does not create /dev/serial/by-id symlinks. Please submit an issue to github.com/armmbed/mbed-ls.
> DEBUG:mbedls.lstools_linux:Mount mapping {}
> DEBUG:mbedls.lstools_base:Candidates for display [{'mount_point': None, 'serial_port': None, 'target_id_usb_id': '066FFF544852707267191640'}]
> DEBUG:mbedls.main:Return code: 0
> vagrant@mbed-dev:~/All/All-blinks$

Any input is appreciated!

theotherjimmy, 2018-04-18:
@Soheilica This seems related to #325 and that `mbed detect` requires the target to be mounted. It looks like your disk is not mounted, or your mounted disk mapping is going horribly wrong:
```
DEBUG:mbedls.lstools_linux:Mount mapping {}
```

Soheilica, 2018-04-19:
Thank you @theotherjimmy 

I built the mbedls from source... and came to here:

```
vagrant@mbed-dev:~$ mbedls
WARNING:mbedls.lstools_base:MBED with target id '066FFF544852707267191640' is connected, but not mounted. Use the '-u' flag to include it in the list.
```

How can I mount this device ID with '-u' flag?

bridadan, 2018-04-19:
Hi @Soheilica,

By default we don't show unmounted devices. The `-u` flag doesn't mount the device, but it will show information about the target that's available from the USB data provided by the OS.

If you want to mount the device, you could install an automounter or manually call the `mount` command from a shell.

0xc0170, 2018-06-12:
> Hi @Soheilica, can you let us know if this is still an issue? Otherwise feel free to close this. Thanks!

Closing, should be resolved.

closed 2018-06-12 by 0xc0170