ARMmbed/mbed-ls #332: mbedls v1.4.3 doesn't detect connected targets

toyowata, 2018-04-25:
mbedls v1.4.3 does not show connected targets.

Host: Widows 7 64-bit (Japanese)

```
PS C:\test\usb\mbed-os> mbedls  --version
1.4.2
PS C:\test\usb\mbed-os> mbedls
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| K64F          | K64F[0]              | E:          | COM15       | 0240000028884e450057700f6bf000108021000097969900 | 0246            |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
PS C:\test\usb\mbed-os>
```

```
PS C:\test\usb\mbed-os> mbedls --version
WARNING:mbedls.platform_database:Error loading database C:\Users\toywat01\AppData\Local\mbedls\mbedls\platforms.json: Platform Database is out of date; Recreating
1.4.3
PS C:\test\usb\mbed-os> mbedls
PS C:\test\usb\mbed-os> 
PS C:\test\usb\mbed-os> mbedls --debug
DEBUG:mbedls.main:mbed-ls ver. 1.4.3
DEBUG:mbedls.main:host: ('nt', 'Windows', '7', '6.1.7601', 'win32')
DEBUG:mbedls.lstools_win7:_iter_vals <PyHKEY object at 0x02D64B30>
DEBUG:mbedls.lstools_win7:Mount point E: found for volume _??_USBSTOR#Disk&Ven_MBED&Prod_VFS&Rev_0.1#0240000028884e450057700f6bf000108021000097969900&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}
DEBUG:mbedls.lstools_win7:Fetching mounted devices from volume service registry entry
DEBUG:mbedls.lstools_win7:No volumes service found, no device can be detected
DEBUG:mbedls.lstools_win7:Fetching usb storage devices from USBSTOR service registry entry
DEBUG:mbedls.lstools_win7:_iter_vals <PyHKEY object at 0x02D64B30>
DEBUG:mbedls.lstools_win7:target_id_usb_id -> mount_point mapping: {}
DEBUG:mbedls.lstools_base:Candidates for display []
DEBUG:mbedls.main:Return code: 0
PS C:\test\usb\mbed-os>

```

@bridadan @theotherjimmy

bridadan, 2018-04-25:
I'm wondering if this is an issue specific to Windows 7 OR specifically a Japanese language issue (it's possible I missed a unicode decode somewhere). I'm going to try and get a Windows 7 VM running today to test.

toyowata, 2018-04-25:
I tested this on a Japanese Windows 10 as below:

```
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

PS C:\Users\toyom> mbed --version
1.5.0
PS C:\Users\toyom> mbedls
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| K64F          | K64F[0]              | D:          | COM22       | 0240000028884e450024700f6bf000408021000097969900 | 0243            |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
PS C:\Users\toyom> mbedls --version
1.4.0
PS C:\Users\toyom> pip install -U mbed-ls
Collecting mbed-ls
  Downloading https://files.pythonhosted.org/packages/11/18/86aaad218d3eec994fc3e7159caaed2fedc80cd58e155d64661cbd06e606/mbed-ls-1.4.3.tar.gz (48kB)
    100% |################################| 51kB 197kB/s
Requirement already up-to-date: PrettyTable>=0.7.2 in c:\python27\lib\site-packages (from mbed-ls)
Requirement already up-to-date: fasteners in c:\python27\lib\site-packages (from mbed-ls)
Requirement already up-to-date: appdirs>=1.4 in c:\python27\lib\site-packages (from mbed-ls)
Requirement already up-to-date: monotonic>=0.1 in c:\python27\lib\site-packages (from fasteners->mbed-ls)
Requirement already up-to-date: six in c:\python27\lib\site-packages (from fasteners->mbed-ls)
Installing collected packages: mbed-ls
  Found existing installation: mbed-ls 1.4.0
    Uninstalling mbed-ls-1.4.0:
      Successfully uninstalled mbed-ls-1.4.0
  Running setup.py install for mbed-ls ... done
Successfully installed mbed-ls-1.4.3
You are using pip version 9.0.1, however version 10.0.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.
PS C:\Users\toyom> mbedls
WARNING:mbedls.platform_database:Error loading database C:\Users\toyom\AppData\Local\mbedls\mbedls\platforms.json: Platform Database is out of date; Recreating
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| K64F          | K64F[0]              | D:          | COM22       | 0240000028884e450024700f6bf000408021000097969900 | 0243            |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
PS C:\Users\toyom> mbedls --debug
DEBUG:mbedls.main:mbed-ls ver. 1.4.3
DEBUG:mbedls.main:host: ('nt', 'Windows', '10', '10.0.16299', 'win32')
DEBUG:mbedls.lstools_win7:_iter_vals <PyHKEY object at 0x00000000059F2390>
DEBUG:mbedls.lstools_win7:Mount point D: found for volume _??_USBSTOR#Disk&Ven_MBED&Prod_VFS&Rev_0.1#0240000028884e450024700f6bf000408021000097969900&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}
DEBUG:mbedls.lstools_win7:Fetching mounted devices from volume service registry entry
DEBUG:mbedls.lstools_win7:_iter_vals <PyHKEY object at 0x00000000059F2390>
DEBUG:mbedls.lstools_win7:Fetching usb storage devices from USBSTOR service registry entry
DEBUG:mbedls.lstools_win7:_iter_vals <PyHKEY object at 0x00000000059F2390>
DEBUG:mbedls.lstools_win7:target_id_usb_id -> mount_point mapping: {u'0240000028884e450024700f6bf000408021000097969900': 'D:'}
DEBUG:mbedls.lstools_win7:Device SYSTEM\CurrentControlSet\Enum\USB\VID_0D28&PID_0204\0240000028884e450024700f6bf000408021000097969900 did not have a "ParentIdPrefix" key, sticking with 0240000028884e450024700f6bf000408021000097969900 as entry key string
DEBUG:mbedls.lstools_win7:Composite device subdevice key SYSTEM\CurrentControlSet\Enum\USB\VID_0D28&PID_0204&MI_02 was not found, skipping
DEBUG:mbedls.lstools_win7:Unknown capabilities from the following ids: [u'usb\\class_03&subclass_00&prot_00', u'usb\\class_03&subclass_00', u'usb\\class_03']
DEBUG:mbedls.lstools_win7:Composite device subdevice key SYSTEM\CurrentControlSet\Enum\USB\VID_0D28&PID_0204&MI_04 was not found, skipping
DEBUG:mbedls.lstools_base:Candidates for display [{'mount_point': 'D:', 'serial_port': u'COM22', 'target_id_usb_id': u'0240000028884e450024700f6bf000408021000097969900'}]
DEBUG:mbedls.lstools_win7:Mount point D: is ready
DEBUG:mbedls.lstools_base:Found target id 0240000028884e450024700f6bf000408021000097969900 in htm line window.location.replace("https://mbed.org/device/?code=0240000028884e450024700f6bf000408021000097969900?version=0243?target_id=002effffffffffff4e453105100a002e");

DEBUG:mbedls.lstools_base:Found htm target id, 0240000028884e450024700f6bf000408021000097969900, for usb target id 0240000028884e450024700f6bf000408021000097969900
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| K64F          | K64F[0]              | D:          | COM22       | 0240000028884e450024700f6bf000408021000097969900 | 0243            |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
DEBUG:mbedls.main:Return code: 0
PS C:\Users\toyom>
```

closed 2018-04-25 by theotherjimmy

bridadan, 2018-04-26:
This should be fixed in version 1.4.4, now on pip

PR #333 (2018-04-25): Fix Windows 7 detection
This should fix #332.

The issue was Windows 7 PCs don't have a "volume" service. I have switched the implementation to use the "Disk" service instead. It seems to work well on Windows 10 and Windows 7.

@toyowata Would it be possible for you to test this when you get a chance? I'll be doing my own tests on my Japanese Windows 7 VM, but it'd be good to verify this on your hardware as well.