ARMmbed/mbed-ls #336: Indicating about ASSERT.TXT

jupe, 2018-04-27:
In CI we need to know if interface asserts and ASSERT.TXT appears to mount point. Would be optimal solution if mbed-ls could indicate if ASSERT.TXT file exists when list boards. mbed-ls could e.g. read that file when it exists and include it in board json, or just report all files in mount point so that application could read them when needed. This would help to avoid using bad state boards in CI.

bridadan, 2018-04-27:
Thanks for raising this @jupe.

This information is already present, we just aren't adding it to the JSON blob. The data can be accessed here: https://github.com/ARMmbed/mbed-ls/blob/master/mbed_lstools/lstools_base.py#L212

jupe, 2018-06-05:
Any change to add available files to JSON blob - in CLI interface ?

bridadan, 2018-06-05:
Not at this time. If I get time to patch #352 today I'll add this as well in a different PR.

PR #354 (2018-06-07): Adding directory entries to platform data
This fixes #336
closed 2018-06-07 by theotherjimmy