ARMmbed/mbed-ls #337: NRF51_DK incorrect detection on Windows 10 x64

andrewleech, 2018-05-07:
Hi, I'm trying to run `mbed test` on my windows 10 development machine against a standard NRF51_DK board.
In 1.4.4 and 1.4.3 it's being detected as:
```
+---------------+----------------------+-------------+-------------+--------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id    | daplink_version |
+---------------+----------------------+-------------+-------------+--------------+-----------------+
| LPC2368       | LPC2368[0]           | K:          | COM3        | 000681078661 | unknown         |
+---------------+----------------------+-------------+-------------+--------------+-----------------+
```
That target id is also visible in device manager under the Parent property as `USB\VID_1366&PID_1015\000681078661`
I've traced the detection to looking at the first 4 characters in target_id and mapping that in the json database to "0006": "LPC2368"

mbed_lstools/lstools_base.py:143
`platform_data = self.plat_db.get(device['target_id_usb_id'][0:4], verbose_data=True)`

In 1.4.2 it comes up as unknown. It's unknown all the way back to 1.3.3, 1.3.2 doesn't run due too logging import error then 1.3.1 doesn't detect any device.

1.2.16 detects my board perfectly, though the current mbed-greentea won't run against this old a version of mbedls.
```
+---------------+----------------------+-------------+-------------+--------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id    | daplink_version |
+---------------+----------------------+-------------+-------------+--------------+-----------------+
| NRF51_DK      | NRF51_DK[0]          | K:          | COM3        | 000681078661 | unknown         |
+---------------+----------------------+-------------+-------------+--------------+-----------------+
```

bridadan, 2018-05-08:
Hi @andrewleech, thanks for reporting the issue, we'll take a look.

Can you give us a little more information?

1. The output of `mbedls --debug`
1. What interface firmware is running on the board? stock? JLink? DAPLink?
1. Can you give me a list of all filenames on the device?

bridadan, 2018-05-08:
I was able to reproduce this with a JLink flashed NRF51_DK, so I'm assuming that's the configuration you have for now.

andrewleech, 2018-05-08:
It's stock configuration which is a composite interface segger host chip providing jlink, com port and mass storage. 
The mass storage has mbed.htm, user guide.htm and (iirc) segger.htm
The user guide does have a nordic URL which could be used the same way as the nrf52 one in the jlink section of the parts db however I think my patch implementation will provide compatibility for more existing boards.

theotherjimmy, 2018-09-24:
@bridadan Could you retest with the new jlink firmware detection?

bridadan, 2018-09-24:
I can't for the life me get an nrf51 flashed with the JLink software for whatever reason. I already have a test case for the nrf52 running jlink here: https://github.com/ARMmbed/mbed-ls/tree/master/test/test_data/nrf52_dk_jlink

I think we can be reasonably confident its also working for the nrf51, considering they share the same JLink firmware.

andrewleech, 2018-09-24:
Sorry I don't have an nrf51_dk any more, moved onto different projects

theotherjimmy, 2018-09-25:
Thanks for trying guys!

bridadan, 2019-01-07:
Just tried to reproduce this at my desk, but everything seems to be working now, so closing this.

closed 2019-01-07 by bridadan